esp8266/Arduino #733: mkspiffs partition seems to be ReadOnly when mounted into an ESP

martinayotte, 2015-08-30:
After mounting a partition made with mkspiffs, all files are readable, but opening a new file, either in "a" or "w" mode is failing.
Also, trying to open existing one in "a" mode will not fail but the content is never written.

igrr, 2015-09-08:
I tried a few ways, but it seems I can't reproduce this issue. Here's what I do:

- Prepare FS image:
```bash
mkdir -p data
echo "hello" > data/data.txt
./mkspiffs -b 0x1000 -p 0x100 -s 0x10000 -c data fs.bin
```
- Create and upload the sketch:
```c++
#include "FS.h"

void setup() {
  Serial.begin(115200);
  Serial.setDebugOutput(true);
  for (int i = 0; i < 5; ++i) {delay(1000); Serial.print(".");} // give me some time to open terminal window
  Serial.println("");
  SPIFFS.begin();
  const char* fname = "/data.txt";
  {
    File f = SPIFFS.open(fname, "a");
    if (!f) {
      Serial.printf("%s doesn't exist\r\n", fname);
      return;
    }
    Serial.printf("Opened %s\r\n", fname);
    Serial.printf("Size: %d\r\n", f.size());
    Serial.println("adding more data");
    f.println("more data");
  }
  {
    File f = SPIFFS.open(fname, "r");
    Serial.printf("Size: %d\r\n", f.size());
  }
}

void loop() { }
```
- Upload FS image: 
```bash
esptool-ck -cp /dev/tty.usbserial -cb 921600 -ca 0x6b000 -cf fs.bin
```

Here is what i get (with debug output enabled just in case):
```
.........
SPIFFSImpl: allocating 512+180+1400=2092 bytes
SPIFFSImpl: mounting fs @6b000, size=10000, block=1000, page=100
SPIFFSImpl: mount rc=0
Opened /data.txt
Size: 6
adding more data
SPIFFS_close: fd=1
Size: 17
SPIFFS_close: fd=1
```

So the initial file size was correct and we have added some more data there successfully.

martinayotte, 2015-09-09:
Hi IGRR,
Ok, thanks ! 
I will try your scenario tomorrow.
Maybe this is depends of the context, I don't even know if it is on ArduinoESP side or mkspiffs side.
So, maybe I should provide here my spiff image which contain more files and bigger ones, which can lead to some confusion. "mkspiffs -i myspiffs.bin" showingthe following : 

   0 /ddd/d/d/d/i/dd	era_cnt: 0
   1 d/d/d/d/d/d/d/d	era_cnt: 0
   2 /d/d/d/d/d/d/di	era_cnt: 0
   3 /ddd/d/d/d/i___	era_cnt: 0
   4 _______________	era_cnt: 0
   5 _______________	era_cnt: 0
   6 _______________	era_cnt: 0
   7 _______________	era_cnt: 0
   8 _______________	era_cnt: 0
   9 _______________	era_cnt: 0
  10 _______________	era_cnt: 0
  11 _______________	era_cnt: 0
  12 _______________	era_cnt: 0
  13 _______________	era_cnt: 0
  14 _______________	era_cnt: 0
  15 _______________	era_cnt: 0
era_cnt_max: 1
last_errno:  0
blocks:      16
free_blocks: 12
page_alloc:  32
page_delet:  25
used:        8032 of 52961
total: 52961
used: 8032

Here is the image in attachment, it is a ZIP file, although GIF extension is added simply to bypass attachment github limitation : 

[image]

igrr, 2015-09-09:
Which options did you use to build this image (i.e. something like `-b 0x1000 -p 0x100 -s 0x10000`)?

martinayotte, 2015-09-09:
I didn't provide any, so it was using the defaults, so according to main.cpp, it was using -s 0x10000 -p 256 -b 4096.
And if I remember on the first try with debug output, it iwas reportnig Ok. 
Also, the files can be access properly, but in ReadOnly.

igrr, 2015-09-09:
Oh, but do these defaults match SPIFFS settings on the ESP side? Which board/flash size setting did you use in the IDE?

martinayotte, 2015-09-09:
Hi again Ivan,
I'm using 512K(64K).
I've just re-merged all latest changes for GitHub, it seems the problem is gone ... :-)
So, if it is not reproducible any, I'm closing the issue for now !
(I will re-open it if it comes back)

closed 2015-09-09 by martinayotte