esp8266/Arduino #734: Enable section garbage collection for linker (-Wl,--gc-sections)

spapadim, 2015-08-30:
Current setup does not remove unused code and constants. Fixing this is a simple change to `platforms.txt`; I changed two lines, for `compiler.c.flags` and `compiler.c.elf.flags` to:

`compiler.c.flags=-c -Os -g -Wpointer-arith -Wno-implicit-function-declaration -Wl,-EL -fno-inline-functions -nostdlib -mlongcalls -mtext-section-literals -falign-functions=4 -MMD -std=gnu99 -ffunction-sections -fdata-sections`

`compiler.c.elf.flags=-g -Os -nostdlib -Wl,--gc-sections -Wl,--no-check-sections -u call_user_start -Wl,-static "-L{compiler.sdk.path}/lib" "-L{compiler.sdk.path}/ld" "-T{build.flash_ld}" -Wl,-wrap,system_restart_local -Wl,-wrap,register_chipv6_phy`

Let me know if you'd like a PR, but the changes are very simple: adding `-ffunction-sections -fdata-section` to the first and `-Wl,--gc-sections`.

I've tested this on a port of olikraus's excellent ucglib to the ESP; see https://github.com/spapadim/ucglib and olikraus/ucglib#61.  The code compiles without issues with these simple changes (otherwise fails, since ucglib compiles all font data, several MB, putting each in a separate section, and relying on the linker to pick the ones actually used).

spapadim, 2015-09-13:
Got the ucglib port running, and successfully strip unused global vars for font data.  This appears to need support for multiple progmem sections, which also requires a simple edit to the linker scripts -- changing issue title to better reflect this.

If you can think of a better way to accomplish the same, please do let me know.  Anything else I tried to get those global font vars removed by linker failed.  The current approach mirrors what olikraus did for the AVR arch.

The patches are https://gist.github.com/spapadim/a4bc258df47f00831006 and the ucglib port that makes use of this feature is https://github.com/spapadim/ucglib -- note that one of the patches is in the gcc toolchain.

igrr, 2015-09-14:
I understand the need for `-fdata-sections` to strip font data, but why does this require `-ffunction-sections`?

spapadim, 2015-09-14:
I basically copied avr-gcc flags Arduino uses on this :) -- can try w/o -ffunction-sections.  

However, the essential bit seems to be support for explicit separate irom sections (and progmem sections on AVR), relevant defines are here:
https://github.com/spapadim/ucglib/blob/master/csrc/ucg.h#L105
and how they're used e.g., here: https://github.com/spapadim/ucglib/blob/master/csrc/ucg.h#L1292

I can confirm that just just setting `UCG_FONT_SECTION` to `.irom.text` for all font data does not strip the unused ones, even with -fdata-sections and -Wl,-gc-sections flags.

igrr, 2015-09-14:
That's right, `-fdata-sections` doesn't work when you specify section explicitly in an attribute.
Data gets placed into one of the 'common' sections which will not get garbage collected.
Same thing applies to functions as well. We need an additional `objcopy --rename-section` step in order for this to work.

I think for the change above you only need `-Wl,-gc-sections` plus the change to the linker script.

spapadim, 2015-09-15:
Yes, you are right, the -f flags are unnecessary -- just verified.  Gist updated.

Now it makes sense btw, thanks -- unused code/data gc unit is always section, and if you want to remove unused data from progmem/irom (which necessarily has an explicit __attribute__), you need to play the tricks olikraus does (and I was wondering why he did it!).

So this would also imply that ICACHE_FLASH_ATTR functions also won't be removed, even if unused?  I need to RTFM on objcopy, but it would be nice if removal of unused functions is also enabled. E.g., adding only `#include <Wire.h>` without ever using it adds all of the Wire library ( + soft-spi code) in the executable on the ESP, but not on AVR -- I got bitten by this a couple of weeks back.

igrr, 2015-10-06:
@spapadim I have merged #857, so you can now remove ICACHE_FLASH_ATTR from your functions, as contents of .c files is now placed into flash automatically. You can use ICACHE_RAM_ATTR to override this, placing code into RAM.
gc-sections should also work now.

closed 2015-11-30 by igrr