esp8266/Arduino #735: Jason library -

asetyde, 2015-08-31:
I've tried to use this very good library 

https://github.com/bblanchon/ArduinoJson

But we 've a stack error i think and we need help to find this bug .. if someone can help  

original bug post :   https://github.com/bblanchon/ArduinoJson/issues/104

Also noticing on the ESP8266, that parsing anything but a statically defined const char*, kills the parser.

#include <ArduinoJson.h>

void setup() {

  Serial.begin( 115200 );
  delay( 10 );
  Serial.println( "" );

  StaticJsonBuffer<200> jsonBuffer;

  String json = "{\"mqtt.host\":\"mqtt.test.com\",\"mqtt.port\":1883}";

  Serial.println( "Parsing json string" );
  JsonObject& root = jsonBuffer.parseObject( json );

  if( root.success() ) {
    Serial.println( "parsed" );
  } else {
    Serial.println( "failed" );
  }
}

void loop() {
}
Results in

Parsing json string

Exception (9):
epc1=0x402021cd epc2=0x00000000 epc3=0x00000000 excvaddr=0x3ffe9d67 depc=0x00000000

ctx: cont 
sp: 3ffe9b40 end: 3ffe9e30 offset: 01a0

>>>stack>>>
3ffe9ce0:  3fffdc20 00000013 3ffe8de4 402039a9  
3ffe9cf0:  3ffe8968 00000000 3ffe8de4 3ffe9e5c  
3ffe9d00:  3fffdc20 00000000 3ffe8de4 40202216  
3ffe9d10:  3ffe9d30 3ffe9d34 3ffe9d34 4020450a  
3ffe9d20:  3fffdc20 00000000 3ffe8de4 4020214e  
3ffe9d30:  3ffe8878 716d227b 682e7474 2274736f  
3ffe9d40:  716d223a 742e7474 2e747365 226d6f63  
3ffe9d50:  716d222c 702e7474 2274726f 3838313a  
3ffe9d60:  00007d33 00000000 00000000 00000000  
3ffe9d70:  00000000 00000000 00000000 00000000  
3ffe9d80:  00000000 00000000 00000000 00000000  
3ffe9d90:  00000000 00000000 00000000 00000000  
3ffe9da0:  00000000 00000000 00000000 00000000  
3ffe9db0:  00000000 00000000 00000000 00000000  
3ffe9dc0:  00000000 00000000 00000000 00000000  
3ffe9dd0:  00000000 00000000 00000000 00000000  
3ffe9de0:  00000000 00000000 00000000 00000000  
3ffe9df0:  00000000 00000000 00000000 00000037  
3ffe9e00:  3fff38a8 0000002e 0000002e 00000000  
3ffe9e10:  00000000 00000000 3ffe9e54 40201a07  
3ffe9e20:  00000000 00000000 3ffe8e10 40100378  
<<<stack<<<

my :

Exception (9):
epc1=0x40202359 epc2=0x00000000 epc3=0x00000000 excvaddr=0x3ffe93f7 depc=0x00000000

ctx: cont 
sp: 3ffe91c0 end: 3ffe97e0 offset: 01a0

>>>stack>>>
3ffe9360:  3ffe9838 00000012 3ffe93a0 40202a25  
3ffe9370:  3fffdc20 00000013 3ffe9838 402039c5  
3ffe9380:  3ffe8504 00000000 3ffe9838 3ffe980c  
3ffe9390:  3fffdc20 3fff09f8 3ffe9838 402023b0  
3ffe93a0:  3ffe93c0 3ffe93c4 3ffe93c4 4020480a  
3ffe93b0:  3fffdc20 3fff09f8 3ffe9838 402021ee  
3ffe93c0:  3ffe8480 716d227b 682e7474 2274736f  
3ffe93d0:  716d223a 742e7474 2e747365 226d6f63  
3ffe93e0:  716d222c 702e7474 2274726f 3838313a  
3ffe93f0:  00007d33 00000000 00000000 00000000  
3ffe9400:  00000000 00000000 00000000 00000000  
3ffe9410:  00000000 00000000 00000000 00000000  
3ffe9420:  00000000 00000000 00000000 00000000  
3ffe9430:  00000000 00000000 00000000 00000000  
3ffe9440:  00000000 00000000 00000000 00000000  
3ffe9450:  00000000 00000000 00000000 00000000  
3ffe9460:  00000000 00000000 00000000 00000000  
3ffe9470:  00000000 00000000 00000000 00000000  
3ffe9480:  00000000 00000000 00000000 00000000  
3ffe9490:  00000000 00000000 00000000 00000000  
3ffe94a0:  00000000 00000000 00000000 00000000  
3ffe94b0:  00000000 00000000 00000000 00000000  
3ffe94c0:  00000000 00000000 00000000 00000000  
3ffe94d0:  00000000 00000000 00000000 00000000  
3ffe94e0:  00000000 00000000 00000000 00000000  
3ffe94f0:  00000000 00000000 00000000 00000000  
3ffe9500:  00000000 00000000 00000000 00000000  
3ffe9510:  00000000 00000000 00000000 00000000  
3ffe9520:  00000000 00000000 00000000 00000000  
3ffe9530:  00000000 00000000 00000000 00000000  
3ffe9540:  00000000 00000000 00000000 00000000  
3ffe9550:  00000000 00000000 00000000 00000000  
3ffe9560:  00000000 00000000 00000000 00000000  
3ffe9570:  00000000 00000000 00000000 00000000  
3ffe9580:  00000000 00000000 00000000 00000000  
3ffe9590:  00000000 00000000 00000000 00000000  
3ffe95a0:  00000000 00000000 00000000 00000000  
3ffe95b0:  00000000 00000000 00000000 00000000  
3ffe95c0:  00000000 00000000 00000000 00000000  
3ffe95d0:  00000000 00000000 00000000 00000000  
3ffe95e0:  00000000 00000000 00000000 00000000  
3ffe95f0:  00000000 00000000 00000000 00000000  
3ffe9600:  00000000 00000000 00000000 00000000  
3ffe9610:  00000000 00000000 00000000 00000000  
3ffe9620:  00000000 00000000 00000000 00000000  
3ffe9630:  00000000 00000000 00000000 00000000  
3ffe9640:  00000000 00000000 00000000 00000000  
3ffe9650:  00000000 00000000 00000000 00000000  
3ffe9660:  00000000 00000000 00000000 00000000  
3ffe9670:  00000000 00000000 00000000 00000000  
3ffe9680:  00000000 00000000 00000000 00000000  
3ffe9690:  00000000 00000000 00000000 00000000  
3ffe96a0:  00000000 00000000 00000000 00000000  
3ffe96b0:  00000000 00000000 00000000 00000000  
3ffe96c0:  00000000 00000000 00000000 00000000  
3ffe96d0:  00000000 00000000 00000000 00000000  
3ffe96e0:  00000000 00000000 00000000 00000000  
3ffe96f0:  00000000 00000000 00000000 00000000  
3ffe9700:  00000000 00000000 00000000 00000000  
3ffe9710:  00000000 00000000 00000000 00000000  
3ffe9720:  00000000 00000000 00000000 00000000  
3ffe9730:  00000000 00000000 00000000 00000000  
3ffe9740:  00000000 00000000 00000000 00000000  
3ffe9750:  00000000 00000000 00000000 00000000  
3ffe9760:  00000000 00000000 00000000 00000000  
3ffe9770:  00000000 00000000 00000000 00000000  
3ffe9780:  00000000 00000000 00000000 00000000  
3ffe9790:  00000000 00000000 00000000 00000000  
3ffe97a0:  00000000 00000000 00000000 00000037  
3ffe97b0:  3fff09f8 0000002e 0000002e 00000000  
3ffe97c0:  00000000 00000000 3ffe9804 4020185f  
3ffe97d0:  00000000 00000000 3ffe87c0 40100374  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,0)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld

Parsing json string

Exception (9):
epc1=0x40202359 epc2=0x00000000 epc3=0x00000000 excvaddr=0x3ffe93f7 depc=0x00000000

ctx: cont 
sp: 3ffe91c0 end: 3ffe97e0 offset: 01a0

>>>stack>>>
3ffe9360:  3ffe9838 00000012 3ffe93a0 40202a25  
3ffe9370:  3fffdc20 00000013 3ffe9838 402039c5  
3ffe9380:  3ffe8504 00000000 3ffe9838 3ffe980c  
3ffe9390:  3fffdc20 3fff09f8 3ffe9838 402023b0  
3ffe93a0:  3ffe93c0 3ffe93c4 3ffe93c4 4020480a  
3ffe93b0:  3fffdc20 3fff09f8 3ffe9838 402021ee  
3ffe93c0:  3ffe8480 716d227b 682e7474 2274736f  
3ffe93d0:  716d223a 742e7474 2e747365 226d6f63  
3ffe93e0:  716d222c 702e7474 2274726f 3838313a  
3ffe93f0:  00007d33 00000000 00000000 00000000  
3ffe9400:  00000000 00000000 00000000 00000000  
3ffe9410:  00000000 00000000 00000000 00000000  
3ffe9420:  00000000 00000000 00000000 00000000  
3ffe9430:  00000000 00000000 00000000 00000000  
3ffe9440:  00000000 00000000 00000000 00000000  
3ffe9450:  00000000 00000000 00000000 00000000  
3ffe9460:  00000000 00000000 00000000 00000000  
3ffe9470:  00000000 00000000 00000000 00000000  
3ffe9480:  00000000 00000000 00000000 00000000  
3ffe9490:  00000000 00000000 00000000 00000000  
3ffe94a0:  00000000 00000000 00000000 00000000  
3ffe94b0:  00000000 00000000 00000000 00000000  
3ffe94c0:  00000000 00000000 00000000 00000000  
3ffe94d0:  00000000 00000000 00000000 00000000  
3ffe94e0:  00000000 00000000 00000000 00000000  
3ffe94f0:  00000000 00000000 00000000 00000000  
3ffe9500:  00000000 00000000 00000000 00000000  
3ffe9510:  00000000 00000000 00000000 00000000  
3ffe9520:  00000000 00000000 00000000 00000000  
3ffe9530:  00000000 00000000 00000000 00000000  
3ffe9540:  00000000 00000000 00000000 00000000  
3ffe9550:  00000000 00000000 00000000 00000000  
3ffe9560:  00000000 00000000 00000000 00000000  
3ffe9570:  00000000 00000000 00000000 00000000  
3ffe9580:  00000000 00000000 00000000 00000000  
3ffe9590:  00000000 00000000 00000000 00000000  
3ffe95a0:  00000000 00000000 00000000 00000000  
3ffe95b0:  00000000 00000000 00000000 00000000  
3ffe95c0:  00000000 00000000 00000000 00000000  
3ffe95d0:  00000000 00000000 00000000 00000000  
3ffe95e0:  00000000 00000000 00000000 00000000  
3ffe95f0:  00000000 00000000 00000000 00000000  
3ffe9600:  00000000 00000000 00000000 00000000  
3ffe9610:  00000000 00000000 00000000 00000000  
3ffe9620:  00000000 00000000 00000000 00000000  
3ffe9630:  00000000 00000000 00000000 00000000  
3ffe9640:  00000000 00000000 00000000 00000000

asetyde, 2015-08-31:
this is StaticJsonBuffer.hpp , probably bug affected 

i think something is not correct .. but no problem with avr , only esp
#pragma once

#include "JsonBuffer.hpp"

namespace ArduinoJson {

// Implements a JsonBuffer with fixed memory allocation.
// The template paramenter CAPACITY specifies the capacity of the buffer in
// bytes.
template <size_t CAPACITY>
class StaticJsonBuffer : public JsonBuffer {
 public:
  explicit StaticJsonBuffer() : _size(0) {}

  size_t capacity() const { return CAPACITY; }
  size_t size() const { return _size; }

 protected:
  virtual void* alloc(size_t bytes) {
    if (_size + bytes > CAPACITY) return NULL;
    void* p = &_buffer[_size];
    _size += bytes;
    return p;
  }

 private:
  uint8_t _buffer[CAPACITY];
  size_t _size;
};
}

igrr, 2015-08-31:
Relevant discussion happens in https://github.com/bblanchon/ArduinoJson/issues/104, closing this one.

closed 2015-08-31 by igrr

asetyde, 2015-08-31:
Russian is the best :)  :+1:  thanks @igrr  , also when it'll work , i think is good idea to add this library , very useful to platform data crossing ( i use between c and java )