esp8266/Arduino #737: Dynamic WiFi.begin() issue

overdrive2max, 2015-08-31:
When I call WiFi.begin("xxxx","xxxxx"); my module connects immediately but when I use variables, i.e. WiFi.begin(ssid,pass); or WiFi.begin((const char*)ssid,(const char*)pass); it does not connect at all.
What can be the problem?

Example does NOT work:
WiFi.begin((const char*)ssid, (const char*)pass);
			int retries = 0;
			while ((WiFi.status() != WL_CONNECTED) && (retries < 10)) {
				retries++;
				delay(500);
				Serial.println("Connecting");
			}

Example does work:
WiFi.begin("TEST", "TESTPASS");
			int retries = 0;
			while ((WiFi.status() != WL_CONNECTED) && (retries < 10)) {
				retries++;
				delay(500);
				Serial.println("Connecting");
			}

Links2004, 2015-09-01:
you do a wrong type cast, the variables are pointer ( ```const char*```).
normally you not need to typecast simply use it like:
```cpp
const char ssid[] = "SSID";
const char pass[] = "PW";
WiFi.begin(ssid, pass);
```

igrr, 2015-09-01:
The code actually has the asterisk before const, but due to lack of code escaping this asterisk is eaten by Markdown parser which results in italic text.

I'll edit the question to add code tags.

@overdrive2max can you please show how you define `ssid` and `pass` variables in your sketch?

overdrive2max, 2015-09-01:
@igrr Thanks, I updated my post. The ssid and pass variables are reset when certain commands are received from serial. I uses memset(ssid,0,28) to initialize then read the new ssid/pass via serial into the variable, then call WiFi.begin.

igrr, 2015-09-01:
Okay, the piece of code you have posted should work. There might be issues elsewhere — like in the code where you read from serial. Something like trailing `\r\n` characters in SSID and password perhaps?
If you paste the complete sketch which demonstrates this issue, i may be able to assist.

overdrive2max, 2015-09-01:
As requested
```c++
/*
MQTT bridge 29-08-2015
==IN==
[$MQSEND [500] => [$SENT => Send normal message to set topic
[$MQBCAST [500] => [$SENT => Send broadcast to set broadcast topic
[$SSID [23] => SSID of wifi
[$WPASS [64] => Password of WIFI network
[$MQSERV [64] => MQ server IP or names
[$MQUSER [23] => MQ user name if required
[$MQPASS [23] => MQ password if required
[$MQSUB [23] => MQ subscribe to command topic
[$MQSUBB [23] =>MQ subscribe to broadcast topic
[$MQPUB [23] => set normal MQ topic for response to commands
[$MQPUBB [23] => set broadcast topic to post to if needed
[$MQPORT [long] => MQ port
[$STOP! => Stop all connections and go into IDLE mode
[$GO! => Start connections and subscriptions. write [$READY! when all is connected.
[$STS? => [$MQ OFFLINE , [$MQ CONNECTED, [$OFFLINE , [WIFI CONNECTED , [$IDLE
[$DEBUG => Toggle debug state

==OUT==
[$MQM [500] - Normal message
[$MQB [500] - Broadcast message
[$MQ? [500] - unknow topic message

==Messages==
[$IDLE => Waiting for commands, not connected
[$WIFI CONNECTED => ...
[$MQ CONNECTED => ...
[$MQ DISCONNECTED, RETRY IN 2sec => ..
[$MQ DROPPED => MQ disconnected
[$OFFLINE => not connected to wifi
[$READY! => connected, you can abuse me
*/

#include <ESP8266WiFi.h>
#include <PubSubClient.h>

//const char *ssid 	// cannot be longer than 32 characters!
 char ssid[28];
 char pass[50];		//
 char mqServer[64] ;
 char mqUser[23];
 char mqPass[23];
 char subCMD[38];
 char subBCast[38];
 char pubNormal[38] ;
 char pubBCast[38];
 uint16_t mqPort = 1883;
 bool eStop = false;
 bool GO = false;

 bool debug = false;
 WiFiClient wclient;
 PubSubClient client(wclient, mqServer, mqPort);

void callback(const MQTT::Publish& pub) {
	// handle message arrived
	if (pub.topic().startsWith(subCMD))
	{
		Serial.print("[$MQM ");
	}
	else if (pub.topic().startsWith(subBCast))
	{
		Serial.print("[$MQB ");
	}
	else
	{
[... 470 lines cut ...]
				Serial.print(F("DEBUG: MQ SUB: "));
				Serial.println(subCMD);
				delay(1);
			
				Serial.print(F("DEBUG: MQ SUB BCAST: "));
				Serial.println(subBCast);
				delay(1);
		
				Serial.print(F("DEBUG: PUB: "));
				Serial.println(pubNormal);
				delay(1);
			
				Serial.print(F("DEBUG: PUB BCAST: "));
				Serial.println(pubBCast);
				delay(1);
			
				Serial.print(F("DEBUG: MQ PORT: "));
				Serial.println(mqPort);
				delay(1);
			//==========================================
		}

	
	}
```

igrr, 2015-09-01:
Like I said, your code captures trailing newline character and puts it into the ssid/password strings.

Added the following after `Serial.println(ssid);`:
 
```c++
        for (int i = 0; i < strlen(ssid); ++i) {
          Serial.printf("%02x ", ssid[i]);
        }
        Serial.println("");
```

And got this:
```
[$IDLE
DEBUG: SERIAL MESSAGE: SSID wifinet123
DEBUG: GOT SSID: wifinet123
77 69 66 69 6e 65 74 31 32 33 0d 
```

See the `0d` character at the end of the string?

overdrive2max, 2015-09-01:
Thank you for pointing that out, I will go have a look quickly and report back.

overdrive2max, 2015-09-01:
It seems to be working now. Thank you very much!

closed 2015-09-01 by overdrive2max

mickaylov, 2015-10-03:
@igrr I just want to say thank you for your implication and deep analyses on esp8266. This 0d is not easy to see :+1: thanks for everything (OTA, spiffs, mkspiffs ........)