esp8266/Arduino #738: SPIFFS limit of 500 characters while adding a a line.

JeroenBeemster, 2015-09-01:
Hi Igrr,
 Here my 2 issues.

1) I try to use the filesystem SPIFFS (currently only available in staging environment).
 After opening the an existing file with append 'File f = SPIFFS.open(AccessPointsFileName, "a") ' and try to add some text with 'f.println("something") ' everything works fine until I step over a limit of 500 characters (I guess). The worst part of it, is that it will also corrupt the file system and have to reflash it. I solved the reflashing by trial and error. 
 2) How do I "format" the flash memory in the area where the file system is stored.

Here is my test application:

#include "FS.h"
 #define AccessPointsFileName "/accesspoints.txt" 

boolean halt = false;

void setup() {
 Serial.begin(115200);
 if(!SPIFFS.begin()) {
 halt = true;
 Serial.println("Begin failed");

}

if(!SPIFFS.remove(AccessPointsFileName)) {
 //halt = true;
 Serial.println("remove failed");
 }

Serial.println("Setup done");
 delay(10);
 }

void loop() { 
 if (halt == false) {
 halt = test();
 }
 }

boolean test() {
 Serial.print(".");
 File f = SPIFFS.open(AccessPointsFileName, "a");
 if (!f) {
 Serial.println("file open failed");
 return true;
 }
 Serial.print("Size :");Serial.println(f.size());

f.println("buffer");

f.close();

f = SPIFFS.open(AccessPointsFileName, "r");
 if (!f) {
 Serial.println("file open failed");
 return true;

}

Serial.print("Size2:");Serial.println(f.size());

f.close();

delay(10);
 return false;
 }

This is the results:

Setup done
 Size :0
 Size2:8
 Size :8
 Size2:16
 Size :16
 Size2:24
 Size :24
 Size2:32
 Size :32
 Size2:40
 Size :40
 .
 .
 Size2:480
 Size :480
 Size2:488
 Size :488
 Size2:496
 Size :496
 file open failed

plerup, 2015-09-06:
I have had the same experience. File operations works for a while and then it crashes. Often leaving the flash corrupted and the ESP refuses to boot until reflashed. I think this is strange as SPIFFS works very well in NodeMCU...

igrr, 2015-09-08:
Thanks, issue confirmed — this was a bug in my flash HAL. I'm surprised it did actually work in many cases though. Fix coming up.

closed 2015-09-14 by igrr