esp8266/Arduino #739: esptool not listed in programmer menu

AkhilReddyMandadi, 2015-09-01:


igrr, 2015-09-01:
@AkhilReddyMandadi 
Please provide some description for this, i.e. why do you believe this is an issue.

AkhilReddyMandadi, 2015-09-01:
actually i selected the genericesp in the boards menu but esptool is not listed under the programmer menu. I tried uploading the code but it showing error

igrr, 2015-09-01:
Esptool is not supposed to be listed in programmers list. It doesn't matter what you select in that menu — upload will always happen using esptool.

Check your connections according to the [docs](https://github.com/esp8266/Arduino/blob/esp8266/hardware/esp8266com/esp8266/doc/boards.md). Visit the esp8266.com forum for guidance to troubleshooth this error.

closed 2015-09-01 by igrr

AkhilReddyMandadi, 2015-09-01:
thank you @igrr