esp8266/Arduino #741: HelloMesh, no connections.

gamecompiler, 2015-09-01:
Hi, id uploaded the HelloMesh example to six Esp´s. But no one connect to each other.
It fails while connecting to any _suitable_ APs which contain _ssid_prefix.

Id checked if the sketch scans correct for Ap s and if it filter the chip id s out.
That works fine. 

Maybe its import: current_ssid is always empty if id print them out over Serial with Serial.println(current_ssid);

The if sentence
if (index >= 0 && (target_chip_id < _chip_id)) {}
does never happen.

Sorry for my English.
Thanks in Advance.

igrr, 2015-09-01:
/cc @jtfell

grahamehorner, 2015-09-25:
I to have this issue I have 10 ESP8266 developer kits all running the same firmware version and the HelloMesh sketch, no data is getting transmitted/received by any node; or the Raspberry PI Internet gateway.

devyte, 2017-10-20:
@gamecompiler is this issue still valid with latest git?

gamecompiler, 2018-01-20:
No, it works now. Thank you very much!

closed 2018-01-20 by gamecompiler

WillingLearner, 2018-01-25:
I also have the same problem, I loaded the same code into the two ESP module but doesn't work. How did you solve your problem gamecompiler ?

sriksh9, 2018-01-30:
@WillingLearner I've uploaded the "HelloMesh" example to three ESP12Es without any modifications, they've connected each other and printed the messages. I don't think there is any such sort of problem anymore.