esp8266/Arduino #742: ESP8266 Crashes in random time after it starts running Arduino 'C' script

lnsriravichandran22, 2015-09-02:
Hello Everyone!!
I have been trying to run a script on my ESP8266 which would do the following things
Gets SSID and password from the user on reset, and would connect with the specified network, then it starts posting data to a remote host.  

Now the problem is that the device hangs while running the script. I thought that this would be because of too many prints in Serial port. Hence I removed the unnecessary prints. This didn't help.

Then I thought this would be because of in-sufficient power supply. Hence I powered the device with a 5V / 2A dc power supply and with the help of a regulator, I got 3.3V DC to the unit. This also didn't work out.

The device at times, posts data for 200 times, and at some time it posts up to 408 times. But hangs for sure at any random number.

Can someone please help me in this regard??

lnsriravichandran22, 2015-09-02:
To be more descriptive, 
for the very first time I would the device enters the Soft AP mode and gets SSID and password from user and then enters the station mode to connect with the specified network and then starts posting the data continuously . The device exactly hangs when posting data continuously!!

igrr, 2015-09-02:
Could you please post the sketch which demonstrates this issue?

lnsriravichandran22, 2015-09-03:
@igrr Thank you so much for the response!!. I didn't terminate the client's connection with server properly. That's where the issue was. For some reason, it goofed up and crashed. Now I have closed the client every time after posting the data. It goes good!!

Once again thanks a lot for the immediate responses!!

closed 2015-09-07 by lnsriravichandran22