esp8266/Arduino #744: How to get silent restart (ESP.restart()) on ESP01 ?

luc-github, 2015-09-03:
I do not want any system message when doing a restart (ESP.restart();) because any garbage on serial may affect the host connected on serial currently a 3D Printer.

what ever I do to hide it, it display this output when restarting:
```
 ets Jan  8 2013,rst cause:1, boot mode:(3,7)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld
```

I tried :

```Serial.end();``` before ```ESP.restart();``` and doing nothing
I set ```system_set_os_print(0);``` and ```Serial1.setDebugOutput(true);``` still have the anoying output

I tried ```Serial.swap()``` which does the trick but then it also crash the Serial completly so I loose the serial after restart,only a unplug/replug recover it. I guess it is because no GPIO15/13 on ESP01

Is there any way to hide this kind of output on ESP01 to get a clean restart ?

Thanks in advance

igrr, 2015-09-03:
Except for the final line, this output is from ROM bootloader of ESP8266.
There is no way to disable it in software.

luc-github, 2015-09-03:
I see
but any way to redirect it like Serial.swap() do without crashing the serial ?

igrr, 2015-09-03:
Please post the full sketch to help reproduce this issue.

igrr, 2015-09-03:
Plus, take note that GPIO15 (TX) is connected to ground on ESP01.

luc-github, 2015-09-03:
I just did a small sketch to show how swap crash serial -  

```

#include <ESP8266WiFi.h>

void setup() {
  // put your setup code here, to run once:
 Serial.begin(9600);
 delay(1000);
 Serial.print("Init done\n");
}
int i = 0;
void loop() {
  delay(1000);
  i=i+1;
  Serial.print("One loop\n");
  if (i==10)
	{
		Serial.print("Restart\n");
		//if adding swap serial is crashed - only unplug/replug recover serial
		//Serial.swap();
	    ESP.restart();
	    while (1){
			delay(1);};
	}
}
```

luc-github, 2015-09-04:
I have added a reset function in HardwareSerial.cpp that recover the swap function making serial lost 

```
void uart_reset(uart_t* uart) {
    pinMode(1, SPECIAL); //TX
    pinMode(3, SPECIAL); //RX
    IOSWAP &= ~(1 << IOSWAPU0);
    uart->txPin = 1;
    uart->rxPin = 3;
    pinMode(15, INPUT);
    pinMode(13, INPUT); //RX
}
```
and 
```
void HardwareSerial::reset() {
    if(_uart == 0)
        return;
    uart_reset(_uart);
}
```

This seems solving my issue with ESP01 and allow to hide all restart message using swap as wished

```
Init done
One loop
One loop
One loop
One loop
One loop
One loop
One loop
One loop
One loop
One loop
Restart
Init done
....
````
here the test sketch
```

#include <ESP8266WiFi.h>

void setup() {
  delay(1000);
  // put your setup code here, to run once:
 Serial.begin(115200);
 Serial.reset();
 delay(1000);
 Serial.print("Init done\n");
}
int i = 0;
void loop() {
  delay(1000);
  i=i+1;
  Serial.print("One loop\n");
  if (i==10)
	{
		Serial.print("Restart\n");
		Serial.flush();
		Serial.swap();
		delay(1000);
	    ESP.restart();
	    while (1){
			delay(1);};
	}
}
```

luc-github, 2015-09-05:
I changed title to match issue

luc-github, 2015-09-07:
Hi should I do a PR ? Or may be reset function is not correct approach ?

luc-github, 2015-09-08:
after several tests the only working solution to recover after a swap on ESP01 is to force the pins to 1 and 3, also forcing pins 13/15 generate a reset

so I have completed the reset function to check same conditions as swap one but forcing pin 1/3

```
void uart_reset(uart_t* uart ) {
	 if(uart == 0)
        return;
    switch(uart->uart_nr) {
        case UART0:
			pinMode(1, SPECIAL); //TX
			pinMode(3, SPECIAL); //RX
			IOSWAP &= ~(1 << IOSWAPU0);
			uart->txPin = 1;
			uart->rxPin = 3;
			pinMode(15, INPUT);
			pinMode(13, INPUT); 
		break;
        case UART1:
            // current no swap possible! see GPIO pins used by UART
            break;
        default:
            break;
    }
}
```
If it is ok, I will do a PR - if not, please let me know how to proceed, I would be more than happy to follow advice

Thank you

igrr, 2015-09-09:
Okay, i had trouble understanding what you mean by "serial is crashing". Apparently nothing is crashing, just the IOSWAP is not reset along with the core, and serial continues to run on pins 13/15 after reset. Fix coming up.

luc-github, 2015-09-09:
sorry my poor English should be the root cause, I meant I loose serial even after a restart, need unplug to recover

luc-github, 2015-09-09:
Verified fixed on ESP01- excellent - thank you

closed 2015-09-09 by luc-github