esp8266/Arduino #747: Heat up problem + post-boot problems

sybeck2k, 2015-09-03:
I've an esp03 connected via i2c to an attiny85. The esp03 has been working wonderfully, with both the original NodeMCU fw, and with the Arduino fw.
I've uploaded a new script to be able to do web-upload (WebUpdate.ino): I've uploaded the script, and now the esp is non responding - and it warms up (a lot).
I can still upload new firmware - I've even uploaded the NodeMCU fw back - but the firmware does not pass the boot phase. On the other hand, it does not become hot!

Here is the output of the boot phase when I've the Arduino sketch uploaded:
```
 ets Jan  8 2013,rst cause:1, boot mode:(3,7)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld
```

And here is the output when using NodeMCU:
```
ets Jan  8 2013,rst cause:1, boot mode:(3,7)

load 0x40100000, len 32028, room 16 
tail 12
chksum 0xd7
ho 0 tail 12 room 4
load 0x3ffe8000, len 2888, room 12 
tail 12
chksum 0xef
ho 0 tail 12 room 4
load 0x3ffe8b50, len 15348, room 12 
tail 8
chksum 0xc5
csum 0xc5
(+a sequence of unreadeable characters)
```
I've plenty of power available (1.5A), connection has been tested over and over (it's the same as the one that was working before uploading the WebUpdate script)... what could it be?

dcremonini, 2015-11-15:
Hi,
I have a similar problem: I uploaded through the Arduino 1.6.5-r5 IDE a program using the ESP8266WiFi library.

Application worked good for a while but after a few boots, the ESP8266 chip started becoming hot.
The power consumption at 3,3V raised to 480-490 mA that is more of the double of the declared value.

For me too, available current is not an issue. Any idea?

asetyde, 2015-11-15:
Power come from ? Usb ?

sybeck2k, 2015-11-15:
In my case power was from usb with a 3.3v regulator with 500mA max power

dcremonini, 2015-11-30:
@setyde: I used before USB, then with USB detached I used a DC  power supply.
Today I started testing with a brand new and different model: it's working. I still have to test it long time though.

asetyde, 2015-11-30:
However quality is not simple to find on esp ! Few reference product . 
Ps a tentoni :)

devyte, 2017-10-20:
This sounds like a board configuration error, i.e.: some pin was left in a state where it was drawing power in a bad way.
I don't have any heating issues with any of the ESP12-based devices.
Closing as can't reproduce, and due to age.

closed 2017-10-20 by devyte

kamalesh-p, 2022-01-26:
I have three ESP8266, among them one works fine.
But other two causes 
1) heating problem when CH_PD is connected to 3.3v, 
2) also the red light started glowing dim
3) shiowing this code as output
**********************
Lets Jan  8 2013,rst cause:1, boot mode:(3,6)

load 0x40100000, len 1396, room 16 
tail 4
chksum 0x89
load 0x3ffe8000, len 776, room 4 
tail 4
chksum 0xe8
load 0x3ffe8308, len 540, room 4 
tail 8
chksum 0xc0
csum 0xc0

2nd boot version : 1.4(b1)
  SPI Speed      : 40MHz
  SPI Mode       : DIO
  SPI Flash Size & Map: 8Mbit(512KB+512KB)
jump to run user1 @ 1000
Ai-Thinker Technology Co.,Ltd.

invalid
**********************

I do use the below circuit connection
[image]