esp8266/Arduino #748: Calling WiFi.status() changes status from WL_DISCONNECTED to WL_CONNECTED

6sn7, 2015-09-05:
A call to WiFi.status() before any connection is established does return WL_DISCONNECTED as expected. However, a subsequent call to WIFI.status() will return WL_CONNECTED even though no connection has been initiated.

My fix is to force a WiFi.disconnect(1);  first.

Version: esp8266/1.6.5-947-g39819f0

// example code:
void setup() {
  Serial.begin(115200);
  delay(100);

 //WiFi.disconnect(1);     //<<<<<<<<<<<<< calling disconnect first fixes the problem.
  int stat = WiFi.status();
 Serial.print("setup: wifi stat=");
 Serial.println(stat);     //  <<<<<<<<<<<<<  returns  WL_DISCONNECTED as expected.
 delay(2000);
  stat = WiFi.status();   //  <<<<<<<<<<<<<  returns  WL_CONNECTED
 Serial.print("setup: wifi stat2=");
 Serial.println(stat);
}

Output:
setup: wifi stat=6
setup: wifi stat2=3

Is there something I may be overlooking or doing incorrectly?

Links2004, 2015-09-05:
the SDK stores the connection info and connects automatically to the last config after boot.
this config is keeped eaven at re flash so its normal.

6sn7, 2015-09-05:
OK. Thanks for the clarification.

closed 2015-09-05 by 6sn7