esp8266/Arduino #749: CRC error on installing "Stable" Esp8266 support from boards manager

gordonendersby, 2015-09-05:
Xubuntu 14.04 and Arduino 1.6.5
When adding support for the esp8266 to the Arduino IDE using the stable json link the install fails and states theres a CRC error and the install fails.
http://arduino.esp8266.com/stable/package_esp8266com_index.json

Ive tried acomplete delete and re-install but still get the same crc error.
The staging version works fine.

Others have the same problem. Please see link to forum.
http://www.esp8266.com/viewtopic.php?f=26&t=4784

Gordon

igrr, 2015-09-05:
Fixed the package json file, please try again.

gordonendersby, 2015-09-05:
Thanks, works a treat.
This arduino ide support makes things very easy.
I still havnt got round to using the sdk.
But this is a quick way to get going with these boards.
Nice work.

gordonendersby, 2015-09-05:
Tested as working on arduino 1.6.5 under xubuntu 14.04.
Fixed, thanks.

closed 2015-09-05 by gordonendersby