esp8266/Arduino #750: Document spiffs image creation and flashing

giannoug, 2015-09-05:
It would be nice if someone gathered some information about creating spiffs images on the PC and uploading them to the ESP8266.

Also, it would be perfect if the IDE flashed any spiffs images it found in the sketch directory to the ESP8266.

pgollor, 2015-09-07:
Hi giannoug,

it is maybe not exactly what do you wat, but there are some informations to create a binary image with igrr mkspiffs tool an upload it to the flash:
https://github.com/pgollor/esp8266-webconf-mDNS-OTA#flashing-file-system

Greets
Kalle

pgollor, 2015-09-10:
Your wish should work with the newest staging release from yesterday. Therefore you can close this issue.

Greets
Kalle

giannoug, 2015-09-10:
It feels like Christmas already :smiley:

Thank you everyone!

closed 2015-09-10 by giannoug