esp8266/Arduino #752: A small nothing left aside. serialEvent()

fredericplante, 2015-09-06:
I, you guys seeem to forgot the serialEvent() trigger, that is required to be fully compatible with the basics example of the Arduino IDE.

I guess we could just program a trigger from within the program, you while you guys are at it, well here is the constructive remark ;)

Thanx :)

igrr, 2015-09-06:
I never new about it, thanks for pointing this out.
However, reading Arduino website,

serialEvent() is not compatible with the Esplora, Leonardo, or Micro

So it's not even supported by some of the Arduino brand boards.
Still, I will look into this once I add the callback queue and some
scheduling functions.

fredericplante, 2015-09-06:
Might as well be more discipline then them. ;)

fredericplante, 2015-09-06:
Esplora, Leonardo, or Micro are basically all Léonardo tech, that is made to become computer interface(mouse, keyboard, SPI and I2c interface that that lack on a computer) using usb. While all the other arduinos of the first génération, the one using an external UART2USB  are made to be small stand alone computing device, that can totally function without serial port, well that is what i deduce of what I get out of the product.

According to me, the emphasis of the SerialEvent() should of been put on the Léonardo class, and less on the stand alone class, but they did not think about it that way. Any ways, according to me both should have it, there is never enough  possible internal trigger and external interrupt.

Take care.

drmpf, 2015-09-07:
Is there any notable library that uses SerialEvent()?
Unless there is a particular need for SerialEvent(), I would vote against it.

I see no advantage over calling a method directly from the loop
Calling a method directly from the loop keeps user focused on what happens when and how long it takes. 

Most users, particularly casual users, will never use interrupts (apart form library code)  and the SerialEvent() runs like an interrupt with some of the associated problems. 
 
I note even on the boards the support SerialEvent() it is not uniform,  Mega does not have it, instead having SerialEvent1(), etc.

fredericplante, 2015-09-07:
Actually, the sérialevent() event is absolutly independant from the Loop() it's never call from there. T'it's actually the library that manage to trigger event by calling SerialEvent(), from the library.

Actually mini, uno, mega128 and mega256, and all the clone, has it. It is the atmega32u that does handle the event. Has for Due and Zero, im not sure.

I think it's important to put it in, to make sure that the new guy that try esp and Adruino ide together get the same look and feel as if it was a real Arduino. Just saying. I look at your work, and I see that you are trying to make it as transparent as possible, well there is a spot there. ;)

philomatic, 2016-02-05:
Anything new on this? Would glad to see this coming!

> Most users, particularly casual users, will never use interrupts (apart form library code) and the SerialEvent() runs like an interrupt with some of the associated problems.

I disagree. In my experience, almost everyone who programmed an Arduino, AtMega or MSP will use interrupts if available. And these are the users which will use Arduino on ESP8266.  All others will stick to Lua (or similiar HL languages).

Best regards

ArGaspar, 2016-11-27:
Hi, is SerialEvent() Available now?. im trying to send information via serial from arduino to esp and back the other way, ESP is acting as a web server so if use serial.read is going to block all the others function. Any suggestions?

Best Regards.

fredericplante, 2016-11-27:
I knew this was going to come up. It's not because it seam useless for some that it is for every one. I will take a look at the progerss

fredericplante, 2016-11-27:
Well I check at the https://github.com/esp8266/Arduino/blob/master/cores/esp8266/HardwareSerial.h

at line 146 and line 147 the procedure seam to be put in place. So maybe you should give it a try

fredericplante, 2016-11-27:
This part:

`
void serialEventRun(void)
{
#if defined(HAVE_HWSERIAL0)
  if (Serial0_available && serialEvent && Serial0_available()) serialEvent();
#endif
#if defined(HAVE_HWSERIAL1)
  if (Serial1_available && serialEvent1 && Serial1_available()) serialEvent1();
#endif

}`

in the  https://github.com/esp8266/Arduino/blob/master/cores/esp8266/HardwareSerial.cpp

ArGaspar, 2016-11-27:
Is it possible to modify that myself? I cant find that file/folder on my pc.

Best Regards

fredericplante, 2016-11-27:
Yeah! probably, if you know where your arduino platform distribution is, you can change it no problem. But every time you will reinstall, you will loose all, same when they will update.

Best would be to change at the root, but for some reason, some people think that if it's useless for them,  it has to be the same for every one. :(

ArGaspar, 2016-12-24:
Hey, I know where it its, but i cant find the file it should be on libraries right???? did you make it work?

philomatic, 2017-03-03:
I'm also out of luck, can't find the location of the files on my disk.
Any ideas?

Best Regards
Patrick

bertmelis, 2017-05-29:
Is there still an idea to reimplement serialEvent?
It's not so much of a deal since it would only add a check on `Serial.available()` on each `loop()` and run the serialEvent-routine if necessary.

It is **not** interrupt driven as in this snippet from [main.cpp](https://github.com/arduino/Arduino/blob/master/hardware/arduino/avr/cores/arduino/main.cpp):
```c++
for (;;) {
		loop();
		if (serialEventRun) serialEventRun();
	}
```
Obviously, it does add some overhead for those who haven't got serialEvent defined.

rsegecin, 2017-10-11:
@fredericplante when you added those lines you commented to make the serialEvent work were you working on ESP8266 Arduino version 2.3? Was it all you added? Could you make a pull request please?

devyte, 2017-10-20:
@igrr I'm rather confused with this one. Was the HardwareSerial code massively changed? The code references in this thread don't match with the current code for HardwareSerial for me.
What are your thoughts on this request?

laercionit, 2018-06-04:
Hello, I saw in the post history the adjustment in the Lib to support SerialEvent, it is a function very used in projects.

I would like to know if this improvement will be incorporated or is it more appropriate to use the Ticker to execute Serial.available?

fredericplante, 2018-06-12:
Humm good question! rsegecin, what was the actual platform in 2015?

To be honest, I'm back to study and it takes a good part of my life. lol

But it's summer time now, baby :D I will get back to programming, I have a whole lot of modules I received during the winter, compulsive buying did not go away, unlike coding, during those long hours of study, ;)

I will give news within the near future.

bertmelis, 2018-06-12:
While you could use Ticker to execute Serial.available I would'nt do that. Both use interrupts and they don't mix well together. Maybe use [schedule.h](https://github.com/esp8266/Arduino/blob/master/cores/esp8266/Schedule.h) instead?

Disclaimer: I didn't look into the inner workings of schedule yet. You'll have to reschedule after calling but I don't know if it's possible from within the called function.

hreintke, 2018-06-12:
@bertmelis @laercionit 
Yes to can call schedule within scheduled function.

This is part of my CommandExecutor class, I use it also with Serial.

```
void CommandExecutor::getStreamChars(Stream* reqStream)
{
	if (reqStream->available())
	{
		executorReceive(reqStream->read());
	}
	schedule_function(std::bind(&CommandExecutor::getStreamChars,this,reqStream));
}
```

HugoML, 2018-12-19:
Arduino 1.8.7 and ESP for Arduino 2.4.0 : 
serialEvent() still does not trigger for ESP. 
Not sure how to use HardwareSerial ...

d-a-v, 2018-12-19:
> Not sure how to use HardwareSerial ...

Until we have a real `serialEvent();`, you can do this in `loop()` (as stated above):
```
  if (Serial.available()) {
    serialEvent();
  }
```

ttkdroid, 2020-08-05:
Hi guys, almost 5 years have passed since this issue was reported and looks like it is still not solved. Is there any perspective that this would be resolved any time soon? Don't take me wrong...I understand that there are a lot of voluntary work and people need to accommodate this work with their schedules, but would be nice to have an update on this issue.

earlephilhower, 2020-08-05:
@ttkdroid please report in the PR if it works/fails.  It seems pretty trivial and not all that useful, honestly.

PR #7505 (2020-08-06): Add SerialEvent() callback to loop processing
Match the AVR SerialEvent implicit callback.  Callback is executed
in normal user mode, not IRQ, so standard processing can be uses.

Fixes #752 after 5 years. :)
commit 3e567e94 (2020-08-06): Add SerialEvent() callback to loop processing (#7505)

* Add SerialEvent() callback to loop processing

Match the AVR SerialEvent implicit callback.  Callback is executed
in normal user mode, not IRQ, so standard processing can be uses.

Fixes #752 after 5 years. :)

* Fix style
closed 2020-08-06 by devyte