esp8266/Arduino #755: Sending requests does cause LmacRxBlk error

bardiir, 2015-09-07:
I've got a webserver which is running quite nicely on an ESP. Now I do want to signal a second ESP with that.

So I added this and call it when a request came in to the webserver:

```
void setClientColor(char ID, char R, char G, char B, char W)
{
  char message[13] = "color|0RGBW|";

  message[6] = ID;
  message[7] = R;
  message[8] = G;
  message[9] = B;
  message[10] = W;
  
  udp.beginPacket(client_ip, UDP_PORT);
  udp.write(message, 13);
  udp.endPacket();
  delay(3);
  udp.flush();
}
```

Which does work, but now the ESP will fail after a few requests with the LmacRxBlk error.

sdrsdr, 2015-09-07:
Try without delay or with bigger delay :) Can you control the request rate to the webserver? throttle them down a bit (1 pers second) and see if  this helps. BTW UDP was never designed to be reliable so you must be able to accept some send or receive mishaps  :)

bardiir, 2015-09-07:
Yes I could of course live with the requests from the UDP not coming through, but they seem to get stuck somewhere. On the other hand I did try this with a TCP request too and it got stuck pretty bad too. Receiving requests itself doesn't pose any issue for me, but sending seems to be an issue, sort of.

The requests to the web server are infrequent (browser requests). But even a few single calls over a time will eventually lead to the LmacRxBlk error, even with far less than 1 request per second.

The webserver itself is quite stable anyway, the only issue the server is facing is because of the connection time_wait timeout which causes heap to go down with many requests coming in in short succession.

The delay there has been any value between 0 and 1000 already, or at least sort of any value. 3 seems to work best. No delay will prevent the UDP message from beeing sent sometimes. Having a bigger delay causes the webserver to stall for that time which is bad for reaction time while doing nothing good towards the LmacRxBlk issue.

sdrsdr, 2015-09-07:
Seems like a driver issue try reverting back to 9f2f2f4414b9cf583dd57c41360e74a53fb4bdde before the experimental TLS support was added

bardiir, 2015-09-07:
I'm using the stable version through the board manager. So I should be way before that :)

bardiir, 2015-09-07:
I'll check if I can create a minimal (non-)working solution based on my sketch to track the issue. Probably this evening or so. It seems like I'm not doing anything plainly wrong with the udp connect so it might be a sideeffect from something else ;)

bardiir, 2015-09-07:
Ok this can be closed I suppose, there still seems to be an issue when having an SoftAP when sending UDP packets to a client which is unresponsive (I'd expect UDP packets getting "lost" then) but that is seldomly the case with my setup.

The main issue was a typo causing the UDP packet with which the client registers it's ip on the server to be repeated quite often instead of only every 20 seconds or so, basically spamming the server until death.

closed 2015-09-07 by bardiir