esp8266/Arduino #756: Stack overflow

gordonendersby, 2015-09-07:
Hi chaps,

Ive just updated my arduino ide support for the ESP8266 to the latest stable version.

But a previously working program using the sleep timers is now playing up.
Im using an ESP8266-12 on a breadboard exactly as I did when it was working under the previous Stable version.
Usually I use the board definition for the Nodemcu for the esp8266-12.
I have also tried the generic definition but it still fails.
I notice that the board definitions now mentions spiffs does this reduce the memory available?

Im getting what seems to be a stack overflow after it connects to the wifi router.
This will happen after a couple of sleep and restarts
The ESP then reboots and can no longer connect to the wifi router.
I get this in the serial monitor:
```ctx: cont 
sp: 3ffea830 end: 3ffeaa90 offset: 01b0

>>>stack>>>
3ffea9e0:  40208d19 3ffe9a44 3ffeaad0 42020000  
3ffea9f0:  41ae6666 3ffe9778 3ffe9a44 3ffe9728  
3ffeaa00:  3ffe9728 00000000 00000000 3ffe9728  
3ffeaa10:  3ffe9728 4020234c 3ffe9a44 3ffe9778  
3ffeaa20:  42020000 3ffe9778 3ffe9a44 40202347  
3ffeaa30:  00000000 00000000 00000000 00000000  
3ffeaa40:  00000000 00000000 00000000 00000000  
3ffeaa50:  00000000 00000000 00000000 00000000  
3ffeaa60:  00000000 00000000 00000000 3ffeaabc  
3ffeaa70:  3fffdc20 00000000 3ffeaab4 4020183f  
3ffeaa80:  00000000 00000000 3ffe9a70 40100378  
<<<stack<<<
i�߈�
JZ�!!)���Zn�����DnZ��*��Xj(ȽC����H   �ܿHn:���O��d```

Ive added a count into the loop waiting for the connection so that the ESP restarts after 20 iterations of the loop. As previously it would have connected by then. Hoping that a restart would clear whatever is causing the problem.
But it hasnt helped.

Heres my code:
''' 
#include <DHT.h>
#include <ESP8266WiFi.h>
extern "C" {
  #include "user_interface.h"
  uint16 readvdd33(void);
}
 
// replace with your channel's thingspeak API key, 
String apiKey = "N33F4IRBI8EI8K8J";
const char* ssid = "ThisOldHouse";
const char* password = "thingstodoinalandrover";
 
const char* server = "api.thingspeak.com";
#define DHTPIN 2 // what pin we're connected to

//const unsigned long SLEEP_INFTERVAL = 10 * 60 * 1000 * 1000; // 10 minutes
const unsigned long SLEEP_INFTERVAL = 30 * 1000 * 1000; // 30 seconds
 
DHT dht(DHTPIN, DHT22,15);
WiFiClient client;
   
 
void setup() {                
  Serial.begin(57600);
  delay(10);
  dht.begin();
  
  WiFi.begin(ssid, password);
 
  Serial.println();
  Serial.println();
  Serial.print("Connecting to ");
  Serial.println(ssid);
   
  WiFi.begin(ssid, password);
  int c = 0; 
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
    c++;
    if (c > 20){ESP.restart();}
  }
  Serial.println("");
  Serial.println("WiFi connected");

  // ################################ get and send reading

   
  float h = dht.readHumidity();
  float t = dht.readTemperature();
  float vdd = readvdd33() / 1000.0;
  if (isnan(h) || isnan(t)) {
    Serial.println("Failed to read from DHT sensor!");
    return;
  }

   Serial.println(); 
   Serial.print("Temperature: ");
   Serial.print(t);
   Serial.print("c  - Humidity: "); 
   Serial.print(h);
   Serial.print("%  - Voltage: "); 
   Serial.print(vdd);
   Serial.println("v "); 

   Serial.println(); 
   Serial.println("Send to Thingspeak");    
   Serial.println();

  while (!client.connect(server, 80)) {
    Serial.println("connection failed, retrying...");
  }

  String postStr = apiKey;
         postStr +="&field1=";
         postStr += String(t);
         postStr +="&field2=";
         postStr += String(h);
         postStr +="&field3=";
         postStr += String(vdd);
         
         postStr += "\r\n\r\n";

   client.print("POST /update HTTP/1.1\n"); 
   client.print("Host: api.thingspeak.com\n"); 
   client.print("Connection: close\n"); 
   client.print("X-THINGSPEAKAPIKEY: "+apiKey+"\n"); 
   client.print("Content-Type: application/x-www-form-urlencoded\n"); 
   client.print("Content-Length: "); 
   client.print(postStr.length()); 
   client.print("\n\n"); 
   client.print(postStr);
         
 

   // wait for response
    while(!!!client.available()) {
      delay(10);
    }
  
    // Read all the lines of the reply from server and print them to Serial
    Serial.println("Response: "); 
    while(client.available()){
      String line = client.readStringUntil('\r');
      Serial.print(line);
    }

    Serial.println();
    Serial.println("closing connection");
  
  client.stop();

  Serial.print("up time: ");
  Serial.print(millis());

  Serial.print(", deep sleep for ");
  Serial.print(SLEEP_INFTERVAL / 1000000);
  Serial.println(" secs...");
  system_deep_sleep_set_option(0);
  system_deep_sleep(SLEEP_INFTERVAL - micros()); 
  
}
 
 
void loop() {

}'''

Sorry about the formatting in the message.
I couldn work out how to wrap the code
Thanks

Gordon

igrr, 2015-09-07:
This looks like software watchdog reset. Could you please check if removing
the call to readvdd33 function resolves the issue?

gordonendersby, 2015-09-07:
Ive also posted this on ESP8266.COM.
The code is more readable there.
http://www.esp8266.com/viewtopic.php?f=32&t=4894

Igrr,  thanks, Ill check that in the next hour or so.
Its confusing as it was fine before i got the latest stable version of the board support.
Is there a way to revert to an earlier version to double check it was ok?

Gordon

gordonendersby, 2015-09-07:
Yep, removed the calls to  readvdd33 and it works a charm.
Sleeps, wakes, connects and uploads the data without any problems.

Is this a bug in the arduino support or a problem with how Im using it?
Why does the watchdog reset kick in with readvdd33?
Is there any way I can get this working and use readvdd33?

Thanks for looking at this.
Gordon

chaeplin, 2015-09-07:
use ADC_MODE(ADC_VCC), ESP.getVcc(). check ESP-specific APIs at 
https://github.com/esp8266/Arduino/blob/esp8266/hardware/esp8266com/esp8266/doc/reference.md

gordonendersby, 2015-09-07:
That stops the stack overflow.
Im getting the occasional hang connecting to thingspeak but thats just my code.
I hacked this together quickly from examples so ill rewrite it properly with tests on the connection and test further.

Whats the difference between readvdd33 and ESP.getVcc()?
Is the readvdd33 an older method to get the voltage and has been replaced by ESP.getVcc() because of these problems and implemented in a newer version of the stable version?

Thanks

Gordon

igrr, 2015-09-07:
Great to know that everything is fine now!

Just for the record, this wasn't a stack overflow :)
When you get something like
```
ctx: cont 
sp: 3ffea830 end: 3ffeaa90 offset: 01b0
```
without `Exception: ...` lines above it, this means that software watchdog timer has been triggered.
Stack overflows usually cause some `Exception: ...` lines to be generated.

Now regarding `readvdd33`...
`readvdd33` is a function which is internal to the SDK, it never was part of public API neither in Espressif SDK nor in Arduino package. Why it happened to work in the previous version but not in the current one, i have no idea. Espressif SDK provides a public function, `system_get_vdd33`, which calls `readvdd33` internally but also ensures that WiFi hardware is an a proper state before this call. In arduino environment we have this `ESP.getVcc()` function which basically wraps `system_get_vdd33` call. You also need to add ADC_MODE(ADC_VCC) to your sketch as @chaeplin pointed out (this is also mentioned in the documentation).

The bottom line is, don't use `readvdd33` in your sketches.

gordonendersby, 2015-09-07:
Thanks for the explanation.
Ill write up later about the difference and put it in the post on esp3866.com
Hopefully it will help others.

There are a few examples of reading the input voltage around that use readvdd33.
I did a google search but found very little that said it shouldnt be used and we should be using ESP.getVcc(). Maybe searches in the future will find this post and your explanation.
Its a pity theres not a warning in the ide at compile time to warn users.

Thanks again.
Gordon

closed 2015-09-07 by gordonendersby

Charlyo, 2016-04-22:
I have the same issue... Any thoughts?

#include <SoftwareSerial.h>
#include <MFRC522.h>
#include <ESP8266WiFi.h>
#include <SPI.h>
#include <PubSubClient.h>

/* wiring the MFRC522 to ESP8266 (ESP-12)
  RST     = GPIO16
  SDA(SS) = GPIO4
  MOSI    = GPIO13
  MISO    = GPIO12
  SCK     = GPIO14
  GND     = GND
  3.3V    = 3.3V
*/

#define RST_PIN  16 // RST-PIN für RC522 - RFID - SPI - Modul GPIO15 
#define SS_PIN  15  // SDA-PIN für RC522 - RFID - SPI - Modul GPIO2 

const char *ssid =  "thethings.iO - HQ";     // change according to your Network - cannot be longer than 32 characters!
const char *pass =  "ilovethethingsioTTIO007"; // change according to your Network

MFRC522 mfrc522(SS_PIN, RST_PIN); // Create MFRC522 instance

SoftwareSerial swSer(D1, D2, false, 256);

#define TOKEN "TtoajojQcU1KXGNQti1hTMolsrC519tf32HBD7bZf1s"

WiFiClient wifiClient;
String UID;

void callback(char* topic, byte* payload, unsigned int length) {
  // handle message arrived
  String text = "";
  for (int i = 0; i < length; i++)
    text.concat((char)payload[i]);
  Serial.println(text);
}
PubSubClient client("mqtt.thethings.io", 1883, callback, wifiClient);

String topic = "v2/things/" + String(TOKEN);
String s, resp, message;

// Helper routine to dump a byte array as hex values to Serial
void dump_byte_array(byte *buffer, byte bufferSize) {
  for (byte i = 0; i < bufferSize; i++) {
    Serial.print(buffer[i] < 0x10 ? " 0" : " ");
    Serial.print(buffer[i], HEX);
  }
}

String byteToString(byte *buffer, byte bufferSize) {
  String s = "";
  for (byte i = 0; i < bufferSize; i++) {
    if (buffer[i] < 0x10) {
      s = s + "0";
    }
    s = s + String(buffer[i], HEX);
  }
  return s;
}

void reconnect() {
  // Loop until we're reconnected
  while (!client.connected()) {
    Serial.print("Attempting MQTT connection...");
    // Attempt to connect
    if (client.connect("ESP12Client")) {
      Serial.println("connected");
      // Once connected, publish an announcement...
      // ... and resubscribe
      client.subscribe((char*)topic.c_str());
    } else {
      Serial.print("failed, rc=");
      Serial.print(client.state());
      Serial.println(" try again in 5 seconds");
      // Wait 5 seconds before retrying
      delay(5000);
    }
  }
}

void setup() {
  Serial.begin(9600);    // Initialize serial communications
  swSer.begin(9600);
  delay(250);
  Serial.println(F(""));
  Serial.println(F("Booting...."));
  SPI.begin();           // Init SPI bus
  
  WiFi.begin(ssid, pass);

  int retries = 0;
  while ((WiFi.status() != WL_CONNECTED)) {
    delay(500);
    Serial.print(".");
  }
  if (WiFi.status() == WL_CONNECTED) {
    Serial.println(F("WiFi connected"));
  }

  while (!client.connect("ESP12Client")) {

  }
  Serial.println(F("Client connected to mqtt broker!"));
  client.subscribe((char*)topic.c_str());
  
  mfrc522.PCD_Init();    // Init MFRC522
  Serial.println(F("Ready!"));
  Serial.println(F("======================================================"));
  Serial.println(F("Scan for Card and print UID:"));
  message = "{\"values\":[{\"key\":\"uid\",\"value\":";
}

void loop() {
  yield();
  client.loop();
  yield();
  
   while (swSer.available() > 0) {
    Serial.write(swSer.read());
  }
  while (Serial.available() > 0) {
    swSer.write("AT\r\n");
  }
  if (!client.connected()) {
    reconnect();
  }
  // Look for new cards
  if (!mfrc522.PICC_IsNewCardPresent()) {
    delay(50);
    return;
  }
  Serial.println("UID FOUND");
  // Select one of the cards
  if (!mfrc522.PICC_ReadCardSerial()) {
    delay(50);
    return;
  }
  // Show some details of the PICC (that is: the tag/card)
  UID = byteToString(mfrc522.uid.uidByte, mfrc522.uid.size);
  String toSend = message + UID + "}]}";
  yield();
  if (client.publish((char*)topic.c_str(), (char*) toSend.c_str())) {
    Serial.println("Publish ok");
  }
  else {
    Serial.println("Publish failed");
  }
  yield();
  delay(1000);
}