esp8266/Arduino #760: MDNS-SD is broken?

syadykin, 2015-09-07:
Hello,

Today I updated code from git and rebuilt my project which uses mdns service discovery. Surprisingly it stops working for me — simple sketch with addService call do not resolve service with command ```avahi-browse -r _http._tcp -p``` anymore, while name resolution works fine. Is mdns-sd broken again?

syadykin, 2015-09-08:
My project built with less recent version of studio works perfectly:

    me@supersonic:~$ avahi-browse -r _http._tcp -p -t
    +;wlan0;IPv4;switch;_http._tcp;local
    =;wlan0;IPv4;switch;_http._tcp;local;switch.local;192.168.78.206;3333;"board=generic"
    me@supersonic:~$

igrr, 2015-09-08:
I don't have access to a Linux box at the moment to try with avahi, but DNS-SD seems to work fine with OS X and Bonjour. I have opened mDNS_Web_Server sample, changed SSID and password, and flashed the sketch to NodeMCU 1.0 board.

From console, i get this output:
```bash
[igrokhotkov@igrokhot-mbp ~/e/spiffs]$ dns-sd -B _http._tcp
Browsing for _http._tcp
DATE: ---Wed 09 Sep 2015---
 1:19:34.676  ...STARTING...
Timestamp     A/R    Flags  if Domain               Service Type         Instance Name
 1:19:35.052  Add        2   4 local.               _http._tcp.          esp8266
^C
[igrokhotkov@igrokhot-mbp ~/e]$ dns-sd -L esp8266 _http._tcp
Lookup esp8266._http._tcp.local
DATE: ---Wed 09 Sep 2015---
 1:20:16.569  ...STARTING...
 1:20:16.570  esp8266._http._tcp.local. can be reached at esp8266.local.:80 (interface 4)
 board=generic
```
Perhaps you can run tcpdump and filter for mDNS packets?

syadykin, 2015-09-09:
Magically it started to work again. Sorry.

closed 2015-09-09 by syadykin