esp8266/Arduino #761: Websocket

Accesse, 2015-09-07:
Please add websocket. Found a git and changed the demo a little bit.

https://github.com/Accesse/Arduino/tree/esp8266/hardware/esp8266com/esp8266/libraries/ESP8266Websocket

Thx

Links2004, 2015-09-08:
is the src based of this:
https://github.com/ejeklint/ArduinoWebsocketServer ?

we have a list of libarys for the ESP here:
https://github.com/esp8266/Arduino/blob/esp8266/hardware/esp8266com/esp8266/doc/reference.md#other-libraries-not-included-with-the-ide

and a working websocket Server + client without the limitations here:
https://github.com/Links2004/arduinoWebSockets

have you improved the libary?

Accesse, 2015-09-08:
I got it from: https://github.com/morrissinger/ESP8266-Websocket
and as far as I know it is based on the source you named.
Haven't improved the one of morrissinger but changed the demo, so that it works.

Thx for the links didn't know about those libraries. Why don't you include your websocket to the ide? I would like that and suppose others too.

Cheers

Links2004, 2015-09-09:
to keep the but tracking and version management on a external place.
@igrr is it possible to modify the zip building to include external git / library?
or is it possible to use the  git-submodule?

igrr, 2015-09-09:
Yes you can add external libraries while building boards manager package.
https://github.com/esp8266/Arduino/blob/esp8266/build/build_board_manager_package.sh#L16-L18
You can use wget to download certain release of an external library.

git-submodule will also work, although i would prefer doing it with wget.

Accesse, 2015-09-10:
@Links2004 wanted to thank you! I checked out your websocket library and really enjoy it, thx! Love the broadcasting! Cheers

Closing the topic here

closed 2015-09-10 by Accesse