esp8266/Arduino #763: TMRpcm has reported some issue that you might want to look at:

fredericplante, 2015-09-08:
https://github.com/TMRh20/TMRpcm/wiki

When I compile the basic example cc1plus.exe get stock so I end up closing it, by force ant his error message came up.:

"

basic.ino: In function 'void setup()':
basic.ino:19:22: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
basic.ino: In function 'void loop()':
basic.ino:28:26: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
In file included from C:\Documents and Settings\Cath.CHAMBRE\My Documents\Arduino_ESP\libraries\TMRpcm/pcmConfig.h:20:0,
                 from C:\Documents and Settings\Cath.CHAMBRE\My Documents\Arduino_ESP\libraries\TMRpcm\TMRpcm.cpp:4:
C:\Documents and Settings\Cath.CHAMBRE\My Documents\Arduino_ESP\libraries\TMRpcm\TMRpcm.cpp:19:27: error: 'ICIE1' was not declared in this scope
  const byte togByte = _BV(ICIE1); //Get the value for toggling the buffer interrupt on/off
                           ^
C:\Documents and Settings\Cath.CHAMBRE\Application Data\Arduino15\packages\esp8266\hardware\esp8266\1.6.5-1084-ga39ce29\cores\esp8266/Arduino.h:184:25: note: in definition of macro '_BV'
 #define _BV(b) (1UL << (b))
                         ^
C:\Documents and Settings\Cath.CHAMBRE\My Documents\Arduino_ESP\libraries\TMRpcm\TMRpcm.cpp:102:30: error: 'TIMSK1' was not declared in this scope
   volatile byte *TIMSK[] = {&TIMSK1};
                              ^
C:\Documents and Settings\Cath.CHAMBRE\My Documents\Arduino_ESP\libraries\TMRpcm\TMRpcm.cpp:103:31: error: 'TCCR1A' was not declared in this scope
   volatile byte *TCCRnA[] = {&TCCR1A};
                               ^
C:\Documents and Settings\Cath.CHAMBRE\My Documents\Arduino_ESP\libraries\TMRpcm\TMRpcm.cpp:104:31: error: 'TCCR1B' was not declared in this scope
   volatile byte *TCCRnB[] = {&TCCR1B};
                               ^
C:\Documents and Settings\Cath.CHAMBRE\My Documents\Arduino_ESP\libraries\TMRpcm\TMRpcm.cpp:105:38: error: 'OCR1A' was not declared in this scope
   volatile unsigned int *OCRnA[] = {&OCR1A};
                                      ^
C:\Documents and Settings\Cath.CHAMBRE\My Documents\Arduino_ESP\libraries\TMRpcm\TMRpcm.cpp:106:37: error: 'ICR1' was not declared in this scope
   volatile unsigned int *ICRn[] = {&ICR1};
                                     ^
C:\Documents and Settings\Cath.CHAMBRE\My Documents\Arduino_ESP\libraries\TMRpcm\TMRpcm.cpp:107:37: error: 'TCNT1' was not declared in this scope
   volatile unsigned int *TCNT[] = {&TCNT1};
                                     ^
C:\Documents and Settings\Cath.CHAMBRE\My Documents\Arduino_ESP\libraries\TMRpcm\TMRpcm.cpp:110:39: error: 'OCR1B' was not declared in this scope
    volatile unsigned int *OCRnB[] = {&OCR1B};
                                       ^
In file included from C:\Documents and Settings\Cath.CHAMBRE\My Documents\Arduino_ESP\libraries\TMRpcm/pcmConfig.h:20:0,
                 from C:\Documents and Settings\Cath.CHAMBRE\My Documents\Arduino_ESP\libraries\TMRpcm\TMRpcm.cpp:4:
C:\Documents and Settings\Cath.CHAMBRE\My Documents\Arduino_ESP\libraries\TMRpcm\TMRpcm.cpp: In member function 'void TMRpcm::timerSt()':
C:\Documents and Settings\Cath.CHAMBRE\My Documents\Arduino_ESP\libraries\TMRpcm\TMRpcm.cpp:205:21: error: 'WGM11' was not declared in this scope
   *TCCRnA[tt] = _BV(WGM11) | _BV(COM1A1) | _BV(COM1B0) | _BV(COM1B1); //WGM11,12,13 all set to 1 = fast PWM/w ICR TOP
                     ^
C:\Documents and Settings\Cath.CHAMBRE\Application Data\Arduino15\packages\esp8266\hardware\esp8266\1.6.5-1084-ga39ce29\cores\esp8266/Arduino.h:184:25: note: in definition of macro '_BV'
 #define _BV(b) (1UL << (b))
                         ^
C:\Documents and Settings\Cath.CHAMBRE\My Documents\Arduino_ESP\libraries\TMRpcm\TMRpcm.cpp:205:34: error: 'COM1A1' was not declared in this scope
   *TCCRnA[tt] = _BV(WGM11) | _BV(COM1A1) | _BV(COM1B0) | _BV(COM1B1); //WGM11,12,13 all set to 1 = fast PWM/w ICR TOP
                                  ^
C:\Documents and Settings\Cath.CHAMBRE\Application Data\Arduino15\packages\esp8266\hardware\esp8266\1.6.5-1084-ga39ce29\cores\esp8266/Arduino.h:184:25: note: in definition of macro '_BV'
 #define _BV(b) (1UL << (b))
                         ^
Error compiling.

"

Links2004, 2015-09-08:
the missing stuff are AVR specific register, so can not work an ESP.
to be exact its the Timer of the AVR (ATmega), the library needs to be rewrite to work on the ESP.

devyte, 2017-10-20:
3rd party lib written for AVR. I suggest asking the lib author for help with porting, or find an alternative.
Closing.

closed 2017-10-20 by devyte