esp8266/Arduino #764: API to set RF TX Power Level

Humancell, 2015-09-09:
I have looked around and can not find an API to set the Wifi RF TX Power Level.  According to this post, the SDK esp_iot_sdk_v1.1.0_15_05_22 added this capability.

Is this already exposed someplace, or can this API be exposed?  I have not been able to locate it in any documentation here.

http://www.esp8266.com/viewtopic.php?f=13&t=3072&sid=39bf5cb2ff5f17bbc6faf06503064cc3&start=5#p18209

avivgr, 2015-12-05:
Any update? i am looking for the same thing. Also wondering what is the default max power in case the set api is not explicitly called?

chaeplin, 2016-01-28:
use WiFI.setOutputPower
https://github.com/esp8266/Arduino/blob/41074245a8d3029270622a8b489d53fdfedb2a22/libraries/ESP8266WiFi/src/ESP8266WiFiGeneric.cpp#L169-L179

babylonx, 2016-07-10:
What happens if you never call setOutputPower? What is the default transmission power set?

Stefano-Orsolini, 2016-08-09:
@babylonx 
from measurements I can tell that it sets to maximum TX power.
I see 19dBm difference between default and using setOutputPower(0)

devyte, 2017-10-20:
The NONOS SDK api reference for SDK v2.1.0 explains that the function takes values in the range [0-82], which means the argument for our wrapper is in the range [0-20.5].
Closing.

closed 2017-10-20 by devyte