esp8266/Arduino #768: wifi_station_set_hostname()

mtnbrit, 2015-09-10:
Hi, when I check my DHCP server, the esp8266's don't have hostnames, but when I look in my Ubiquiti UniFi controller, I see ESP_xxyyzz hostnames made up from the last 3 of the mac address. Wondering how this name was set, I found the sdk has this function wifi_station_set_hostname(). Is this function exposed in Arduino/esp8266? Or some other way to set this hostname outside of dhcp or mDNS?

Humancell, 2015-09-10:
The first three octets of a MAC Address are the registered vendor ID.  My guess is that your Ubiquiti UniFi is using this information to look up and display an abbreviation of the vendor name?

pgollor, 2015-09-11:
Hi,
You can call the function directly in yout `setup()` function. Please see the [CallSDKFunctions](https://github.com/esp8266/Arduino/blob/esp8266/hardware/esp8266com/esp8266/libraries/esp8266/examples/CallSDKFunctions/CallSDKFunctions.ino) example sketch for more details.

```
wifi_station_set_hostname("foobar");
```
This is working for me. :-)
Please close your issue if it works.

Links2004, 2015-09-11:
here is an APIfor this in the WiFi class

https://github.com/esp8266/Arduino/blob/esp8266/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/ESP8266WiFi.h#L310-L321

mtnbrit, 2015-09-11:
When I include the espressif sdk, I get several compile errors

```error: 'size_t' was not declared in this scope```

I tried including ```#include <stdlib.h>``` but it didn't help.

mtnbrit, 2015-09-11:
never mind, i fixed the compile error by putting stdlib.h include before the sdk include...

mtnbrit, 2015-09-11:
```wifi_station_set_hostname("foobar")``` works a treat, thanks!

closed 2015-09-11 by mtnbrit