esp8266/Arduino #769: how does .serveStatic(...) works?

Yoshi20, 2015-09-11:
I saved index.html in the FS ("/data/index.html") and tryed the following functioncall:

server.serveStatic("/", SPIFFS, "/data/index.html");

but it doesn't seems to work.
Is this function still a work in progress or do I do something wrong?

pgollor, 2015-09-11:
```
void ESP8266WebServer::serveStatic(const char* uri, FS& fs, const char* path) {
    _addRequestHandler(new StaticRequestHandler(fs, uri));
}
```

This code snippet is out `ESP8266WebServer.cpp`. The function seems to be not finished yet.

igrr, 2015-09-11:
Well spotted, looks like StaticRequestHandler doesn't take FS "path" into account. Will fix.

Yoshi20, 2015-09-11:
nice :+1: 
I'm looking forward to it.

Yoshi20, 2015-09-14:
The Fix: "ESP8266WebServer::serveStatic to work for both files and directories"
Seems to work perfectly  -> Close issue

closed 2015-09-14 by Yoshi20

mbenitez01, 2018-08-25:
was this (server.serveStatic) ever fixed?!?!? I have NEVER been able to serve a miserable file with this method... 8-(