esp8266/Arduino #770: warning: espcomm_sync failed error: espcomm_open failed // ESP-01 // Wiring up the right way shuts my ESP off

ildr3him, 2015-09-11:
Using 1.6.5 Arduino IDE with stable esp8266 distribution on mac osx. In the beginning (and still to this point) I always get the error:

warning: espcomm_sync failed
error: espcomm_open failed

when trying to upload any sketch.

I feel like i tried the most things i could find on google regarding this error.

But while trying to figure this out I got somewhat confusing information. Some sources say GPI00 needs to be connected to GND, some say it needs to be a PullUp. Some sources say ch_pd needs to be at 3.3v, some say it needs to be at DTR with a PullUp Resistor and some sources leave it completely disconnected and dont mention it anywhere. Can somebody who got a ESP-01 to work give me some clarification on this?

I'm really confused but I noticed some things that seem odd and maybe you can help me figure this out:

- RX and TX seem to be wired the right way. VCC (3.3v) and GND too. When I start uploading a Sketch the blue led blinks, even though i get the error at the end
- Connecting GPIO0 to GND doesn't resolve the issue like in the [Minimal Hardware Setup for Bootloading only example on this page](https://github.com/esp8266/Arduino/blob/esp8266/hardware/esp8266com/esp8266/doc/boards.md#generic-esp8266-modules) // the ESP-01 has no GPIO15 I can connect anything too btw.
- Rather odd: Connecting the GPIO0 to DTR and a PullUp like in [this example from the same page] (https://raw.githubusercontent.com/esp8266/Arduino/esp8266/docs/ESP_to_serial.png) shuts down the ESP-01. The Red LED is off and the blue LED isn't blinking either when uploading anything to it. So because I think its rather unlikely that the example is wrong I guess I missunderstand something about the wirring of the ESP-01 fundamentaly wrong, so wrong that I can't figure my head around why this is happening.

Can you help me in any way clarify my confusion?

pgollor, 2015-09-11:
Do you have pull ups for CH_PD and REST?
Did you connect the RTS out from your UART USB converter chip to REST or CH_PD?

If no RTS out available you have to pull down the REST or the CH_PD to ground manually just before you want to upload the sketch.
Flashing is only available if GPIO0 and REST (or CH_PD works too) are pulled to ground just before uploading the image.

Pleas connect all pins from your ESP-01 module except GPIO2 like the [example](https://raw.githubusercontent.com/esp8266/Arduino/esp8266/docs/ESP_to_serial.png).

Links2004, 2015-09-11:
the example is working I have six ESP-12 working with this. 
when GPIO0 is low (0V) the ESP goes to boot mode on next reset 
if it is high (3.3V) the ESP will boot your sketch.

don't connect the GPIOs directly to VCC or GND always use a resistor for all between 2k to 20k  will work, most used for this is 10k. 

for the ESP-01 the GPIO15 is connected correct on the board itself.
(its the pin left to GPIO2 on the SOC)
[image]

ildr3him, 2015-09-11:
@pgollor In the example the CH_PD and GPIO0 are pulled up, aren't they? 

But you write GPIO0 (or CH_PD) need to be connected to ground? In the example there is no GND connection to the GPIO0

Do I overlook something here?

pgollor, 2015-09-11:
For normal mode GPIO0 have to pulled up. This is the reason for the pull up resistor.
To go into the bootloader after reboot the GPIO0 have once pulled down...

Please read the available documentation and close this issue. Your issue is not a software problem.

drmpf, 2015-09-11:
Check out http://www.forward.com.au/pfod/CheapWifiShield/ESP2866_01_WiFi_Shield/index.html
for a complete circuit and programming instructions and programming serial cable,

closed 2015-09-30 by igrr

jrperin, 2015-10-20:
Hi guys! I was having the same problem: 
warning: espcomm_sync failed
error: espcomm_open failed

For me, that was solved with an external 3.3v. My FT232 USB hasn't enough power to provide to ESP8266.

osharim, 2015-10-21:
I found a solution.

this is why using Arduino 1.6.5 doesn't seems to work with esptool.

the solution can be found here http://hpclab.blogspot.mx/2015/

igrr, 2015-10-21:
This is because esptool-ck doesn't work with PL2303 based converters on OS X.

Seyig, 2015-10-27:
@igrr , please is there a way to fix this issue?

tanmayee30, 2015-12-24:
Hello!!
Even I am having same problem.
"warning: espcomm_sync failed
error: espcomm_open failed"
I have installed all the required libraries required for ESp8266. I am working on windows OS also using arduino_IDE version 1.6.7

bluman74, 2016-02-28:
Make sure you disconnect everything from GPIO0 and only leave the GND connection in place.

kolya182, 2016-03-03:
Had the same issue. Added 10k resistor between GPIO0 and GND. Arduino 1.6.5 mac El Capitan.

rengelking, 2016-03-17:
Add more power.  When the Wifi powers up it crashes everything if you don't have enough power; you can add +5V power to the Vbat along with whatever power is coming from the FTDI cable you are trying to program it with.

tscha70, 2016-04-30:
I spent hours to find the ISSUE!!!

REMOVE LED on GPIO2 and it works!!!!

or better, DO NOT USE low cost LED as I did!!

abbiy, 2016-05-15:
I am also facing this error and have tried all the above listed troubleshoots but still nothing happens . I am felling so low as i this is my first IOT project and i am failed. PLzz someone help me out with this ..

**warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed**

romanakozak, 2016-06-05:
I am having this exact same issue on Mac and Arduino 1.6.9, very frustrating. Please help!

freedolin, 2016-06-08:
I was having the same issue. 

Solution: 
Select the correct COM-Port in the Arduino IDE under Tools -> Port:

ragd011, 2016-06-09:
Same error here on a windows 10 machine.  Worked yesterday before upgrading to 1.6.9.  I think I was running 1.6.6 and realized there was a new version so I upgraded and I can't load the same board I built and actually updated this morning on 1.6.6.  Rebooted, tried different settings, tried different sketches.

djpavlovic, 2016-06-17:
I have solved the issue with Ubuntu running Arduino IDE 1.6.8 as sudo user eg. sudo ./arduino

hjpatel91, 2016-06-28:
i am also facing same problem. I am using Arduino 1.6.9. i have connected GPIO0 and ch_PD  to 3.3v and GPIO15 to GND. Plz help me with this.
 

warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed

[image]

krzychb, 2016-06-28:
Hi @hjpatel91,

I am just working on [troubleshooting guide](https://github.com/krzychb/Arduino/blob/master/doc/faq/a01-espcomm_sync-failed.md) for this issue.
Could you follow it step by step and provide result of each step?

Krzysztof

kylemusco, 2016-07-19:
I fixed this problem by removing the LED on GPIO_2, using an external power supply for the ESP and power cycling before every upload. Also, GPIO_0 is set to GND

amanpreetjpr, 2016-08-27:
Hi,
I was having the same error when I tried uploading the code:
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed

My problem got solved when I applied external 3.3V supply (not from the USB port) and changed the baud rate to 57600bps
Aman

zurexalexis, 2016-09-05:
Problem - using arduino IDE 1.6.11

warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed

My ESP - ESP12F

Connection

VCC ----- VCC (3.3 Volt using 12 volt adaptor with voltage regulator LM 2596, to 3.3 volt)
CH_PD (i don't havr CH_PD on ESP12F but i have EN instead) ----- VCC (3.3 Volt)

GND ----- GND (any ground, the adaptor or FTDI232 GND)
GPIO15 ----- GND
GPIO0 ----- GND

RX (on esp) ----- TX (FTDI232)
TX (on ESP) ----- RX (FTDI232)

What solving my problem is :
1. i unpluged the VCC on ESP (power off)
2. i klik UPLOAD BUTTON (on arduino IDE)
3. watching the COMPILING SKETCH text.... (on arduino IDE) 
4. When the text is changing into UPLOADING (on arduino IDE), i plugged back my VCC (on ESP) to VCC 3.3V (power it on)
5. and its start uploading......
6. that's it

source of learning this technique is [HERE](https://learn.sparkfun.com/tutorials/esp8266-thing-hookup-guide/discuss)

my first post on git hub, sorry for my bad english, thanks

diimdeep, 2016-10-18:
Turns out my FT232R FTDI pinout was `GND, CTS, VCC, TXD, RXD, DTR` and I was usin `DTR, CTS` for reset, [that is wrong](https://github.com/esp8266/Arduino/blob/master/doc/boards.md#serial-adapter). I had to add `RTS` to my FTDI and use `DTR, RTS` 
[image]

harleydk, 2016-11-04:
@kylemusco Thanks for that, did the trick for me.

Krunal-Desai, 2016-11-14:
Hi guys just today after spending hours like crazy i finally solved the problem.

To use this library with Sketch IDE you need following things installed properly.

1.sketch IDE

2.This above Library 

3.Device driver which was missing in my case which you can find here https://www.silabs.com/products/mcu/Pages/USBtoUARTBridgeVCPDrivers.aspx

Feel free to ask or post any other error if you are getting.

shenery, 2016-11-19:
Hi Krunal-Desai. I am having trouble getting my sonoff in flash mode. I have tried everything. When I plug it in via ft232rl I see com port 4 but when I flash I get espcomm mem failed. Do you think if I replace the driver with the one you mentioned this will fix my problem? Did you see a com port in arduino ide before you installed the new driver?

Kind regards
Steven

Krunal-Desai, 2016-11-19:
Hi Steven. 

Well I do not have any idea about your board so no sure this will fix your problem for sure. Answer to your second question is no.

Before i install these driver I was able see only COM3 port in IDE but once i installed these drivers i was able to see COM4 and that was working nicely. COM3 even after installation did not work but COM4 worked out best.

So what i believe is this particular chip require a certain set of serial communication protocol set up on particular COM port and that is what these driver do. So no matter what board you use or what IDE you take it i believe this driver is just a way to carry the information to chip and they might solve your problem.

Feel free to share any further queries if you have and do post comment if things work out just fine so other can also get help from it.

Happy to help you...!
Krunal

Jairos2015, 2016-11-29:
Tenía el mismo problema y estaba trabajando con tarjeta generica. La cambie a esp-E y otra vez todo ok.
OK with esp-E.

kumarmaxwell, 2016-12-12:
Hi all,
By Connecting RESET,CH_PD pins of E01 module to 3.3V and GPIO0 to GND. These error can resolved..A brief post on Errors are [here](https://esp8266e01.wordpress.com/2016/12/05/346/)

BJSingh, 2017-02-03:
Hi,
I tried various possibilities mentioned above by others for same error. So I would like to share what works for me and really I don't why but it works.
Connections I followed:

[image]

I switch on power supply keeping switch position(connected to GPIO0) to F side for flashing code using arduino ide 1.6.11 and once done moves switch to N position. Now keeping switch position to N if I off and on power of esp8266 its runs uploaded program.
Now to upload new code I just have to switch off power of esp8266 and change position of switch to F side again. After this when ESP8266 given power then it goes into flashing and now I can upload new code.

duncan-a, 2017-02-03:
It works because it's correct...

Rather than have GPIO0 'floating'  when the switch is set to N, I would
pull it high via a resistor from the switch contact to +3.3V - but the
process is  correct.

BJSingh, 2017-02-03:
Ok and @duncan-a thanks for suggestion.

ritwik552, 2017-02-19:
I have same Problem  
am getting this error 

warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed
error: espcomm_upload_mem failed

what is that problem 
am using ESP8266 (2.2.0)

igrr, 2017-02-19:
@ritwik552 please check the FAQ first http://arduino.readthedocs.io/en/docs_to_readthedocs/faq/a01-espcomm_sync-failed.html

ritwik552, 2017-02-19:
i just make it by changing porn ty...but when am attacking some network at deauth all showing 0pkts/s ant nothing happen

igrr, 2017-02-19:
Which is good, if you'd ask me.

ritwik552, 2017-02-19:
means? i don't get it

chanuradha, 2017-02-28:
Hello,
         can anyone suggest me how to send a alert sms using esp8266. plzzzzzzzzz

mayurshilame, 2017-04-04:
Connecting CH_PD to Vcc solved my problem.....

Sephok, 2017-04-09:
TL,DR: I solved my problems (warning: espcomm_sync failed     etc.) by replacing the 10k resistor between GPIO00 and GND I used first with a 3.9k resistor.

Long Version:
I was also starting to feel a little irritated after sitting in front of this problem last night and today. But just about an hour ago I managed to solve my problem.
Initially my Setup is the ESP-12S on the typical white adapter board. VCC to 3.3V, GND to GND, CH_PD (or EN on other chips) and GPIO02 pulled high with 10k (although CH_PD should be high anyways with the layout of the adapter board, turning the two parallel 10k resistors effectively into a 5k), GPIO15 and GPIO00 pulled low with 10k (although GPIO15 should also be pulled low by the 10k on the board thus making the duo 5k as well). 0.1uf and 47uf are placed across VCC/GND. Power is supplied by the PSU from the Elegoo "The Most Complete Starter Kit". USB-to-Serial is done with one of these
http://www.ebay.de/itm/272461294802?_trksid=p2060353.m1438.l2649&ssPageName=STRK%3AMEBIDX%3AIT
, connecting the TX/RX/GND pins. For programming I use the Arduino IDE 1.6.12 with the esp-library.

The problem: I get the
"
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed
error: espcomm_upload_mem failed
"
messages. Turning up the verbosity during upload reveals these lines as well:
"
trying to connect
     flush start
     setting serial port timeouts to 1 ms
     setting serial port timeouts to 1000 ms
     flush complete
     espcomm_send_command: sending command header
     espcomm_send_command: sending command payload
     espcomm_send_command: receiving 2013 bytes of data
     read 0, requested 1
error: failed reading byte
warning: espcomm_send_command: cant receive slip payload data
"

It is the Serial Monitor that will give me an idea what might be the problem:
On 115200 Baud it shows me something like this:
"
*garbage_data* (This is actually the bootloader reporting stuff on 74880 Baud)
AI-Thinker *something something*

ready
"
On 74880 Baud it reports a bunch of stuff, but most interesting is the very first line:
 ets Jan  8 2013,rst cause:2, boot mode:(3,6).

Testing reveals: When connecting GPIO00 to high for normal boot - the Serial Monitor shows exactly the same.

Oh.

Ahhhh!

No wonder I can't program the chip if it isn't even in the right mode!
The 10k resistor seems to be too much for GND to properly register at GPIO00, thus not starting the UART Mode (programming mode) but Flash Mode (normal boot) instead.

I replaced the 10k (probably, I mean, I didn't measure it, but the other ones from the same batch from China seem to be 10k) resistor between GPIO0 and GND with a 3.9k. Voila, 
"
writing flash
................................................................................ [ 34% ]
................................................................................ [ 69% ]
.....................................................................            [ 100% ]
starting app without reboot
" (and some more lines before and after).

Also, the proper behavior of the Serial Monitor when starting the esp in UART Mode seems to be to show only one line of garbage data on 115200 Baud which is revealed as
"
 ets Jan  8 2013,rst cause:2, boot mode:(1,7)
"
on 74880 Baud. Dates might differ, of course.

StefanHermannBerlin, 2017-05-11:
On my NodeJS it was a sensor connected to D8 what caused the trouble. Connecting it to D7 solved it. I recommend trying to replicate the error with a NodeJS disconnected from everything (expect the USB of course).

larryn46, 2017-06-18:
I got the same error message when I selected the wrong board. I selected "generic ESP8266" when I should have selected "NodeMCU 1.0 (ESP 12-E Module)"

Veer034, 2017-10-18:
I had same problem for couple of days.
Follow steps shown in this site: https://diyhacking.com/esp8266-tutorial/

Edena87, 2018-02-05:
Buenas tardes. Tengo problemas con subir mi programa a mi placa Wemos D1 wifi 

El Sketch usa 261163 bytes (52%) del espacio de almacenamiento de programa. El máximo es 499696 bytes.
Las variables Globales usan 34192 bytes (41%) de la memoria dinámica, dejando 47728 bytes para las variables locales. El máximo es 81920 bytes.
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed
error: espcomm_upload_mem failed

Alguien me puede ayudar? Ya cambin los baudios y nada

larryn46, 2018-02-06:
There is more than one cause for this message. Make sure the Tools > Upload speed is either 115200 or 230400. Check that the USB port is working and the correct port is selected. : Tools: Port. My computer changes the USB port every time I unplug it and re-plug it in. Sometimes a wire connected to a pin will keep the program from uploading. I had this happen with D0 conneted to RST. Other pins can do it also, possibly when D1 was connected to a LED to ground. I haven't used Wemos D1, so there may be other causes that are specific to this board.

Baypenguen, 2018-02-22:
Hello friends,
I was having the same problem.
*********
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed
error: espcomm_upload_mem failed
*********

I have read the forum and tried to. I worked on it :)

Solution:

I connected a GND crossover cable between GPIO0.

[image]

mavega998, 2018-06-14:
@Sephok I had the problem with ESP8266 ESP-01

> warning: espcomm_sync failed
> error: espcomm_open failed
> error: espcomm_upload_mem failed
> error: espcomm_upload_mem failed

I solved doing similar that you. With this connection, didn't problem
[image]
[image]

khoiwalvikram2, 2018-11-17:
try selecting nodemcu  v1.0 board it will fix that issue... it worked for me

USvER, 2018-11-17:
Fixed my issue by playing with upload speed, only "921600" worked... all other showed this error...

lincronin, 2019-02-01:
i have download latest version of tasmota, using arduino ide to flash and upload this firmware to my sonoff TH16: I need help on how to fix this issue.  
Sketch uses 582272 bytes (60%) of program storage space. Maximum is 958448 bytes.
Global variables use 50524 bytes (61%) of dynamic memory, leaving 31396 bytes for local variables. Maximum is 81920 bytes.
Uploading 586416 bytes from C:\Users\Linda\AppData\Local\Temp\arduino_build_724539/sonoff.ino.bin to flash at 0x00000000
................................................................................ [ 13% ]
................................................................................ [ 27% ]
................................................................................ [ 41% ]
................................................................................ [ 55% ]
................................................................................ [ 69% ]
................................................................................ [ 83% ]
................................................................................ [ 97% ]
.............                                                                    [ 100% ]
Invalid library found in C:\Program Files (x86)\arduino\libraries\lib: no headers files (.h) found in C:\Program Files (x86)\arduino\libraries\lib
Invalid library found in C:\Program Files (x86)\arduino\libraries\lib: no headers files (.h) found in C:\Program Files (x86)\arduino\libraries\lib

thank you linda

devyte, 2019-02-02:
Hi @lincronin I think that last error is harmless, it just means you have an invalid library, e.g. lib may be empty.

darrahts, 2019-02-07:
I know the issue is closed but **there should be no reason to go through all of the stuff mentioned here**. First time using the nodeMCU's I got the blinky sketch to work, and then something happened between my laptop updating and developing on different machines. I spent days trying to figure out why some of my old sketches, when edited or deployed to new NodeMCUs would work, and nothing else would.
```
Board: "NodeMCU 1.0 (ESP-12E Module)"
Upload Speed: "115200"
CPU Frequency: "80 MHz"
Flash Size: "4M (no SPIFFS)"
Debug port: "Disabled"
Debug Level: "None"
lwIP Variant: "v2 Lower Memory"
VTables: "Flash"
Exceptions: "disabled"
Erase Flash: "Only Sketch"
Port: <your_port>
```
This worked across different machines, using[ this link](http://arduino.esp8266.com/stable/package_esp8266com_index.json) to add the board file to boards manager. I downloaded "**esp8266** by **ESP8266 Community** version 2.5.0".

Sigurd00, 2019-04-11:
I had this problem as well, but none of these solutions fixed the probem. What did though, was changing _Erase Flash_ option from _Only sketch _ to _All Flash Contents_. Hope it helps

supernovae, 2019-04-12:
I had this problem in Windows 10. Once my NodeMCU was connected i opened up device driver, clicked com ports and right clicked on siliconlabs cp210x usb uart and clicked "search for updates" - windows updated the driver and rebooted. No its working without issue.

ChanJeff123, 2019-06-23:
as i did ,if U have sensor connected to 8266 or nodemcu ,before upload u can remove them .after uplaoded than can connect  them.maybe cause of the power

vladutzzz24, 2019-07-10:
I had the same problem. My bachelor degree is an IOT project, i have to present it next week and today this happened and i nearly had a heart attack :)). After some research i find out that the some pins are using the internal timers and the sensor connected to a certain pin (in my case, D4) was resetting the timer and i could not upload code anymore. The system had an external power supply and that would't fix the issue either. So, try removing every sensor one by one and find out if the problem persists. Good day!

B-vaishnavi, 2019-11-25:
I had the same problem..
Warning:espcomm_upload_mem failed 
 but none of the above mentioned solutions fixed the problem.

intiharj, 2020-02-09:
> I had this problem in Windows 10. Once my NodeMCU was connected i opened up device driver, clicked com ports and right clicked on siliconlabs cp210x usb uart and clicked "search for updates" - windows updated the driver and rebooted. Now its working without issue.

thanks, @supernovae, this fixed it for me

[2 more replies]