esp8266/Arduino #772: WiFiWebServer example not stable

ThomasCLee, 2015-09-12:
I am not sure why the WiFiWebServer example sometimes work and sometimes doesn't.

When it doesn't work, it seems to reach to this point but never detects a client . Any idea why?

 WiFiClient client = server.available();
  if (!client) {
    return;
  }

igrr, 2015-09-14:
This may happen for a number of reasons. For instance you might end up in a situation when all the TCP connections at the ESP side are in TIME_WAIT state. The server will not respond to another request until one of the connections gets freed up after timeout (four minutes as per TCP spec).

You might want to use wireshark to see what's going on at the network level when this condition happens.

drmpf, 2015-09-14:
I am looking at this problem at the moment.
Would you add an option to change the connection timeout if no traffic in either direction,
I my case I would set it to 5secs.

ThomasCLee, 2015-09-15:
How do I add connection timeout?

ThomasCLee, 2015-09-27:
Problem is resolved. It turns out that it is an issue with the router. My router has a feature called 'HT 20/40 Coexistence'. After turning it off, it works.

closed 2015-09-27 by ThomasCLee