esp8266/Arduino #773: Ubidots code oddity

slaw54, 2015-09-12:
Would somebody be kind enough to try and compile the following code supplied by the Ubidots site;
I am new to programming in any language.
// Created by Ty Tower in June 11 2015

#include stdio.h>
#include ESP8266WiFi.h>
#define errorPin 13
const int sleepTimeS = 20;// Time to sleep (in seconds):
long lastReadingTime = 0;
WiFiClient client;
char results[4];
String idvariable = "55ea68c476254???";
String token = "9SRNOj52Hkgge1CKuAKVyuvG1RrJ67YBlwxDxAy7hTcDXnrMcZg0T????";

//////////////////////////////////////////////////////////////////////////////////

void setup(){

  pinMode(errorPin, OUTPUT);
  const char* ssid = "wa??";
  const char* password = "la?????";

  for (int i=0;i<4; i++){   // let know we are working
    digitalWrite(errorPin ,HIGH);
    delay(200);
    digitalWrite(errorPin ,LOW);
    delay(200);
  }

  // Create an instance of the server
  // specify the port to listen on as an argument

  WiFiServer server(80);
  Serial.begin(115200);
  delay(10);

  // Connect to WiFi network
  Serial.println();
  Serial.println();
  Serial.print("Connecting to ");
  Serial.println(ssid);

  // WiFi.scanNetworks will return the number of networks found
  int n = WiFi.scanNetworks();
  Serial.println("scan done");

  if (n == 0){
    Serial.println("no networks found");
    Serial.println("Going into sleep");
// ESP.deepSleep(sleepTimeS * 1000000);
  }

  WiFi.begin(ssid, password);

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }

  Serial.println("");
  Serial.println("WiFi connected");

  // Start the server
  server.begin();
  Serial.println("WiFi Server started");
  Serial.println(WiFi.localIP());

}

////////////////////////////////////////////////////////////////////////////////

void loop(){
  int value = (analogRead(A0));
  ubiSave_value(String(value));

  Serial.println("Ubidots data");
  Serial.println(value);
  Serial.println(" Going to Sleep for a while !" );

  // deepSleep time is defined in microseconds. Multiply seconds by 1e6
  //ESP.deepSleep(sleepTimeS * 1000000);//one or other
  delay(20000);
}

void ubiSave_value(String value) {
  Serial.println("help");
  // if you get a connection, report back via serial:
  int num=0;
  String var = "{\"value\": " + String(value)+"}";
  num = var.length();
  if (client.connect("things.ubidots.com", 80)) {
    Serial.println("connected ubidots");
    delay(100);
    client.println("POST /api/v1.6/variables/"+idvariable+"/values HTTP/1.1");
    Serial.println("POST /api/v1.6/variables/"+idvariable+"/values HTTP/1.1");
    client.println("Content-Type: application/json");
    Serial.println("Content-Type: application/json");
    client.println("Content-Length: "+String(num));
    Serial.println("Content-Length: "+String(num));
    client.println("X-Auth-Token: "+token);
    Serial.println("X-Auth-Token: "+token);
    client.println("Host: things.ubidots.com\n");
    Serial.println("Host: things.ubidots.com\n");
    client.print(var);
    Serial.print(var+"\n");

  }
  else {
    // if you didn't get a connection to the server:
    Serial.println("ubidots connection failed");
  }

  if (!client.connected()) {
    Serial.println("NotConnected");
    Serial.println("disconnecting ubidots.");
    client.stop();
    // do nothing forevermore:
    for(;;);
  }

  if (client.available()) {
    char c = client.read();
    Serial.print(c);
  }
}
I have two 32bit windows 7 desktops same version of ide and esp library one machine gives standard stack error other machine uploads code that runs OK but never prints any reply from Ubidots server although the data is posted ok. Mac machine has same problem stack error. I guess the "// do nothing forever more:" is causing the stack WDT problem and I think a disconnect occurs after Content-Length but this is getting way beyond me> all machines compile and upload working Thingspeak code. Any guidance appreciated.

Links2004, 2015-09-12:
yes ```for(;;);``` will trigger a wtd for sure.
your http header is not correct only ```\n``` isn't valid for HTTP.

can not try it but this shut work better:
```cpp
String head = 	"POST /api/v1.6/variables/" + idvariable + "/values HTTP/1.1\r\n"
				"Host: things.ubidots.com\r\n"
				"Content-Type: application/json\r\n"
				"Content-Length: " + String(num) + "\r\n"
				"X-Auth-Token: " + token + "\r\n"
				"\r\n";

client.print(head);
Serial.print(head);
				
client.print(var);
Serial.print(var);
```

slaw54, 2015-09-14:
Thanks for the info. your code is definitely more stable. The problem is appears to be network related, a conventional copper broadband connection gives far less errors than a 3G broadband connection and the do nothing forever was giving the stack errors and wdt reset, I will try removing that and use a timed watchdog reset. Still not sure why thingpeak give a much more stable connection, just that ubidots has a much nicer interface.
Thanks for you help

closed 2015-09-14 by igrr