esp8266/Arduino #776: FS.h no such file

shirish47, 2015-09-13:
ino:3:16: fatal error: FS.h: No such file or directory
compilation terminated.

when I include "FS.h"

it does not find the file ?? 

I have installed esp8266 right now.. whats the problem ??

igrr, 2015-09-13:
Please enable verbose compiler output in Arduino preferences and copy
complete output here. Thanks.

shirish47, 2015-09-13:
C:\Users\ASUS-PC\AppData\Roaming\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9/bin/xtensa-lx106-elf-g++ -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ -IC:\Users\ASUS-PC\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\1.6.5-947-g39819f0/tools/sdk//include -c -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11 -MMD -DF_CPU=80000000L -DARDUINO=10604 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266 -DESP8266 -IC:\Users\ASUS-PC\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\1.6.5-947-g39819f0\cores\esp8266 -IC:\Users\ASUS-PC\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\1.6.5-947-g39819f0\variants\generic C:\Users\ASUS-PC\AppData\Local\Temp\build5901705143095250808.tmp\sketch_sep13a.cpp -o C:\Users\ASUS-PC\AppData\Local\Temp\build5901705143095250808.tmp\sketch_sep13a.cpp.o 
sketch_sep13a.ino:2:16: fatal error: FS.h: No such file or directory
compilation terminated.
Error compiling.

igrr, 2015-09-13:
You are using 'stable' version of the esp8266 core, which doesn't include
FS support yet. Follow the correct documentation link on the main page (in
'stable version' section).

shirish47, 2015-09-13:
OK thanks .. :)

shirish47, 2015-09-13:
so I should use staging version for using FS.h ? right ...

closed 2015-09-14 by igrr

andig, 2015-11-07:
Just hit the same stop. Is there a list of what package is on what version (or whats new in staging) anywhere?

pilipicus, 2015-12-08:
Hi,
I get this error and I don't know how to solve this problem. Please help me!
Thank you in advance

/Applications/Arduino.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware "/Applications/Arduino.app/Contents/Java/hardware" -hardware "/Users/filippoaiello/Library/Arduino15/packages" -tools "/Applications/Arduino.app/Contents/Java/tools-builder" -tools "/Applications/Arduino.app/Contents/Java/hardware/tools/avr" -tools "/Users/filippoaiello/Library/Arduino15/packages" -built-in-libraries "/Applications/Arduino.app/Contents/Java/libraries" -libraries "/Users/filippoaiello/Documents/Arduino/libraries" -fqbn=esp8266:esp8266:generic:UploadTool=esptool,CpuFrequency=80,FlashFreq=40,FlashMode=dio,UploadSpeed=115200,FlashSize=512K64,ResetMethod=ck -ide-version=10606 -build-path "/var/folders/_m/m7fwr7sj4j5c53v2f0zvmhcw0000gp/T/build28d2ebf6aca2ada7c4ee6064d4477f03.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose "/Users/filippoaiello/Documents/Arduino/libraries/arduinoWebSockets-master/examples/WebSocketServer/WebSocketServer.ino"
/Applications/Arduino.app/Contents/Java/arduino-builder -compile -logger=machine -hardware "/Applications/Arduino.app/Contents/Java/hardware" -hardware "/Users/filippoaiello/Library/Arduino15/packages" -tools "/Applications/Arduino.app/Contents/Java/tools-builder" -tools "/Applications/Arduino.app/Contents/Java/hardware/tools/avr" -tools "/Users/filippoaiello/Library/Arduino15/packages" -built-in-libraries "/Applications/Arduino.app/Contents/Java/libraries" -libraries "/Users/filippoaiello/Documents/Arduino/libraries" -fqbn=esp8266:esp8266:generic:UploadTool=esptool,CpuFrequency=80,FlashFreq=40,FlashMode=dio,UploadSpeed=115200,FlashSize=512K64,ResetMethod=ck -ide-version=10606 -build-path "/var/folders/_m/m7fwr7sj4j5c53v2f0zvmhcw0000gp/T/build28d2ebf6aca2ada7c4ee6064d4477f03.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose "/Users/filippoaiello/Documents/Arduino/libraries/arduinoWebSockets-master/examples/WebSocketServer/WebSocketServer.ino"
"/Users/filippoaiello/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -w -x c++ -M -MG -MP -DF_CPU=80000000L -DARDUINO=10606 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266  -DESP8266 "-I/Users/filippoaiello/Library/Arduino15/packages/esp8266/hardware/esp8266/2.0.0-rc2/cores/esp8266" "-I/Users/filippoaiello/Library/Arduino15/packages/esp8266/hardware/esp8266/2.0.0-rc2/variants/generic" "/var/folders/_m/m7fwr7sj4j5c53v2f0zvmhcw0000gp/T/build28d2ebf6aca2ada7c4ee6064d4477f03.tmp/sketch/WebSocketServer.ino.cpp"
"/Users/filippoaiello/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -w -x c++ -M -MG -MP -DF_CPU=80000000L -DARDUINO=10606 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266  -DESP8266 "-I/Users/filippoaiello/Library/Arduino15/packages/esp8266/hardware/esp8266/2.0.0-rc2/cores/esp8266" "-I/Users/filippoaiello/Library/Arduino15/packages/esp8266/hardware/esp8266/2.0.0-rc2/variants/generic" "-I/Users/filippoaiello/Library/Arduino15/packages/esp8266/hardware/esp8266/2.0.0-rc2/libraries/ESP8266WiFi/src" "-I/Users/filippoaiello/Documents/Arduino/libraries/arduinoWebSockets-master/src" "-I/Users/filippoaiello/Library/Arduino15/packages/esp8266/hardware/esp8266/2.0.0-rc2/libraries/Hash/src" "/var/folders/_m/m7fwr7sj4j5c53v2f0zvmhcw0000gp/T/build28d2ebf6aca2ada7c4ee6064d4477f03.tmp/sketch/WebSocketServer.ino.cpp"
"/Users/filippoaiello/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -w -x c++ -M -MG -MP -DF_CPU=80000000L -DARDUINO=10606 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266  -DESP8266 "-I/Users/filippoaiello/Library/Arduino15/packages/esp8266/hardware/esp8266/2.0.0-rc2/cores/esp8266" "-I/Users/filippoaiello/Library/Arduino15/packages/esp8266/hardware/esp8266/2.0.0-rc2/variants/generic" "-I/Users/filippoaiello/Library/Arduino15/packages/esp8266/hardware/esp8266/2.0.0-rc2/libraries/ESP8266WiFi/src" "-I/Users/filippoaiello/Documents/Arduino/libraries/arduinoWebSockets-master/src" "-I/Users/filippoaiello/Library/Arduino15/packages/esp8266/hardware/esp8266/2.0.0-rc2/libraries/Hash/src" "/var/folders/_m/m7fwr7sj4j5c53v2f0zvmhcw0000gp/T/build28d2ebf6aca2ada7c4ee6064d4477f03.tmp/sketch/WebSocketServer.ino.cpp"
"/Users/filippoaiello/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -w -x c++ -M -MG -MP -DF_CPU=80000000L -DARDUINO=10606 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266  -DESP8266 "-I/Users/filippoaiello/Library/Arduino15/packages/esp8266/hardware/esp8266/2.0.0-rc2/cores/esp8266" "-I/Users/filippoaiello/Library/Arduino15/packages/esp8266/hardware/esp8266/2.0.0-rc2/variants/generic" "-I/Users/filippoaiello/Library/Arduino15/packages/esp8266/hardware/esp8266/2.0.0-rc2/libraries/Hash/src" "-I/Users/filippoaiello/Library/Arduino15/packages/esp8266/hardware/esp8266/2.0.0-rc2/libraries/ESP8266WiFi/src" "-I/Users/filippoaiello/Documents/Arduino/libraries/arduinoWebSockets-master/src" "/var/folders/_m/m7fwr7sj4j5c53v2f0zvmhcw0000gp/T/build28d2ebf6aca2ada7c4ee6064d4477f03.tmp/sketch/WebSocketServer.ino.cpp"
"/Users/filippoaiello/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -w -x c++ -M -MG -MP -DF_CPU=80000000L -DARDUINO=10606 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266  -DESP8266 "-I/Users [... cut ...]
                                   ^
compilation terminated.
Using library ESP8266WiFi at version 1.0 in folder: /Users/filippoaiello/Library/Arduino15/packages/esp8266/hardware/esp8266/2.0.0-rc2/libraries/ESP8266WiFi 
Using library arduinoWebSockets-master at version 1.2 in folder: /Users/filippoaiello/Documents/Arduino/libraries/arduinoWebSockets-master 
Using library Hash at version 1.0 in folder: /Users/filippoaiello/Library/Arduino15/packages/esp8266/hardware/esp8266/2.0.0-rc2/libraries/Hash 
exit status 1
Error compiling.

Links2004, 2015-12-08:
the ```core_esp8266_features.h``` is added in the latest git.
it is not included in any release.

simple create a empty one and the WebSockets will work in your case.
or update your core to the latest git version.

pilipicus, 2015-12-08:
Many thanks for your prompt reply. Really appreciated! :+1:

-I have deleted the Arduino library to avoid in possible conflict between the Esp8266 relases.
(In the past I had problems switching between the Stable and the Staging version replacing the json link ).
-I have created a new folder in Arduino/hardware/esp8266com/esp8266 and inside I have copied all the files included in the .zip file downloaded from Git (esp8266/Arduino).
-I have relaunched Arduino IDE and my sketch (https://github.com/Links2004/arduinoWebSockets/blob/master/examples/WebSocketServer/WebSocketServer.ino)

I get the folowing error:

/Applications/Arduino.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware "/Applications/Arduino.app/Contents/Java/hardware" -hardware "/Users/filippoaiello/Documents/Arduino/hardware" -tools "/Applications/Arduino.app/Contents/Java/tools-builder" -tools "/Applications/Arduino.app/Contents/Java/hardware/tools/avr" -built-in-libraries "/Applications/Arduino.app/Contents/Java/libraries" -libraries "/Users/filippoaiello/Documents/Arduino/libraries" -fqbn=esp8266com:esp8266:generic:UploadTool=esptool,CpuFrequency=80,FlashFreq=40,FlashMode=dio,UploadSpeed=115200,FlashSize=512K64,ResetMethod=ck -ide-version=10606 -build-path "/var/folders/_m/m7fwr7sj4j5c53v2f0zvmhcw0000gp/T/builda689ab7f6c8bbc359133bb69a688b8c2.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose "/Users/filippoaiello/Documents/Arduino/libraries/arduinoWebSockets-master/examples/WebSocketClient/WebSocketClient.ino"
/Applications/Arduino.app/Contents/Java/arduino-builder -compile -logger=machine -hardware "/Applications/Arduino.app/Contents/Java/hardware" -hardware "/Users/filippoaiello/Documents/Arduino/hardware" -tools "/Applications/Arduino.app/Contents/Java/tools-builder" -tools "/Applications/Arduino.app/Contents/Java/hardware/tools/avr" -built-in-libraries "/Applications/Arduino.app/Contents/Java/libraries" -libraries "/Users/filippoaiello/Documents/Arduino/libraries" -fqbn=esp8266com:esp8266:generic:UploadTool=esptool,CpuFrequency=80,FlashFreq=40,FlashMode=dio,UploadSpeed=115200,FlashSize=512K64,ResetMethod=ck -ide-version=10606 -build-path "/var/folders/_m/m7fwr7sj4j5c53v2f0zvmhcw0000gp/T/builda689ab7f6c8bbc359133bb69a688b8c2.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose "/Users/filippoaiello/Documents/Arduino/libraries/arduinoWebSockets-master/examples/WebSocketClient/WebSocketClient.ino"
"/Users/filippoaiello/Documents/Arduino/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++" -w -x c++ -M -MG -MP -DF_CPU=80000000L -DARDUINO=10606 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266  -DESP8266 "-I/Users/filippoaiello/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266" "-I/Users/filippoaiello/Documents/Arduino/hardware/esp8266com/esp8266/variants/generic" "/var/folders/_m/m7fwr7sj4j5c53v2f0zvmhcw0000gp/T/builda689ab7f6c8bbc359133bb69a688b8c2.tmp/sketch/WebSocketClient.ino.cpp"
fork/exec /Users/filippoaiello/Documents/Arduino/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++: no such file or directory
Error compiling.

Links2004, 2015-12-08:
follow this 
https://github.com/esp8266/Arduino#using-git-version-

too use the git version

pilipicus, 2015-12-08:
This time I have follwed the instructions on the link that you just gave me:
cd hardware
mkdir esp8266com
cd esp8266com
git clone https://github.com/esp8266/Arduino.git esp8266

The error seems to be exactly the same:

/Applications/Arduino.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware "/Applications/Arduino.app/Contents/Java/hardware" -hardware "/Users/filippoaiello/Documents/Arduino/hardware" -tools "/Applications/Arduino.app/Contents/Java/tools-builder" -tools "/Applications/Arduino.app/Contents/Java/hardware/tools/avr" -built-in-libraries "/Applications/Arduino.app/Contents/Java/libraries" -libraries "/Users/filippoaiello/Documents/Arduino/libraries" -fqbn=esp8266com:esp8266:generic:UploadTool=esptool,CpuFrequency=80,FlashFreq=40,FlashMode=dio,UploadSpeed=115200,FlashSize=512K64,ResetMethod=ck -ide-version=10607 -build-path "/var/folders/_m/m7fwr7sj4j5c53v2f0zvmhcw0000gp/T/build28d2ebf6aca2ada7c4ee6064d4477f03.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose "/Users/filippoaiello/Documents/Arduino/libraries/arduinoWebSockets-master/examples/WebSocketServer/WebSocketServer.ino"
/Applications/Arduino.app/Contents/Java/arduino-builder -compile -logger=machine -hardware "/Applications/Arduino.app/Contents/Java/hardware" -hardware "/Users/filippoaiello/Documents/Arduino/hardware" -tools "/Applications/Arduino.app/Contents/Java/tools-builder" -tools "/Applications/Arduino.app/Contents/Java/hardware/tools/avr" -built-in-libraries "/Applications/Arduino.app/Contents/Java/libraries" -libraries "/Users/filippoaiello/Documents/Arduino/libraries" -fqbn=esp8266com:esp8266:generic:UploadTool=esptool,CpuFrequency=80,FlashFreq=40,FlashMode=dio,UploadSpeed=115200,FlashSize=512K64,ResetMethod=ck -ide-version=10607 -build-path "/var/folders/_m/m7fwr7sj4j5c53v2f0zvmhcw0000gp/T/build28d2ebf6aca2ada7c4ee6064d4477f03.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose "/Users/filippoaiello/Documents/Arduino/libraries/arduinoWebSockets-master/examples/WebSocketServer/WebSocketServer.ino"
"/Users/filippoaiello/Documents/Arduino/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-I/Users/filippoaiello/Documents/Arduino/hardware/esp8266com/esp8266/tools/sdk/include" -c -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DARDUINO=10607 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266  -DESP8266       "-I/Users/filippoaiello/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266" "-I/Users/filippoaiello/Documents/Arduino/hardware/esp8266com/esp8266/variants/generic" "/var/folders/_m/m7fwr7sj4j5c53v2f0zvmhcw0000gp/T/build28d2ebf6aca2ada7c4ee6064d4477f03.tmp/sketch/WebSocketServer.ino.cpp" -o "/dev/null"
"/Users/filippoaiello/Documents/Arduino/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-I/Users/filippoaiello/Documents/Arduino/hardware/esp8266com/esp8266/tools/sdk/include" -c -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DARDUINO=10607 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266  -DESP8266       "-I/Users/filippoaiello/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266" "-I/Users/filippoaiello/Documents/Arduino/hardware/esp8266com/esp8266/variants/generic" "/var/folders/_m/m7fwr7sj4j5c53v2f0zvmhcw0000gp/T/build28d2ebf6aca2ada7c4ee6064d4477f03.tmp/sketch/WebSocketServer.ino.cpp" -o "/dev/null"
"/Users/filippoaiello/Documents/Arduino/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-I/Users/filippoaiello/Documents/Arduino/hardware/esp8266com/esp8266/tools/sdk/include" -c -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DARDUINO=10607 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266  -DESP8266       "-I/Users/filippoaiello/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266" "-I/Users/filippoaiello/Documents/Arduino/hardware/esp8266com/esp8266/variants/generic" "/var/folders/_m/m7fwr7sj4j5c53v2f0zvmhcw0000gp/T/build28d2ebf6aca2ada7c4ee6064d4477f03.tmp/sketch/WebSocketServer.ino.cpp" -o "/var/folders/_m/m7fwr7sj4j5c53v2f0zvmhcw0000gp/T/build28d2ebf6aca2ada7c4ee6064d4477f03.tmp/preproc/ctags_target_for_gcc_minus_e.cpp"
fork/exec /Users/filippoaiello/Documents/Arduino/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++: no such file or directory
Error compiling.

Links2004, 2015-12-08:
you miss the 
```
cd esp8266/tools
python get.py
```
part

pilipicus, 2015-12-08:
Finally it works! Thank you very much ;)