esp8266/Arduino #779: SPIFFS Space

EUA, 2015-09-14:
Please provide a tool for reporting about how much space is available / used / capacity of SPIFFS system.
Here is a hint about how to do it.
http://www.esp8266.com/viewtopic.php?f=23&t=1598

pgollor, 2015-09-14:
Do you mean a function which calculates the free space at runtime?
For this you can use a lop like this:
https://github.com/esp8266/Arduino/blob/esp8266/hardware/esp8266com/esp8266/doc/reference.md#directory-object-dir
And make a sum of all file sizes.

Otherwise you can download the SPIFFS image from your ESP8266 and use the `mkspiffs` tool with the `-l` option.

igrr, 2015-09-14:
No it should be more like [statvfs](http://linux.die.net/man/2/statvfs) POSIX function. I'll add smth like that.

pgollor, 2015-09-14:
Ok, i misunderstand the post. Thanks for your great work for the hole project Iggr.

closed 2015-11-30 by igrr