esp8266/Arduino #782: SPIFFS.exists() doesn't exist

EUA, 2015-09-14:
On git version:
The code : SPIFFS.exists(path) returns 

error: 'class fs::FS' has no member named 'exists'

https://github.com/esp8266/Arduino/blob/esp8266/hardware/esp8266com/esp8266/doc/reference.md#exists

igrr, 2015-09-14:
Could you please enable verbose compiler output in Arduino preferences and copy the full output here?

EUA, 2015-09-14:
I think it's not related with it. Simply there is no exists() function.
Anyway:
```
Build options changed, rebuilding all
Using library Time in folder: /home/death/Arduino/libraries/Time 
Using library ESP8266WiFi in folder: /home/PIC/arduino-1.6.6/hardware/esp8266com/esp8266/libraries/ESP8266WiFi 
Using library ESP8266WebServer in folder: /home/PIC/arduino-1.6.6/hardware/esp8266com/esp8266/libraries/ESP8266WebServer 
Using library ESP8266mDNS in folder: /home/PIC/arduino-1.6.6/hardware/esp8266com/esp8266/libraries/ESP8266mDNS (legacy)
Using library EEPROM in folder: /home/PIC/arduino-1.6.6/hardware/esp8266com/esp8266/libraries/EEPROM 
Using library Ticker in folder: /home/PIC/arduino-1.6.6/hardware/esp8266com/esp8266/libraries/Ticker 

/home/PIC/arduino-1.6.6/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++ -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ -I/home/PIC/arduino-1.6.6/hardware/esp8266com/esp8266/tools/sdk//include -c -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11 -MMD -DF_CPU=160000000L -DARDUINO=10606 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266 -DESP8266 -I/home/PIC/arduino-1.6.6/hardware/esp8266com/esp8266/cores/esp8266 -I/home/PIC/arduino-1.6.6/hardware/esp8266com/esp8266/variants/generic -I/home/death/Arduino/libraries/Time -I/home/PIC/arduino-1.6.6/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src -I/home/PIC/arduino-1.6.6/hardware/esp8266com/esp8266/libraries/ESP8266WebServer/src -I/home/PIC/arduino-1.6.6/hardware/esp8266com/esp8266/libraries/ESP8266mDNS -I/home/PIC/arduino-1.6.6/hardware/esp8266com/esp8266/libraries/EEPROM -I/home/PIC/arduino-1.6.6/hardware/esp8266com/esp8266/libraries/Ticker /tmp/build9121599321242415614.tmp/CS5490Powermeter3.cpp -o /tmp/build9121599321242415614.tmp/CS5490Powermeter3.cpp.o 
CS5490.ino: In function 'double CS5490_GetData2(int32_t, int, timing)':
CS5490.ino:853:18: error: 'class fs::FS' has no member named 'exists'
'class fs::FS' has no member named 'exists'
```
Problem is, I think it's forgotten to implement in in new SPIFFS library.
And wiki is probably for old FS library.

igrr, 2015-09-15:
This function was added yesterday in 75a2928 along with documentation:
https://github.com/esp8266/Arduino/blob/esp8266/hardware/esp8266com/esp8266/cores/esp8266/FS.h#L100-L101

It is currently used in ESP8266WebServer library, and all the compilation tests have passed, which leads me to believe the function is in place. 
Could you please post the output of `git rev-parse HEAD` as well?

EUA, 2015-09-16:
I re-update now and it's working.
Thanks for implementing it +1

closed 2015-09-16 by EUA