esp8266/Arduino #784: Help with HTTPS

torntrousers, 2015-09-14:
I'm trying to use the new https support but can't get it to work so wonder if someone could see what the problem is (soory if this is leaping ahead, i know the https code hasn't been released yet). I've built the latest git code from today (1f8c14d) and started with the HTTPSRequest sample. That works ok (though does always seem to get a wdt reset after closing the connection), so i then changed it to do a GET to https:/www.btopenzone.com:8443, and that worked ok too, so then updated it to try to do a POST to http:/www.btopenzone.com:8443/tbbLogon but that doesn't work and all i get back is a stream of ÿ characters.

This is the code i have, can anyone see a problem? 
```
/*
 *  HTTP over TLS (HTTPS) example sketch
 *
 *  This example demonstrates how to use
 *  WiFiClientSecure class to access HTTPS API.
 *  We fetch and display the status of
 *  esp8266/Arduino project continous integration
 *  build.
 *
 *  Created by Ivan Grokhotkov, 2015.
 *  This example is in public domain.
 */

#include <ESP8266WiFi.h>
#include <WiFiClientSecure.h>

const char* ssid = "BTHub5-72W5";
const char* password = "46d38ec753";

//const char* host = "api.github.com";
//const int httpsPort = 443;
const char* host = "www.btopenzone.com";
const int httpsPort = 8443;

// Use web browser to view and copy
// SHA1 fingerprint of the certificate
const char* fingerprint = "CF 05 98 89 CA FF 8E D8 5E 5C E0 C2 E4 F7 E6 C3 C7 50 DD 5C";

void setup() {
  Serial.begin(115200);
  Serial.println();
  Serial.print("connecting to ");
  Serial.println(ssid);
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());

  // Use WiFiClientSecure class to create TLS connection
  WiFiClientSecure client;
  Serial.print("connecting to ");
  Serial.println(host);
  while (!client.connect(host, httpsPort)) {
    Serial.println("connection failed");
//    return;
  }

  if (client.verify(fingerprint, host)) {
    Serial.println("certificate matches");
  } else {
    Serial.println("certificate doesn't match");
  }

//  String url = "/repos/esp8266/Arduino/commits/esp8266/status";
  String url = "/tbbLogon";
  String postData = "username=qaz123@btinternet.com&password=abc123&xhtmlLogon=https://www.btopenzone.com:8443/tbbLogon";
  Serial.print("requesting URL: ");
  Serial.println(url);
/*
  client.print(String("POST ") + url + " HTTP/1.1\r\n" +
               "Host: " + host + "\r\n" +
               "User-Agent: BuildFailureDetectorESP8266\r\n" +
               "Connection: close\r\n\r\n");
*/
  client.print("POST /tbbLogon HTTP/1.1\n");
  client.print("Host: www.btopenzone.com:8443\n");
  client.print("Connection: close\n");
//  client.print("Connection: keep-alive\n");
//  client.print("User-Agent: Mozilla/5.0 (Windows NT 6.1; WOW64; rv:38.0) Gecko/20100101 Firefox/38.0\n");
  client.print("Content-Type: application/x-www-form-urlencoded\n");
  client.print("Content-Length: ");
  client.print(postData.length()+2);
  client.print("\n\n");
  client.print(postData);

  Serial.println("request sent");

  Serial.println("Receiving response");

  while (client.connected()) {
    /*
    String line = client.readStringUntil('\n');
    Serial.println(line);
    if (line == "\r") {
      Serial.println("headers received");
      break;
    }
    */
    Serial.write(client.read());
    delay(10);
  }
  Serial.println("2");
  String line = client.readStringUntil('\n');
  Serial.println(line);
  Serial.println("closing connection");
}

void loop() {
    delay(1000);
    Serial.println("done");
}
```
As an alternative test this does work returning an html page:

curl --data "username=qaz123%40btinternet.com&password=abc123&xhtmlLogon=https://www.btopenzone.com:8443/tbbLogon" https://www.btopenzone.com:8443/tbbLogon

igrr, 2015-09-14:
```
Serial.write(client.read());
```

When there is no character to be read, client.read() returns -1, which is then converted to a character by Serial.write. This results in a character with ASCII code of 255, which may look like ÿ depending on the terminal. You might want to check if `client.available() > 0` before calling `read`.

There may be other issues, but this one is the most obvious.

torntrousers, 2015-09-14:
Ah ok, i should have known that. But i did have the client.available() check before and then it just hangs never getting back a response. Trying again now adding this code after the Serial.println("Receiving response") then it just sits forever printing out dots:
```
  while (client.available() < 1) {
    Serial.print(".");
    delay(100);
  } 
```

igrr, 2015-09-14:
What happens if you change Content-Length header to match the real size of content?

Links2004, 2015-09-14:
your header lineending need to be \r\n look in the http RFC for more info

torntrousers, 2015-09-14:
Something like this?
```
  client.print("POST /tbbLogon HTTP/1.1\r\n");
  client.print("Host: www.btopenzone.com:8443\r\n");
  client.print("Connection: close\r\n");
  client.print("Content-Type: application/x-www-form-urlencoded\r\n");
  client.print("Content-Length: ");
  client.print(postData.length());
  client.print("\r\n");
  client.print(postData);
  Serial.println("request sent");
  Serial.println("Receiving response");

  while (client.available() < 1) {
    Serial.print(".");
    delay(100);
  }
```
Still just dots and no response comes back.

igrr, 2015-09-14:
You still seem to be missing an empty line after the headers (see the `\r\n\r\n` sequence in the original example).

torntrousers, 2015-09-14:
Ok, rearanging to be closely like the original example:
```
  client.print(String("POST ") + url + " HTTP/1.1\r\n" +
               "Host: " + host + "\r\n" +
               "User-Agent: BuildFailureDetectorESP8266\r\n" +
               "Content-Type: application/x-www-form-urlencoded\r\n" +
               "Content-Length: " + postData.length() + "\r\n" +
               "Connection: close\r\n\r\n");

  client.print(postData);

  Serial.println("request sent");
  Serial.println("Receiving response");

  while (client.available() < 1) {
    Serial.print(".");
    delay(100);
  }
```
but still just dots and no response.

torntrousers, 2015-09-15:
Now i've tried running tcpmon on a local computer with a local non-ssl http port being forwarded to www.btopenzone.com port 8443 as SSL, and then changing the esp sketch to use WiFiClient instead of WiFiClientSecure and to use the tcpmon host address. 
That works, tcpmon forwards the post request to btopenzone as ssl which returns an html response which the esp sketch receives and prints out. So the esp sketch must be formating the POST request and headers ok. 
What else could the problem be, some bug with WiFiClientSecure?

igrr, 2015-09-15:
Correct, if the same POST request works with WiFiClient then it's likely a bug in WiFiClientSecure.

Links2004, 2015-09-15:
at every print you send a tcp package some servers dont like this.
you can try it like this:
https://github.com/esp8266/Arduino/issues/773#issuecomment-139753667

you also shut have "\r\n\r\n" before  postData or the server will interpreter it as invalidate request.

torntrousers, 2015-09-15:
Getting a bit further...i think there might be a bug in WifiClientSecure - it looks like after sending the request client.available() is always returning 0 even when a response has come back, but doing client.readStringUntil('\n') does return the first line, and then client.available() starts working properly. You can see this by modifying the HTTPSRequest sample to have a while loop printing client.available() before and after the client.readStringUntil call.

So accounting for that i can get the https POST to work ok, yipee! 

Unfortunately it still doesn't work for what i want, which is trying to have the ESP logon to a [FON Wifi network](https://corp.fon.com/en). Not really for this git issue but i wonder if it needs to have two connections, one for the https connection which kept open and another to do the other http requests on - can the esp do multiple client connections?

Links2004, 2015-09-15:
yes up to 5 TCP and 4 UDP connections at one time.

torntrousers, 2015-09-15:
Ok thanks for confirming that. Doing some debugging with netstat on a pc i don't think it uses multiple connections anyway. Not sure how it works, in case anyone has any suggestions here's how it works:
There are lots of FON access points named (in the UK) BTWifi-with-FON that are open with no security. If you try to use a browser then you can't connect to anything until you go to a FON web page with a form where you have to logon with a userid password, and then after that you're connected and all internet access works normally (for a short while after which you need to reenter the userid/password again). 

I traced the web page form and it just does a POST so can now do the logon with curl - curl --data "username=qaz123%40btinternet.com&password=abc123&xhtmlLogon=https://www.btopenzone.com:8443/tbbLogon" https://www.btopenzone.com:8443/tbbLogon

So i was hoping to replicate that on an ESP and do the https post and then be able to make other http connections to publish data. The https post seems to work fine and the headers and response the esp gets back look the same as what the curl command gets, but then further WifiClient requests on the ESP don't work and just fail to connect.

I thought it might just be using the mac address against the userid/password or somehting like that but i guess there must be more to it.

Any ideas?

Links2004, 2015-09-15:
for me this looks like the FON access points runs a transparent proxy. 
so the allow your ip or mac for some time to use the proxy. 
where do you try to connect to? may they only allow some ports / protocols.

torntrousers, 2015-09-15:
It started working! Don't no what i did, just tidied the code up a bit  i thought, but now it works ok. Very pleased. Thanks for all the help.

Would be good to get the WifiClientSecure.available bug fixed though so i'll leave this open for that.

closed 2015-09-16 by igrr

torntrousers, 2015-09-18:
Thanks for the fix, i can confirm it does fix it for me.

yongkianpeng, 2015-11-14:
Torntrousers, I would love to try out HTTPS request from ESP, could you tell me where can I download the WiFiClientSecure lib please? Thank you.

gicho, 2015-11-14:
You should install the staging version of the ESP8266-Arduino libraries using this URL for board manager in preferences:
http://arduino.esp8266.com/staging/package_esp8266com_index.json

yongkianpeng, 2015-11-14:
Thanks gicho, actually I already have HTTP working with ESP-12E board  using this URL for board manager in preferences: http://arduino.esp8266.com/versions/1.6.5-1160-gef26c5f/package_esp8266com_index.json. Updated the ESP8266WIFI lib and the compiler is recognizing the WiFiClientSecure.h now. I ran the example above, it connected, certificate matches but the return file config.htm has 0 Bytes (File /config.htm 0 Bytes). Web Browser (https://raw.githubusercontent.com/sticilface/ESPmanager/Strings/examples/Settingsmanager-example/data/config.htm) return differently. Any idea? Thank you in advance.

Links2004, 2015-11-14:
may its the same problem then here:
https://github.com/esp8266/Arduino/issues/43#issuecomment-155565679

its most likely the TLS fragment size that is to big.
the ESP can only handle SSL / TLS fragment size of 4K.

electronicsguy, 2016-03-29:
@Links2004 @gicho @torntrousers Can someone tell me if I can make HTTPS GET requests using ESP-01? Or is it only possible on the models with more memory?

Links2004, 2016-03-29:
You can all esp8266 has the same ram size, only the flash is different.

sreeramtkd, 2017-02-12:
Is there a fix for this ? I'm trying to connect to the below URL and getting the -1 as output or at times an empty string is passed to serial output. 

https://api.apifier.com/v1/execs/APIKEY/results

electronicsguy, 2017-02-12:
@sreeramtkd  Yes ESP-01 works with SSL over http. Use the WifiClientSecure library here: [WiFiClientSecure](https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/src/WiFiClientSecure.cpp)

sreeramtkd, 2017-02-12:
@electronicsguy I'm using the latest library but unfortunately I can't get esp8266 to pull https data

connecting to Sree
..............
WiFi connected
IP address: 
192.168.1.4
connecting to api.apifier.com
Certificate matches
requesting URL: /v1/execs/To5wxSEcGZGYzLurA/results
request sent
reply was:
==========
String is
ÿ
==========
closing connection

electronicsguy, 2017-02-12:
@sreeramtkd   Well the library does have some restrictions as @igrr  has mentioned in the documentation for it. (like encryption type, max. certificate size, etc). But you need to post your code with a minimal example for others to check.

sreeramtkd, 2017-02-12:
Here is the code from library

#include <ESP8266WiFi.h>
#include <WiFiClientSecure.h>

const char* ssid = "";
const char* password = "";

const char* host = "api.apifier.com";
const int httpsPort = 443;

// Use web browser to view and copy
// SHA1 fingerprint of the certificate
const char* fingerprint = "17 67 4C 08 F1 22 7B B4 1C 6F 5B 01 22 AB 00 06 24 8E D8 6B";

void setup() {
  Serial.begin(115200);
  Serial.println();
  Serial.print("connecting to ");
  Serial.println(ssid);
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
  // Use WiFiClientSecure class to create TLS connection
  WiFiClientSecure client;
  Serial.print("connecting to ");
  Serial.println(host);
  if (!client.connect(host, httpsPort)) {
    Serial.println("connection failed");
    return;
  }

 if (client.verify(fingerprint, host)) {
  Serial.println("Certificate matches");
 } else {
    Serial.println("Certificate doesn't match");
}
  String url = "/v1/execs/To5wxSEcGZGYzLurA/results";
  Serial.print("requesting URL: ");
  Serial.println(url);
  client.print(String("GET ") + url + " HTTP/1.1\r\n" +
               "Host: " + host + "\r\n" +
               "Connection: close\r\n\r\n");
  Serial.println("request sent");
  int repeatCounter = 10;
  while (!client.available() && repeatCounter--) {
    delay(500);
  }
  String line;
    line  = client.readStringUntil('\n');
  Serial.println("reply was:");
  Serial.println("==========");
  Serial.println(line);
  Serial.println("==========");
  Serial.println("closing connection");
}
void loop() {
}

sreeramtkd, 2017-02-12:
also FYI i'm using an ESP8266 12-F module

electronicsguy, 2017-03-30:
@sreeramtkd   apifier works for me using [HTTPSRedirect](http://bit.ly/1SopEug).  Please check it out:

This is what I get as the expected output:
`
  "_id": "<id removed>",
  "actId": "CwNxxSNdBYw7NWLjb",
  "startedAt": "2017-02-13T14:54:44.162Z",
  "finishedAt": "2017-02-13T14:54:56.197Z",
  "status": "SUCCEEDED",
  "statusMessage": "Crawled 5 pages, limit is 5 pages",
  "tag": null,
  "stats": {
    "downloadedBytes": 153322,
    "pagesInQueue": 0,
    "pagesCrawled": 5,
    "pagesOutputted": 5,
    "pagesFailed": 0,
    "pagesCrashed": 0,
    "pagesRetried": 0,
    "totalPageRetries": 0,
    "storageBytes": 17992
  },
  "meta": {
    "source": "API",
    "method": "POST",
    "clientIp": "redacted",
    "userAgent": "redacted"
  },
  "detailsUrl": "https://api.apifier.com/v1/execs/<id redacted>",
  "resultsUrl": "https://api.apifier.com/v1/execs/<id redacted>/results"
}
`

evbelda, 2017-04-17:
Hi,

I'm trying to use the ESP-01 module+Mega conect to TX/RX 18 and 19 Pin but it doesn't work with ESP8266Wifi library.
How do you configure the board to accept a Mega (only accept ESP8266 boards) with an ESP module?
How do you connect the ESP-01 module pins to work?

sherikapotein, 2017-06-04:
Hi,

Going through various related posts I could get my Wemos to send and accept https requests. Chhers to the community for that !! :). I'd used the similar code as shared by @sreeramtkd above. 

For purpose of testing I'd used a free webhost(Heliohost) and used a LetsEncrypt SSL certificate and it worked smoothly.

My end goal was to implement the same functionality on my local network where my Wemos/ESPs would communicate with a server running Nginx and communicate over https. I could setup the Nginx with a self signed certificate as instructed [here](https://www.digitalocean.com/community/tutorials/how-to-create-a-self-signed-ssl-certificate-for-nginx-in-ubuntu-16-04). 

I did not wish to have any sorts of Internet connectivity from my local network, because of which I could not use the same LetsEncrypt certificate again. 

After setting up the Nginx, the Wemos is unable to connect to the server on port 443 using Https but can communicate with it over port 80. It gives a plain "connection failed". I could have moved ahead had the error been related to certifate, wrong fingerprint or something which happens after connection is established. But a refusal to connect keeps me hanging. Tried to skim through the source code but its probably over my limited knowledge. 

I have verified port 443 (for Https) to be open and accessible by browser clients on the network.

Some inputs on troubleshooting please.

AbdelsalamHaa, 2018-01-27:
Hi 
i'm trying to connect my esp8266 to my http server , i have read so many code and try to understand , a could understand few but im still not very clear how to do it . i think my problem is in the request message i really really hope someone can help me fixing it my code is here thanks in advance 

#include <ESP8266WiFi.h>

const char* ssid = "-----------";
const char* password = "-------------";

const char* host = "35.187.233.153";

void setup()
{
  Serial.begin(115200);
  Serial.println();

  Serial.printf("Connecting to %s ", ssid);
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED)
  {
    delay(500);
    Serial.print(".");
  }
  Serial.println(" connected");
}

void loop()
{
  WiFiClient client;

  Serial.printf("\n[Connecting to %s ... ", host);
  if (client.connect(host, 80))
  {
    Serial.println("connected]");

    Serial.println("[Sending a request]");
    

   client.print("POST ");
   client.print("/api/trash/update?id=1&level=99&battL=9999");// here are the data i wanna transimte they are supposed to part of the url .  
   client.print(" HTTP/1.1\r\n");
   client.print("Host:35.187.233.153\r\n");
   client.print("Content-Type: application/x-www-form-urlencoded\n");

   client.print("Connection: close\r\n");
   

    

    Serial.println("[Response:]");
    while (client.connected())
    {
      if (client.available())
      {
      
        String line = client.readStringUntil('\n');
        Serial.println(line); 
      }
    }
    client.stop();
    Serial.println("\n[Disconnected]");
  }
  else
  {
    Serial.println("connection failed!]");
    client.stop();
  }
  delay(5000);
}

sreeramtkd, 2018-02-13:
@AbdelsalamHaa  - Would be nice to see the error / issue in specific, Use any of the available online client testing services to see if you are getting proper response from the server mentioned in code. https://www.hurl.it/ Or Postman Echo. This should allow you to debug issues.

AbdelsalamHaa, 2018-02-24:
@sreeramtkd thanks for ur responds , I have actually found the problem which was in my request message format .

Craig1516, 2018-03-07:
AbdelsalamHaa, would you mind sharing your final code?  Or does your Jan 27 post include final?  

Thanks,
Craig

edugargon, 2019-02-05:
> Hi,
> 
> I'm trying to use the ESP-01 module+Mega conect to TX/RX 18 and 19 Pin but it doesn't work with ESP8266Wifi library.
> How do you configure the board to accept a Mega (only accept ESP8266 boards) with an ESP module?
> How do you connect the ESP-01 module pins to work?

I'm in the same situation but with an Arduino UNO and ESP-01 with SoftwareSerial.
I can not find a way to add a WiFiClientSecure instance to use as a gateway to UniversalTelegramBot, since the ESP8266WiFi.h library is not compatible with Arduino boards.
Has anyone been able to do it? Thank you very much.
Regards.

electronicsguy, 2019-02-08:
@edugargon  If you have a esp8266, what are you using Arduino for? Unless you need analog inputs, the arduino is unnecessary.

edugargon, 2019-02-09:
> @edugargon If you have a esp8266, what are you using Arduino for? Unless you need analog inputs, the arduino is unnecessary.

You said, if I'm using Arduino, it's because I need analog inputs.

klslowik, 2020-05-12:
> I'm in the same situation but with an Arduino UNO and ESP-01 with SoftwareSerial.
> I can not find a way to add a WiFiClientSecure instance to use as a gateway to UniversalTelegramBot, since the [...]

I have the same problem right now.
Were you able to solve it?

JiriBilek, 2020-05-12:
AFAIK this problem is not related to this repository. You can sort it out by flashing an AT firmware (from Espressif repository) to your ESP8266 board and use an Arduino library to communicate over serial (e.g. WifiEsp library).
But the communication via serial if far from perfect.

commit 89df2854 (2015-09-16): WiFiClientSecure::available fix

Attempt to read data from SSL engine inside WiFiClientSecure::available() if RX buffer is empty.
Fix #784.