esp8266/Arduino #785: ESP.getFlashChipSize() returns different values

robertgregor, 2015-09-14:
Hello,
I have strange behaviour of the ESP.getFlashChipSize(). I have bought several ESP-12 modules. I wanted to check the flash size, to be able to set correct value in the Flash size menu. So I have created simple sketch where I put:
void setup() {
  Serial.begin(115200);
  Serial.print("Flash size: ");
  Serial.println(ESP.getFlashChipSize(), DEC);
}
It gave me 1M flash. OK, I have set it and loaded the big shetch - around Sketch uses 433,064 bytes
Then at the begining I have called this again and I've got Flash size: 524288

How this is possible???

robertgregor, 2015-09-15:
Hello, OK, I have found on google interresting thing, that the size is stored in the flash and it is overwritten when there is a bin download. So this is not relevant and better solution is to check the chipId and based on this dedicate the real flash size. Maybe you can add the comment to the reference manual, because it can be confusing.

igrr, 2015-09-15:
documentation [mentions this](https://github.com/esp8266/Arduino/blob/146072cbba937c7f22c716e54b892ba76b8efdf4/hardware/esp8266com/esp8266/doc/reference.md#esp-specific-apis):

> `ESP.getFlashChipSize()` returns the flash chip size, in bytes, as seen by the SDK (may be less than actual size).

Links2004, 2015-09-15:
may this helps you:
https://github.com/Links2004/Arduino/blob/esp8266/hardware/esp8266com/esp8266/libraries/esp8266/examples/CheckFlashConfig/CheckFlashConfig.ino

```ESP.getFlashChipRealSize();``` will return the size based of the flash chipId

closed 2015-11-27 by Links2004