esp8266/Arduino #786: somtimes no resons or with 4s delay and no entering loop

marc3l, 2015-09-15:
I setup an esp 07 modul as webserver with some webpages in Flash. somtimes the respose is very quick and it is working as aspected. but in a lot of cases the modul needs to long to respose so the browser goes to error.
I triggerd a port pin to see the mainloop. after asking for a webpage the esp needs up to 4s to re-enter the loop() where is only "server.handleClient();". Is this a normal behavior or am I doing something wrong?

pgollor, 2015-09-15:
Please upload your hole sketch here.
Are you using the F() macro or SPIFFS for your pages?

marc3l, 2015-09-15:
I'm not sure what you mean with  F() and SPIFFS. 

I took project from http://www.john-lassen.de/index.php/projects/esp-8266-arduino-ide-webconfig and changed it to my needs

how can I poste code?

marc3l, 2015-09-15:
```  
#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ESP8266WebServer.h>
#include <Ticker.h>
#include <EEPROM.h>
#include <Wire.h>
#include "helpers.h"
#include "global.h"
/*
Include the HTML, STYLE and Script "Pages"
*/
#include "html_index.h"
#include "html_settings.h"
#include "html_taster.h"
#include "php_relais.h"

#define ACCESS_POINT_NAME  "TestNW"				
#define ACCESS_POINT_PASSWORD  "12345678" 
//#define AdminTimeOut 180  // Defines the Time in Seconds, when the Admin-Mode will be diabled

void setup ( void ) {
  Init_IO();
	EEPROM.begin(512);
	Serial.begin(115200);
	delay(500);
	
	WiFi.mode(WIFI_AP_STA);
	WiFi.softAP( ACCESS_POINT_NAME , ACCESS_POINT_PASSWORD);

	ConfigureWifi();
	
	server.on ( "/index.html", []() {     Serial.println("?index.html\r\n");      server.send ( 200, "text/html", PAGE_index );   }  );
  server.on ( "/taster.html", []() {    Serial.println("?taster.html\r\n");     server.send ( 200, "text/html", PAGE_taster );   }  );
  server.on ( "/schalter.html", []() {  Serial.println("?index.html\r\n");      server.send ( 200, "text/html", PAGE_index );   }  );
  server.on ( "/settings.html", []() {  Serial.println("?settings.html\r\n");   server.send ( 200, "text/html", PAGE_settings );   }  );
 
 
  server.on ( "/relais.php", []() {   /* Serial.print("?relais.php\r\n");*/ processPHP_relais();   }  );

	server.onNotFound ( []() { /*Serial.println("Page Not Found"); */         server.send ( 400, "text/html", "Page not Found" );   }  );
	server.begin();
	Serial.println( "HTTP server started" );
	TimerSekunde.attach(1,SekundenTakt);
  TimerSekunde.attach(0.01,HundertstelSekundenTakt);
  Serial.println(String(ESP.getFlashChipSize()));
}
 
void loop ( void ) {
	
	server.handleClient();

  if(IO_changed){
    IO_changed = 0;
    handleIO();
  }
    

 ```
    
}

pgollor, 2015-09-15:
Sorry i don't know the `esp-8266-arduino-ide-webconfig ` project. But i think it is not a general problem with the Arduino IDE or libraries.
Please ask the owner of your project library for help.

For debugging you can try to enable the serial debug output. Maybe you get some information from there.

marc3l, 2015-09-15:
Thank you for your answers.

Well I do think there is a general problem. I changed a couple of things and it seems that the pages which a stored in "PROGMEM", like:
  ```        
const char PAGE_example[] PROGMEM = R"=====(<!DOCTYPE html><html>
  <head></head><body>test</body></html>)=====";         
 ```
where only loaded one time correct.
other pages are working fine. 
 - Is it possible that there is a problem with server.send() of PROGMEM 
 - Can anyone give me a little example of a stable version?
 - how do I know how big the Flash and SPIFFS are to set them correct in the arduino IDE

Links2004, 2015-09-15:
for checking the flash config you can use:
https://github.com/Links2004/Arduino/blob/1393463448d27595c218927cad428b184650a74b/hardware/esp8266com/esp8266/libraries/esp8266/examples/CheckFlashConfig/CheckFlashConfig.ino

marc3l, 2015-09-16:
Thank you that helped me. I just set the correct FLASH size an it is working.

closed 2015-09-30 by igrr