esp8266/Arduino #788: Personnally, forr many reasons, I'm not a big Arduino community fan

fredericplante, 2015-09-15:
Could it be possible to put more emphasis on UECIDE. I feel it could be a big plus for this project.

igrr, 2015-09-16:
This boils down to the bandwidth we have to support different IDEs.
As far as I know, this core works with Arduino, Eclipse (using some plugin), Visual Studio (using some plugin), and UECIDE (probably with some changes, i.e. core.txt file?).
I'm more or less IDE agnostic so I choose to work in the environment which most consumers of this project will use, and that is Arduino. I know @Links2004 uses Eclipse and keeps this core compatible with Arduino Eclipse plugin. So if there is someone who has time to work on better support of other IDEs (including UECIDE), this is totally welcome. Personally I don't have enough bandwidth to keep working on new features and supporting another IDE at the same time, so I choose to work on new features (as time permits) and fix issues.
Perhaps supporting UECIDE core is not all that complicated. I wonder if @majenkotech can give us more insight.

fredericplante, 2015-09-16:
By the way, @majenkotech, the FS.h does not load in UECIDE, and it's the only thing to disable me from only using your IDE.. ;)

Thank you @igrr for you great work as team leader, and coding dude.

majenkotech, 2015-09-16:
There's slightly more to it than just a core.txt file.  You also need a compiler.txt file for the compiler package, a board.txt file for each board, and you need to package it all up properly.

I have scripts that do it all for us in Linux but I haven't recently updated it with any latest changes.

What's the problem you're getting with FS.h?

fredericplante, 2015-09-16:
I will get back to you when I get home about the FS.h thing thing there.

closed 2016-04-11 by igrr