esp8266/Arduino #791: ESP-01 and flash mode after power off...

direk, 2015-09-16:
Hi, 

Currently I am playing with ESP-01 and have only GPIO 0 and 2 avaliable. That is enough for me, but i am wondering if I can solve problem described below in software way...

I have IR LED connected beetwen GPIO0 and GND. After reset (power off and on) module sees GPIO connected to GND (thru LED) and boots in flash mode. I know that I can fix it by driving led via transistor, but is there any other way (software?) to prevent module to go into flash mode after power cycle?

igrr, 2015-09-16:
This is determined by the code written into chip's ROM. There is no
software way to change behavior of bootstrapping pins.

direk, 2015-09-16:
So... for both GPIO0 and 2 and LED (or similar element) the only way is to use transistor circuit. 
What a bummer :/

igrr, 2015-09-16:
Why not connect the LED between VCC and GPIO0/2? Just drive the pin low,
this will turn on the LED.

direk, 2015-09-16:
I was thinking about it, but it also complicates other things - in IR transmission zeroes becomes ones and vice versa. I'll need to modify code responsible for IR signal sending... I'll give it a try.

For now it looks like easiest SW hack. HW solution is transiator.

igrr, 2015-09-16:
Yet another option which doesn't require you to modify IR code is to
connect LED between GPIO0 and GPIO2. Set GPIO0 low and drive LED with
GPIO2. Also be sure to have ~10k pull-up resistors between both GPIOs and
VCC.

drmpf, 2015-09-16:
As igrr said,  check out http://www.forward.com.au/pfod/ESP8266/GPIOpins/ESP8266_01_pin_magic.html

closed 2015-09-16 by igrr

Diaoul, 2015-10-10:
@igrr: with the following setup:

    VCC -> LED -> GPIO 2 (pinMode OUTPUT)

Is it necessary to add a resistor to protect the LED from too much current or is there any built-in resistor in GPIO2 when the pin is set to LOW?

Diaoul, 2015-10-10:
Answering myself: yes, an external resistor is needed.
Tests showed I sank 33mA powering a simple LED directly from VCC and 8mA only with 220ohms resistor.

Later I found this:
>Resistor's should be used on GPIO pins. If you connect GPIO2 directly to 3.3V for example, then >enable GPIO2 as an output, set it low, you are effectively shorting 3.3V to Ground through the GPIO >pin. Internal resistance (and perhaps some protection devices internally) may limit the current so it's >not fatal but it may exceed the allowable current of 25mA. Not to mention it will heat up the chip >package and waste some power. 

Source: http://www.esp8266.com/viewtopic.php?f=13&t=2533#p14826

vijaysurat, 2016-06-04:
i am also facing such prob with my esp01 and pir sensor.. i have connected pir output to esp pin gpio-0 via a npn tr to invert signal. but while startup sometime pir output go high and gpio-0 goes low..so esp goes in flash mode. i tried gpio-02 but same thing happened ..

tioraulv, 2016-07-12:
I think you could use rx like gpio3. I use it for a similar project. http://www.forward.com.au/pfod/ESP8266/GPIOpins/ESP8266_01_pin_magic.html