esp8266/Arduino #792: File content corrupted for large file on SPIFFS

mytrain, 2015-09-16:
Hi, 
I met some issues when try to download then write a file on the SPIFFS file system.
In my project the file is stored with the same size but its content is altered. Therefore I wrote a small sketch to help you analyzing the issue but the sketch ends up with a truncated file that is also an abnormal file.
The sketch downloads a file (on one of my public server), computes it its CRC32 then saves it on the SPIFFS file system.
When reading back the file and computing the CRC32 I get a different result, moreover in the case of the scketch the file size is not equals to the number of byte downloaded.

The sketch is available on http://37.59.92.209/cms/images/mytrain/private/SPIFFSWriteError.ino
The downloaded file : http://37.59.92.209/cms/images/mytrain/private/steelseries-min.js (size=228467,CRC=0x6e3cc165)
The logs of the serial dump when running the sketch : http://37.59.92.209/cms/images/mytrain/private/logs.txt

I would like to know what's wrong in my scketch.
Thanks you.

PS: Moreover from time to time I got the following exception:

: 3ffec4b0 end: 3ffec7c0 offset: 01b0

>>>stack>>>
3ffec660:  0001a4df 3ffeae18 402017ec 3ffeb7a0  
3ffec670:  402010ae 00000005 00000005 3ffeae18  
3ffec680:  0001a4df 3ffec6d8 3ffec818 402095bb  
3ffec690:  3fff52c8 00000001 00000000 00000100  
3ffec6a0:  0000001f 00000000 3fff5318 00000066  
3ffec6b0:  00000066 3ffec854 3ffec850 3fff46d8  
3ffec6c0:  3ffe9ad0 00000000 000003e8 40203e08  
3ffec6d0:  3fff52c8 3fff5318 3ffe96d0 00000000  
3ffec6e0:  000003e8 0000b231 00000000 3fff51d0  
3ffec6f0:  3ffe9b30 d15c3b25 0000000b 3fff53d8  
3ffec700:  0000002d 00000001 00000005 4020188f  
3ffec710:  000005b4 3ffec76c 00000095 402018c8  
3ffec720:  00000040 3fff46d8 3fff4360 3ffea764  
3ffec730:  3ffec848 000255ab 3ffec818 40209957  
3ffec740:  3ffe9ad0 00000000 000003e8 00000000  
3ffec750:  3fff5110 3fff5160 3fff5188 00000016  
3ffec760:  00000016 00000000 00000000 3fff46b8  
3ffec770:  3fff50e8 00000000 00000000 00000000  
3ffec780:  3fff5188 00000000 00000000 00000000  

3ffec790:  00000000 00000000 00000000 3ffec7ec  
3ffec7a0:  3fffdc20 00000000 3ffec7e4 4020185f  
3ffec7b0:  00000000 00000000 3ffeb7a0 40100398  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(1,6)

igrr, 2015-09-16:
Thanks for the detailed report. I'll look into this.
Which flash size/configuration do you have selected in the IDE Tools menu?

mytrain, 2015-09-16:
igrr,
I'm using a 1M + 3M SPIFFS on an ESP-12 with 4M memory
My IDE is in staging version 1.6.5-1106-g8253b82

mytrain, 2015-09-28:
For you information, I found a workaround by downloading a gzipped version of the files (so file size is lower than 64K now).

closed 2015-11-30 by igrr