esp8266/Arduino #793: ESP.restart, KAPUT, :)

fredericplante, 2015-09-16:
When I do this,:

else if (commande == F("redemarer"))ESP.restart();

most of the time I get:

ets Jan  8 2013,rst cause:2, boot mode:(3,6)                                   
                                                                                
load 0x4010f000, len 1264, room 16                                              
tail 0                                                                          
chksum 0x42                                                                     
csum 0x42                                                                       
~ld  

and every thing is fine. But some time I get this:

>redemarer                                                                      
 ets Jan  8 2013,rst cause:2, boot mode:(1,7)                                   
                                                                                
                                                                                
 ets Jan  8 2013,rst cause:4, boot mode:(1,7)                                   
                                                                                
wdt reset  

 and it get stock there. I must pull the reset pin physically  to restart the board correctly.

luc-github, 2015-09-16:
I use ESP.restart() on latest staging and do not have such issue - I even used recently to be able to show another issue (https://github.com/esp8266/Arduino/issues/780)

May be giving more context would help

fredericplante, 2015-09-16:
Wait up, I will make my "os" print some ESP stats before restarting. I will get back to this soon.

I would say, that most of the time, it work well. But not all the time. To do this example, I pasted up there, I got 1 fail for 4 ok.

Links2004, 2015-09-16:
check your pins ```boot mode:(1,7)``` indicates a problem with GPIO 0,2 or 15
it shut look like this:
```ets Jan  8 2013,rst cause:2, boot mode:(3,7)```

fredericplante, 2015-09-16:
Well actually, I have put my restart fuction in my "autoexec" script, basically what it does when it boot, and I have not get another problem, on a regular basis. I get a few bug here and there but nothing else.

Are you saying I should use a resistance to physically pull up or down on each pin, to get this random bug out of my way?

Links2004, 2015-09-16:
yes this may helps, do you have connected some thing to the pins currently?

mtnbrit, 2015-09-16:
i have the same problem on NodeMCU v1.0 board with esp12-e module. 8/10 times it doesn't reboot but stays stuck at wdt. no problem on NodeMCU v0.9, or any bare 12e module. I think its something in the NodeMCU V1.0 circuit.

what board or module are you having this problem on?

asetyde, 2015-09-16:
this is know issue

check gpio0 and 2 not be used or must be hight before reset ,

other things when you flash , if you want test reboot must be cut power and reboot manually , after board reboot normal .

Probably there are a problem when board nodemcu 1.0 is flashed , something not put correct .

fredericplante, 2015-09-16:
Yeah, I get :

ets Jan 8 2013,rst cause:2, boot mode:(3,6) 

And not 

ets Jan 8 2013,rst cause:2, boot mode:(3,7) 

I don't know if this is bad, or evil ;), but any way. when I get this line, the ESP will or has reboot what look normally.

I'v put my board into full loading my autoexecution initial script then reboot, and it has screew once about 2 hour ago. I must admit that this new stagging version is better then the last one.

calozano, 2015-11-29:
I got the same issue showing the reset was for watchdog. I just Insert "ESP.wdtDisable();" inside loop and it doesn´t reset again.

jumpjack, 2016-10-05:
insert where and how?!?

devyte, 2017-10-19:
Closing as duplicate of #1017 .

closed 2017-10-19 by devyte