esp8266/Arduino #795: +full HEADER in ESP8266WebServer

xbary, 2015-09-16:
Hi

I would ask to be added to the official version of such an allowance:

-----------------------------------------------------------------------------------------------------
in file ESP8266WebServer.h (on comment // Add by XBary)
...
  WiFiClient client() { return _currentClient; }
  HTTPUpload& upload() { return _currentUpload; }
  String HEADER() { return _currentHEADER; } // Add by XBary

  String arg(const char* name);   // get request argument value by name
...
and
...
HTTPMethod  _currentMethod;
  String      _currentUri;
  String      _currentHEADER; // Add by XBary

  size_t           _currentArgCount;
  RequestArgument* _currentArgs;
...
-----------------------------------------------------------------------------------------------------
in file Parsing.cpp
...
  String req = client.readStringUntil('\r');
  client.readStringUntil('\n');
  _currentHEADER = req + "\r\n"; // Add by XBary

  // First line of HTTP request looks like "GET /path HTTP/1.1"
  // Retrieve the "/path" part by finding the spaces
...
and
...
      headerName = req.substring(0, headerDiv);
      headerValue = req.substring(headerDiv + 2);
	_currentHEADER += (req + "\r\n"); // Add by XBary
	  
	  #ifdef DEBUG
	  DEBUG_OUTPUT.print("headerName: ");
...
and
...
      if (headerDiv == -1){
        break;
      }
      headerName = req.substring(0, headerDiv);
      headerValue = req.substring(headerDiv + 2);
	 _currentHEADER += (req + "\r\n"); // Add XBary
	  
	  #ifdef DEBUG
	  DEBUG_OUTPUT.print("headerName: ");
	  DEBUG_OUTPUT.println(headerName);
	  DEBUG_OUTPUT.print("headerValue: ");
...
-----------------------------------------------------------------------------------------------------

Example use:
...
void handle_mainmenu()
{
	WiFiClient client = server.client();
	String hostheader = server.HEADER();
	int sesID = GetSession(client, hostheader);
...

The application mainmenu now have access to the entire HTTP header. I read User-agent: or cookies, etc. 

Regards XBary

brianensor, 2015-10-13:
My vote (if I get one) is to make headers accessible in the same way as arguments. Then one could do something like server.header("Cookie") to get the value. I've coded this up reusing RequestArgument and it's working great. I'll submit a pull request later today or tomorrow. Thanks for the great project!

xbary, 2015-10-13:
And so be ... now wait for an official patch. Thanks for brianensor for taking the topic.

Links2004, 2015-11-27:
see #933

closed 2015-11-27 by Links2004