esp8266/Arduino #799: mac os x 10.11 unable to upload

yoyellow, 2015-09-17:
SInce my upgrade to mac os x 10.11 l Capitan (GM) in am unable to upload any sketches using different usb2serial adapters to my esp8266-01.

i tried various different esp-boards and various different wiring setups (i have 3 different usb2serial adapters)

i dont know why i can not get into contact with the esp board.
i get timeout errors and finaly: error: espcomm_open failed.

i tried both staging and stable, same results..

what are next steps to do..?

yoyellow, 2015-09-17:
I have it working, sorry for wasting time, it seems i screwed up something, three times in a row..

closed 2015-09-17 by yoyellow

123dj, 2015-10-06:
@yoyellow do you remember how you got it to work?  I'm on El Capitan too and it's not work.  I've spent ~4 hours on it so far.

igrr, 2015-10-06:
I've upgraded today, and the only issue i had were some kexts which prevented the system from booting. Having fixed that, it's all back to normal.

Ah, aside from the fact that we can no longer use those unsigned CH340 drivers. Bye-bye nodemcu...

yoyellow, 2015-10-07:
The issue i was having was faulty wiring, so that was my own bad.

if you are using a serial2usb with the CH340(G) on El Cap it still works, i have installed the drivers after disabling SIP for the installation (after that you should enable it again)

you can pay for driver: https://www.mac-usb-serial.com

or you can try this free method: http://tzapu.com/2015/09/24/making-ch340-ch341-serial-adapters-work-under-el-capitan-os-x/

free one worked for me.

123dj, 2015-10-07:
I figured it out. I simply was using 5V instead of 3.3V and blew my board :(.