esp8266/Arduino #800: Is this kinda like cert pinning?

cottsak, 2015-09-18:
I don't know much about certificate pinning but I can see the value to it when one is concerned with root CA lookups, MiM attacks, etc.

Does anyone (@igrr?) know if this line https://github.com/esp8266/Arduino/blob/8d20e8e58622d289c6195b9126b9f3674a7892c3/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/examples/HTTPSRequest/HTTPSRequest.ino#L51 is kind of like certificate pinning? ie, checking th cert hash explicitly to make sure it's the one you want rather than just relying on the library logic to verify the cert chain?

igrr, 2015-09-18:
Basically, yes.
When you open https page with a web browser it will inspect certificate chain to verify the certificate. If it gives you a green light, you can copy the certificate hash (SHA-1) and put it into your sketch. Matching the hash successfully tells you that server has sent exactly the same certificate as you expect.

The second part of the story is matching the hostname in the certificate and checking that certificate hasn't expired. At this point, this is not yet done.

cottsak, 2015-09-18:
@igrr thanks mate

closed 2015-09-18 by cottsak