esp8266/Arduino #801: Udp.beginPacket() does not support broadcast addresses

burner-, 2015-09-18:
I am not sure is this bug or missing feature but looks that if you set broadcast address to Udp.beginPacket() it will not able to send it. Naturally broadcast needs logic to stack side because it use static mac address and does not require arp resolv. Anyhow I found from library some code what looks like that it try handle bcast addresses, so it is more bug than missing feature?

pgollor, 2015-09-18:
Please read the references:
https://github.com/esp8266/Arduino/blob/esp8266/hardware/esp8266com/esp8266/doc/reference.md#wifiesp8266wifi-library

The function name your looking for is: `beginPacketMulticast`

igrr, 2015-09-18:
Multicast is not the same as broadcast :)

I haven't tried sending broadcast, will take a look. Perhaps something's
indeed broken.

pgollor, 2015-09-18:
I am so sorry. I wanted to help and red the post to fast.
Please ignore my comment. :-)

I think i tried it to send information over broadcast, but i can not remember if it had worked. I will search for the scetch and will leave a post here.

pgollor, 2015-09-18:
```
#include <ESP8266WiFi.h>
#include <WiFiUdp.h>

WiFiUDP g_udp;

const int g_port = 8888;

void setup()
{
  // connect to WiFi
  WiFi.begin("SSID", "PSK");

  // start UDP server
  g_udp.begin(g_port);
}

void loop()
{
  IPAddress ip = WiFi.localIP();
  ip[3] = 255;
  
  // transmit broadcast package
  g_udp.beginPacket(ip, g_port);
  g_udp.write("Hello\n");
  g_udp.endPacket();
  
  delay(2000);
}
```
This works for me.

burner-, 2015-09-19:
pgollor, You are right. I just verify and it looks that it working. I have no idea why it does not work earlier  because I testes several times so that with multicast it worked and with bcast not. Anyhow looks that I cannot reproduce that and most probably problem was on my first test enviroment so I close that.

closed 2015-09-19 by burner-

nicolasbisi, 2015-12-19:
Hi there. Have any of you made it work using the broadcast ip 255.255.255.255?
By some reason, the esp8266 is not sending the message but its returning an ok code.
I opened another issue: https://github.com/esp8266/Arduino/issues/1252
Thanks