esp8266/Arduino #804: Error while esp8266 is running the code

AkhilReddyMandadi, 2015-09-19:
I connected HX711 to my arduino uno and trying to upload the data from it through the digital pins (7,8) to an website . The code is uploading perfectly but when I open the serial monitor im getting this....

ets Jan 8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld

is thrown by device via serial interface.
any idea what i`m making wrong?

polygontwist, 2015-09-22:
not enough power? Built back-up capacitor?

Links2004, 2015-09-23:
GPIO 7 + 8 are connected to the SPI flash you can not use them.
may this helps:
https://github.com/esp8266/Arduino/blob/esp8266/hardware/esp8266com/esp8266/doc/boards.md

closed 2015-11-30 by igrr