esp8266/Arduino #807: Unable to Control Servos via header file

mbbackus, 2015-09-21:
I'm trying to control some servos using a NodeMCU 1.0. I'm able to do so without a header file and class, but when I try to abstract that layer away into a separate file/class, the program simply doesn't work. This technique works fine with regular Arduinos.

Main program:
```
#include <Servo.h>
#include "NanoMouseMotors.h"

const byte ledPin = 16;
const byte buttonPin = 12;

NanoMouseMotors motors;

Servo leftServo;

void setup()
{
  Serial.begin(9600);
  motors.hello();
  motors.attach(10, 9);  
  Serial.println("Motors attached.");
  motors.forward();
  Serial.println("Signal sent to motors.");
}

void loop()
{
}
```

And here's the header file:
```
#include <Arduino.h>
#include <Servo.h>

class NanoMouseMotors{
private:
    Servo leftServo;
    Servo rightServo;

    static const byte power = 250;

public:
    void hello()
    {
      Serial.println("Hello from inside the NanoMouseMotors class.");
    }

    void attach(byte leftMotor,byte rightMotor)
    {
        leftServo.attach(leftMotor);
        rightServo.attach(rightMotor);
    }
    
    void forward()
    {
      Serial.println("Just entered the forward() function.");
        leftServo.writeMicroseconds(1500+power);
        rightServo.writeMicroseconds(1500-power);
    }
};
```
When I run it, I can tell I've entered the forward() function, thanks to the println inside. But it seems to hang on telling the servo motors what to do. I have not idea why this is the case since it works when I do everything in a single file without classes.

I'm using 1.6.4 and the stable version of the ESP8266 board definitions.

mbbackus, 2015-10-06:
Turns out it's not an issue with classes. It's an issue with controlling servos using GPIO9. Once I switched it to GPIO0 everything worked as it should. Any ideas on how to get GPIO9 controlling servos?

Links2004, 2015-10-06:
have you used ```D9``` or ```9```?
the NodeMCU board have a different Pin mapping.
https://github.com/esp8266/Arduino/blob/esp8266/hardware/esp8266com/esp8266/doc/boards.md#pin-mapping

```GPIO9``` of the ESP chip is connected to the SPI flash chip using it is not possible.

mbbackus, 2015-10-18:
I was using "9" since by referring to "10" I am able to control a servo on the pin labeled for SD3. Using either "D3" or "0" I am able to control the other servo on the pin labeled D3.

Looking at the following pin map it appears that GPIO9 should be accessible: https://github.com/nodemcu/nodemcu-devkit-v1.0#pin-map

The documentation on analog output states "PWM may be used on pins 0 to 16" http://arduino.esp8266.com/versions/1.6.5-947-g39819f0/doc/reference.html

igrr, 2015-10-18:
@mbbackus Just tried using GPIO9 with Servo library and it works fine. 
You need to reset the board after uploading the code though. ESP8266 serial loader runs in QIO mode, so when the sketch is started after upload, it actually runs in QIO mode. After hard reset bootloader runs again and this time it reads flash configuration from binary image header, setting it to DIO.

mbbackus, 2015-10-18:
I tried resetting the board by pushing the reset button, but that doesn't seem to work. I suspect that by "hard reset" you mean something more. I've tried holding down both the FLASH and RST buttons, but that didn't work either. How do I effect a "hard reset" on the NodeMCU 1.0? Thanks for the help.

igrr, 2015-10-19:
I was simply pressing reset button on the board.
Here's the sketch i have used: https://gist.github.com/igrr/a960d9444c26ae0bd40d

mbbackus, 2015-10-20:
Thanks for the response Igrr. I tried your code on my robot. At first it didn't work no matter which pins I attached the servo to. Once I commented out the WiFi lines, gpio10 (labeled SD3) and pin D4 started working (as with my code). Still no luck with gpio9 (labeled SD2). The NodeMCU 1.0 appears to be resetting every couple seconds. I can't for the life of me see why it's working for you but not me. I tried a different NodeMCU 1.0 and had the same problem.

mbbackus, 2015-10-30:
Igrr,

Do you have to reset in order for pin 10 to work? I don't. Seems weird that I would have to reset for 9 but not 10. Also, what system are you using, which version of the Arduino IDE, and which boards manager url (stable or staging). I'm trying to narrow down why yours works and mine does not.

Also, I ran the following code and found that pin 9 only registers .12 volts when set to high which is odd because it registers over 3 volts when uploading the program and when I set pin 10 to high.
```
void setup() {
  pinMode(9,OUTPUT);
  digitalWrite(9,HIGH);
}

void loop() {}
```
Any ideas/help would be greatly appreciated. Thanks.

sebgiles, 2016-11-06:
NodeMCU v1.0
Esp resets upon calling pinMode(9, OUTPUT)

Tested with blink sketch:
~~~~
void setup() {
  Serial.begin(115200);
  Serial.println("about to set pinmode");
  delay(1000);
  pinMode(9, OUTPUT);
  delay(1000);
  Serial.println("survived setting pinmode");
}

void loop() {
  digitalWrite(9, HIGH);   // turn the LED on (HIGH is the voltage level)
  delay(1000);
  digitalWrite(9, LOW);    // turn the LED off by making the voltage LOW
  delay(1000);
}
~~~~

This gives output 

~~~~
about to set pinmode

 ets Jan  8 2013,rst cause:4, boot mode:(3,4)

wdt reset
load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x0f
csum 0x0f
~ld
~~~~

sebgiles, 2016-11-06:
NodeMCU v1.0

Tested with this code:
~~~~
void setup() {
  Serial.begin(115200);
  Serial.println("about to set pinmode");
  delay(1000);
  pinMode(9, INPUT);
  delay(1000);
  Serial.println("survived setting pinmode");
}

void loop() {
  Serial.println(digitalRead(9));
  delay(1000); 
}
~~~~

If GPIO9 is HIGH there are no problems. 
As soon as GPIO9 goes LOW ESP reboots, no zeros appear on serial. 
If GPIO9 is LOW on startup esp reboots upon calling pinmode.

I hope this information helps

devyte, 2017-10-20:
I believe GPIO9 is connected to the flash on Nodemcu boards, and can't be used for GPIO. 
Closing.

closed 2017-10-20 by devyte

Gargo, 2018-02-27:
I have the same problems but independently from gpio and nodemcu version - servo simply doesn't work