esp8266/Arduino #808: SPIFFS crash

mozgy, 2015-09-21:
Following code works if StaticJson buffer size is 300, but crashes if its ie 3000 -

```
#include "FS.h"

#include <ArduinoJson.h>

File fh_netdata;

uint8_t netCount;

void setup() {

  Serial.begin(115200);
  delay(10);

  bool result = SPIFFS.begin();
  if( !result ) {
    Serial.println("SPIFFS open failed!");
  }

  SPIFFS.remove( "/netdata.txt" );

  netCount = 0;

}

void loop() {

  test();

  delay(60000);

}

bool test( void ) {

  StaticJsonBuffer<3000> thisBuffer;

  JsonObject& tmpObj = thisBuffer.createObject();
  tmpObj["count"] = netCount;

  JsonArray& tmpArray  = tmpObj.createNestedArray("networks");

  fh_netdata = SPIFFS.open("/netdata.txt", "r");

  if ( !fh_netdata ) {
    Serial.println("Open file failed");
    
    fh_netdata = SPIFFS.open("/netdata.txt", "w");
    if ( !fh_netdata ) {
      Serial.println("Data file creation failed");
      return false;
    } else {
      Serial.println("File created ..");
      return true;
    }
  } else {
    Serial.println("File open ..");
    return true;
  }

}
```

with crash ->

```
Open file failed

Exception (28):
epc1=0x40208d83 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000008 depc=0x00000000

ctx: sys 
sp: 3ffe8c90 end: 3fffffb0 offset: 01a0

>>>stack>>>
3ffe8e30:  3fff1298 00000007 00000000 402055c9  
3ffe8e40:  00000001 00000080 00000080 ffffffff  
3ffe8e50:  00000001 00000100 3ffe8b18 40206107  
3ffe8e60:  3fff0f98 00000000 00000020 00000100  
3ffe8e70:  00008001 000000f8 ffffffff 656e2f01  
3ffe8e80:  74616474 78742e61 00000074 00000000  
3ffe8e90:  00000000 00000000 00000000 40207700  
3ffe8ea0:  0000000d 3ffe0094 00000000 00000000  
3ffe8eb0:  17d98001 3ffe8ef6 00100000 00000000  
3ffe8ec0:  00002000 3fff0e98 00000000 00000000  
3ffe8ed0:  00000000 00000000 00000000 00000000  
3ffe8ee0:  3ffe8f30 00000016 3ffe8b18 402035b0  
3ffe8ef0:  3fff10a8 00000001 00000000 00000000  
3ffe8f00:  00000000 00000000 00000000 00000000  
3ffe8f10:  00000000 00000000 00000000 00000016  
3ffe8f20:  3ffe8492 3ffe8b14 3ffe8f80 40202344  
3ffe8f30:  74656e2f 61746164 7478742e 4020d700  
3ffe8f40:  3ffe9c28 0000000f 3ffe84bd 3ffe9aa0  
3ffe8f50:  3ffe9b74 00000010 3ffe9c28 4020a621  
3ffe8f60:  3ffe8720 3ffe8aac 3ffe9c28 3ffe9aa0  
3ffe8f70:  3ffe8492 3ffe8b0c 3ffe9b60 4020a4aa  
3ffe8f80:  00000000 00000000 00000002 00000005  
3ffe8f90:  3ffe9b74 3ffe8aac 00000000 402093c9  
3ffe8fa0:  3ffe8500 3ffe8fa0 3ffe8fac 3ffe8fd4  
3ffe8fb0:  00000000 3ffe849f 00000000 00000004  
3ffe8fc0:  00000000 00000000 00000000 3ffe8fa0  
3ffe8fd0:  00000000 00000000 00000000 3ffe84a5  
3ffe8fe0:  00000000 00000005 00000000 3ffe8fcc  
```

is this SPIFFS crash or ArduinoJson library ?

Links2004, 2015-09-21:
you try the get 3KB out of the heap thats may too much. 
try to move ``` StaticJsonBuffer<3000> thisBuffer;```out of the function or 
use the ```DynamicJsonBuffer``` the Dynamic one uses malloc.

igrr, 2015-09-21:
Actually StaticJsonBuffer is allocated on the stack, which is only 4kB. So yes, switching to DynamicJsonBuffer might be a good idea.

mozgy, 2015-09-21:
Ok, I hear you on using DynamicJsonBuffer,
but in this case crash happened a bit later as you can see that first Serial.println() did some output but second Serial.println() never happened, crash happened possibly in second SPIFFS.open() call.

Links2004, 2015-09-21:
if the stack has only 4KB if you use 3KB for JSON then its may too less for working with SPIFFS.
try to enable the DEBUGV define to get more info.

Links2004, 2015-11-27:
If it still a problem with using ```DynamicJsonBuffer``` feel free to reopen.

closed 2015-11-27 by Links2004