esp8266/Arduino #809: SPIFFS Error: mkspiffs not found!

rakeshpai, 2015-09-22:
I followed the instructions here: https://github.com/esp8266/Arduino/blob/esp8266/hardware/esp8266com/esp8266/doc/reference.md#uploading-files-to-file-system

When I hit the "ESP8266 Sketch Data Upload" menu item, I get an error saying "SPIFFS Error: mkspiffs not found!"

Not sure why that is. If there's any additional info I can provide please let me know. I'm on Mac OSX, if that helps.

Thanks for the great work!

hallard, 2015-09-25:
Hi,
Same there on windows 10, I've put it in c:\windows\ and c:\windows\system32 folders with no luck.
It works from command line (it find it) 
[image]

but not from Arduino IDE.
[image]

Does someone knows where to put the exe file ?
Thanks

gparvind, 2015-09-27:
check this: https://gitter.im/esp8266/Arduino/archives/2015/09/09

hallard, 2015-10-01:
@gparvind 
yeah thanks, I solved also a couple of days ago on #775, I posted but didn't referenced from this thread, here it is https://github.com/esp8266/Arduino/pull/775

rakeshpai, 2015-10-16:
Sorry, but this still isn't working for me. My issue must be specific to mac. Any idea where I mkspiffs should be installed, and where the lookup paths are specified?

closed 2015-11-30 by igrr