esp8266/Arduino #810: Connect to config stored by sdk

cacatpunctro, 2015-09-22:
Hi, it would be nice if you could add:
(untested code)
<pre>
int ESP8266WiFiClass::begin() {
  static struct station_config default_config;
  wifi_station_get_config_default(&default_config);
  
  if(default_config.ssid) { //some ssid exists
    ETS_UART_INTR_DISABLE();
    wifi_station_connect(); //connect directly, no need to  wifi_station_set_config
    ETS_UART_INTR_ENABLE();
  } else { //no ssid
    return WL_NO_SSID_AVAIL;
  }

  return status(); 

}
</pre>

atm, im using(if anyone else interested):

<pre>
extern "C" {
  #include "user_interface.h"
}
void setup() {
  Serial.begin(115200);
  WiFi.mode(WIFI_AP_STA);

  if(get_config_default_ssid()) {
    Serial.println(get_config_default_ssid());
    ETS_UART_INTR_DISABLE();
    wifi_station_connect();
    ETS_UART_INTR_ENABLE();
  } else {
    Serial.println("no default config, ask for it");
  }

  Serial.println(WiFi.waitForConnectResult());

  WiFi.printDiag(Serial);

}

char* get_config_default_ssid()
{
    static struct station_config def_conf;
    wifi_station_get_config_default(&def_conf);
    if(def_conf.ssid) {
      return reinterpret_cast<char*>(def_conf.ssid);
    } else {
      return NULL;
    }
}
</pre>

Links2004, 2015-09-22:
you dont need to do any thing in code to connect to the last stored wifi config.
the SDK will do all the work automatically.
even a code like:
```cpp
void setup() {

}

void loop() {

}
```
will connect, if in the SDK config is a valid wifi setting is stored.

cacatpunctro, 2015-09-23:
That's nice to know. 
WiFi.SSID() calls wifi_station_get_config which is not wifi_station_get_config_default which are really the same thing since only wifi_station_set_config is used and that always stores the config.
That was confusing to me :)

cacatpunctro, 2015-09-23:
Ok, so it seems my issue is different. If you are calling softAP before or after you will not connect to stored config. So you need a method to connect to stored config, ie: calling begin() with no arguments or something else.

<pre>
WiFi.softAP(sta_ssid, sta_password);
WiFi.mode(WIFI_AP_STA);
Serial.println(WiFi.SSID());
Serial.println(WiFi.waitForConnectResult());
</pre>

prints this
<pre>
stored_config_ssid
0
Mode: STA+AP
PHY mode: N
Channel: 1
AP id: 0
Status: 0
Auto connect: 1
SSID (10): stored_config_ssid
Passphrase (11): stored_config_pass
BSSID set: 0
0.0.0.0
192.168.4.1
</pre>

devyte, 2017-10-20:
I think this is covered in the current implementation one way or another. 
Closing.

closed 2017-10-20 by devyte