esp8266/Arduino #811: Compiling Error

rokison, 2015-09-23:
I am using the latest stable board manager release, and am trying to control an OLED display using i2c.
I am getting the following errors:

Arduino: 1.6.5 (Windows 7), Board: "Generic ESP8266 Module, Serial, 80 MHz, 40MHz, DIO, 115200, 512K (64K SPIFFS)"

In file included from \\crusader\RokisonO\Arduino\libraries\ChineseTouchScreen\TouchScreen.cpp:6:0:
C:\Users\rokisono\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\1.6.5-947-g39819f0\variants\generic/pins_arduino.h:37:14: error: 'uint8_t' does not name a type
 static const uint8_t SDA = 4;
              ^
C:\Users\rokisono\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\1.6.5-947-g39819f0\variants\generic/pins_arduino.h:38:14: error: 'uint8_t' does not name a type
 static const uint8_t SCL = 5;
              ^
C:\Users\rokisono\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\1.6.5-947-g39819f0\variants\generic/pins_arduino.h:40:14: error: 'uint8_t' does not name a type
 static const uint8_t SS    = 15;
              ^
C:\Users\rokisono\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\1.6.5-947-g39819f0\variants\generic/pins_arduino.h:41:14: error: 'uint8_t' does not name a type
 static const uint8_t MOSI  = 13;
              ^
C:\Users\rokisono\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\1.6.5-947-g39819f0\variants\generic/pins_arduino.h:42:14: error: 'uint8_t' does not name a type
 static const uint8_t MISO  = 12;
              ^
C:\Users\rokisono\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\1.6.5-947-g39819f0\variants\generic/pins_arduino.h:43:14: error: 'uint8_t' does not name a type
 static const uint8_t SCK   = 14;
              ^
C:\Users\rokisono\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\1.6.5-947-g39819f0\variants\generic/pins_arduino.h:45:14: error: 'uint8_t' does not name a type
 static const uint8_t BUILTIN_LED = 1;
              ^
C:\Users\rokisono\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\1.6.5-947-g39819f0\variants\generic/pins_arduino.h:47:14: error: 'uint8_t' does not name a type
 static const uint8_t A0 = 17;
              ^
Error compiling.

  This report would have more information with
  "Show verbose output during compilation"
  enabled in File > Preferences.

This used to work, so I don't know what is going on. 
uint8_t IS a data type!
Any help gratefully received.

Links2004, 2015-09-23:
can you link the library + code example?

rokison, 2015-09-23:
It's OK, I think it was a library redefining some pins that had already been there. I swapped to a version of the library I had tweaked, and it all seems to work OK now.

closed 2015-09-28 by igrr