esp8266/Arduino #815: Chrome: ERR_CONTENT_LENGTH_MISMATCH

awwende, 2015-09-23:
When I run the following code, I'm getting the error "ERR_CONTENT_LENGTH_MISMATCH".

It happens after a few page refreshes, I get this error, and then it takes a minute for the ESP8266 to start serving the webpage again. I can save the html code to my computer and run the html file, without issue, so I doesn't appear to be a syntax issue.

```
#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>

ESP8266WebServer webserver(80);

String buildJavascript(){
  String javaScript;
  javaScript = "<script type='text/javascript'>";

  javaScript += "function getDate()";
  javaScript += "{";
  javaScript += "var d= new Date();";
  javaScript += "var month = d.getMonth()+1;";
  javaScript += "var day = d.getDate();";
  javaScript += "var year = d.getFullYear().toString().substr(2,2);";
  javaScript += "document.getElementById('inputFileNameToSaveAs').value=month+'-'+day+'-'+year;";
  javaScript += "}";

  javaScript += "function saveTextAsFile()";
  javaScript += "{";
  javaScript += "var textToWrite='Test';";
  javaScript += "var textFileAsBlob = new Blob([textToWrite], {type:'text/plain', endings:'native'});";
  javaScript += "var fileNameToSaveAs = document.getElementById('inputFileNameToSaveAs').value;";
  javaScript += "var downloadLink = document.createElement('a');";
  javaScript += "downloadLink.download = fileNameToSaveAs;";
  javaScript += "downloadLink.innerHTML = 'Download File';";
  javaScript += "if (window.webkitURL != null){";
  javaScript += "downloadLink.href = window.webkitURL.createObjectURL(textFileAsBlob);";
  javaScript += "}";
  javaScript += "else{";
  javaScript += "downloadLink.href = window.URL.createObjectURL(textFileAsBlob);";
  javaScript += "downloadLink.onclick = destroyClickedElement;";
  javaScript += "downloadLink.style.display = 'none';";
  javaScript += "document.body.appendChild(downloadLink);";
  javaScript += "}";
  javaScript += "downloadLink.click();";
  javaScript += "}";
  
  javaScript += "function destroyClickedElement(event){";
  javaScript += "document.body.removeChild(event.target);";
  javaScript += "}";

  javaScript += "var defaultValues = '{\\";
  javaScript += "\"RR2\":0.943,\"TOL2p\":0.02,\"TOL2n\":0.02,\\";
  javaScript += "\"RR3\":0.493,\"TOL3p\":0.03,\"TOL3n\":0.03,\\";
  javaScript += "\"RR4\":0.100,\"TOL4p\":0.03,\"TOL4n\":0.01,\\";
  javaScript += "\"TARGET1\":170,\\";
  javaScript += "\"TARGET2\":130,\\";
  javaScript += "\"TARGET3\":87,\\";
  javaScript += "\"TARGET4\":48,\\";
  javaScript += "\"TARGET5\":5}';";

  javaScript += "var currentValues = '{\\";
  javaScript += "\"RR2\":0,\"TOL2p\":0,\"TOL2n\":0,\\";
  javaScript += "\"RR3\":0,\"TOL3p\":0,\"TOL3n\":0,\\";
  javaScript += "\"RR4\":0,\"TOL4p\":0,\"TOL4n\":0,\\";
  javaScript += "\"TARGET1\":0,\\";
  javaScript += "\"TARGET2\":0,\\";
  javaScript += "\"TARGET3\":0,\\";
  javaScript += "\"TARGET4\":0,\\";
  javaScript += "\"TARGET5\":0}';";

  javaScript += "function loadRelRefs(option){";
  javaScript += "if(option == 'DEFAULT'){";
  javaScript += "var obj = JSON.parse(defaultValues);";
[... 136 lines cut ...]

  uint8_t mac[WL_MAC_ADDR_LENGTH];
  WiFi.softAPmacAddress(mac);
  String macID = String(mac[WL_MAC_ADDR_LENGTH - 2], HEX) +
                 String(mac[WL_MAC_ADDR_LENGTH - 1], HEX);
  macID.toUpperCase();
  String AP_NameString = "ESP8266 " + macID;

  char AP_NameChar[AP_NameString.length() + 1];
  memset(AP_NameChar, 0, AP_NameString.length() + 1);

  for (int i=0; i<AP_NameString.length(); i++)
    AP_NameChar[i] = AP_NameString.charAt(i);
  WiFi.softAP(AP_NameChar, "password");
  webserver.on("/", handleWebsite);
  webserver.onNotFound(handleNotFound);
  webserver.begin();
}

void loop(){ 
  webserver.handleClient();
}
```

My strings are pretty long, but running ESP.getFreeHeap() at the end of my webpage, shows I still have 10k of available memory.

mytrain, 2015-09-28:
@awwende 
I've got a similar issue (no yet resolved) using ESP8266WebServer.
When a client download several files (html+png+js) some downloads (not all) get corrupted.
Activating the serial debug I noticed that 'LmacRxBlk:1' appears once or several time on each failed downloads. On the opposite if 'LmacRxBlk:1' does not appear the download is correct.

Corrupted downloads change form time to time especially after a hard reset of the ESP.

I hope this can tip somebody on how to resolve this issue.

awwende, 2015-09-28:
@mytrain This might not help you, but what I believe the issue is, is that my HTML data is too long when I send it. I was able to get it *sort of* working by moving my js to a separate file. I'm currently though having a wdt reset problem when I call `<script src="myscripts.js"></script>`. The page will load and the functions work, but when I hit refresh, I my board crashes.

bhamon, 2017-07-07:
I have the same issue. I leveraged it a bit by:
- Increasing the client socket timeout. It can be achieved by applying [this patch](https://github.com/bhamon/Arduino/commit/5000d1cf77979f0d90e967eb8c6c67d3ef2cde41) and specifying the `timeout` parameter of the `serveStatic` function to a larger value (like `30000` for 30 seconds).
- GZipping the larger files (a simple `gzip FILE_NAME` works, it'll be served as GZip encoded as long as the file extension ends with `.gz`).

However that's not a perfect solution. From time to time, the issue appears again.

d-a-v, 2017-10-20:
I tried the above sketch with no error.
Could you try the latest git version (or 2.4.0-rc2) ?