esp8266/Arduino #817: unreliable access point problem with STA+AP mode

mattvenn, 2015-09-25:
I've had an issue where I get unreliable access points on one ESP and fine on another. Seems the issue is stored in the flash.

When I looked at the diagnostic messagess (using WiFi.printDiag(Serial) from both modules I found this on the bad one:

Mode: STA+AP
PHY mode: N
Channel: 9
AP id: 0
Status: 1
Auto connect: 1

and this on the good one:

Mode: AP
PHY mode: N
Channel: 1
AP id: 0
Status: 255
Auto connect: 1

By adding this line:

WiFi.mode(WIFI_AP);

before I setup the AP with

WiFi.softAP(ssid);

the problem was resolved.

Is this worth adding to the example, or is there a bug somewhere with STA+AP mode?

rakeshpai, 2015-09-28:
I seem to be having a similar issue. Let me know if this is a different problem, and I can create a separate issue.

Until the esp connects to an existing wifi connection (whose config is already stored), everything works fine. I have a button which is supposed to start the AP. On the button click, I do the following:
```
WiFi.mode(WIFI_AP_STA);
WiFi.softAP(ssid);
```
In a terminal, I am pinging the esp from my laptop, while connected to it over wifi through my router. I was expecting the pings to continue working while the esp starts an AP. However, when I press the button, the pings stop immediately, returning no route to host. The AP starts successfully though.

I'm not a absolutely certain, but I can almost swear that this wasn't the case some time ago. I seem to recall that WiFi.softAP() didn't kill the existing connection as a STA. Maybe a recent update broke this? Or have I been doing it wrong, and there's some other way to get into STA+AP mode?

hamidrezamaddah, 2017-05-24:
Dear rakeshpai
I have a problem, same as yours
Have you solved your problem?
thnaks

rakeshpai, 2017-05-24:
Sorry, that was very long ago. I don't remember if/how I solved the issue.

hamidrezamaddah, 2017-05-24:
Thanks a lot :)

devyte, 2017-10-20:
This is most likely caused by the STA and AP parts sharing a single radio. When in STATIONAP mode, the station is trying to connect to an AP on a channel that may be different than the one used by the SoftAP, and it drags the SoftAP along with it when it switches to it.
The only known workaround is to scan for the AP, get its channel, configure the SoftAP side on that same channel, and connect the STA side.
Closing as known issue.

closed 2017-10-20 by devyte

McStick0, 2021-07-09:
only the well-known workaround does not work : /
I have Esp32 Wrover-B(LilyGo) and am connecting to the same Chanel - still not working.
One thing about Wifi Analyzer, while scaning i saw that my ESP have 1+5 chanel, what is that exacly? 
Only one way, is to buy LTE Modem  ESP, for AP+STA?