esp8266/Arduino #819: Access WiFi Network and PWM

plinioseniore, 2015-09-26:
Hi,

I'm working on an ESP8266 based WiFi Bulb based on [Souliss](http://souliss.net/articles/multicolor-wifi-bulb-arduino-based/), is working fine out of the startup.

The ESP8266 cannot connect to a WiFi network (either as STA or AP) if the output pins are used in a PWM, this only at the connection time, after it works fine. We found this because the module need 3-4 seconds to connect with the WiFi network and so, turning the LEDs on only after the connection has made, will result into a delayed light.

So if we turn ON leds, then start the connection the ESP8266 turn off the outputs and reboot due to the watchdog. Is it related to the timer used for the software PWM?

The pin used are the following:
#define	PIN_RED		13
#define	PIN_GREEN	12
#define	PIN_BLUE	14
#define	PIN_WHITE	2
#define	PIN_ENABLE	15

A picture is following for reference

[image]

Thanks.

Links2004, 2015-09-26:
I thin you may have an EMC problem here.
are the 4 Resistors below the antenna for the Leds?
can you try to lace the PCB strait upwards so the WiFi antenna comes away from the Resistors and the PWM.
it may also help to use an ESP-07 or ESP-12 thay have a shield to protect against EMC.

plinioseniore, 2015-09-26:
Hi,

yes the resistor below the antenna are used to drive power to the LED that are then dimmed through the mosfet on the top. 

I don't know if is an EMC problem, because is very well replicable and doesn't appear if we turn ON the LED just after the connection has been completed. Out of this first time, then the module works fine.

Does you know if there is any influence of the use of PWM and connection to the WiFi network?

plinioseniore, 2015-09-26:
For reference the node restarts with *ets Jan 8, 2013, rst cause: 4, boot mode:(1,6)*

Links2004, 2015-09-26:
rst cause: 4 means soft restart, for some reason the software watchdog is trigger.
can you post the setup / connection part of the code?
it the first time i see this kind of problem.

may this line blocks the WiFi too:
https://github.com/esp8266/Arduino/blob/esp8266/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_timer.c#L38

plinioseniore, 2015-09-26:
This is the sketch, thanks

```C
#include <ESP8266WiFi.h>

#define RED_PIN           13
#define GREEN_PIN         12
#define BLUE_PIN          14
#define WHITE_PIN         2
#define POWER_ENABLE_LED  15
#define OFF               0
#define ON                1

uint32_t ui32Counter;
const char * cSSID ="SSID";
const char * cPW ="PW";

void setup()
{
  pinMode(RED_PIN, OUTPUT);           // Red
  pinMode(GREEN_PIN, OUTPUT);         // Green
  pinMode(BLUE_PIN, OUTPUT);          // Blue
  pinMode(WHITE_PIN, OUTPUT);         // White
  pinMode(POWER_ENABLE_LED, OUTPUT);  // Power Led Enable
  analogWriteFreq(500);
  analogWriteRange(255);
  digitalWrite(RED_PIN, 0);
  digitalWrite(GREEN_PIN, 0);
  digitalWrite(BLUE_PIN, 0);
  digitalWrite(WHITE_PIN, 0);
  delay(500);
  digitalWrite(POWER_ENABLE_LED, ON);
  analogWrite(RED_PIN, 20);
  
  ui32Counter=0;
  Serial.begin(115200);
}

void loop()
{
  delay(1000);
  if (ui32Counter==1)
  {
    Serial.println("\n\rConnection to WiFi");
    WiFi.begin(cSSID, cPW);
    while (WiFi.status()!=WL_CONNECTED)
      delay(1);
    Serial.println("Connected");
    Serial.println("\n\rWIFI_STA");
    WiFi.mode(WIFI_STA);
    delay(5000);
    Serial.print("Local IP ");
    Serial.println(WiFi.localIP());
  }
  ui32Counter++; 
}
```

plinioseniore, 2015-09-27:
After some test we have identified that the problem is with PWM only, if we turn ON the LEDs without PWM (full bright) there is no reset or any other type of issue.

Links2004, 2015-09-27:
i get the error here to.
with debug enable ```Serial.setDebugOutput(true);```

```
 Connection to WiFi 
Fatal exception (0): 
epc1=0x40202680, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000 
 
Exception (0): 
epc1=0x40202680 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000 
 
ctx: cont  
sp: 3ffeb9e0 end: 3ffebe40 offset: 01a0 
 
>>>stack>>> 
3ffebb80:  400043da 00000030 00000016 ffffffff   
3ffebb90:  400044ab 3fffc718 3ffebc80 08000000   
3ffebba0:  60000200 08000000 00000003 00000000   
3ffebbb0:  0000ffff 00000001 04000002 003fe000   
3ffebbc0:  3fff32b0 000003fe 3fff1eb4 00000030   
3ffebbd0:  00000000 00000000 00000000 00000000   
3ffebbe0:  00000000 00000000 00000000 00000000   
3ffebbf0:  3fffc200 40202680 00000020 402026c8   
3ffebc00:  00000000 00000000 3fffc230 4000050c   
3ffebc10:  4021a1d6 00000030 00000008 ffffffff   
3ffebc20:  4021a1c6 00000001 00000000 0000003f   
3ffebc30:  3fff1f2a 3fff1e74 3fff1ea4 3ffeadc4   
3ffebc40:  203ffebe 3ffebdc0 00000001 3ffebd70   
3ffebc50:  3fff1da4 00000000 3fff1f74 00000030   
3ffebc60:  40000fb0 00000200 00000411 3fffc200   
3ffebc70:  00000611 ffffffff 00000020 00000003   
3ffebc80:  00000002 4000410f 3ffebda6 00000004   
3ffebc90:  3fffc718 40004a3c 000003fe 3fff32b0   
3ffebca0:  3fffc718 40102304 00000378 00000000   
3ffebcb0:  000003fe 40219db3 00000378 000003ff   
3ffebcc0:  00001000 40219e49 3fff1eb4 000003ff   
3ffebcd0:  000003fd 00000031 003fe000 3fff1eb4   
3ffebce0:  ffffff01 55aa55aa 00001f97 0000001c   
3ffebcf0:  0000001c 000000f1 00000051 000003ff   
3ffebd00:  4021a255 3ffec820 00000001 3ffebd70   
3ffebd10:  00000001 3ffebd90 4021a35b 3ffec820   
3ffebd20:  00000001 3ffebd70 00000031 3fff1da4   
3ffebd30:  3fff1f70 3ffebdd1 00000001 4021a392   
3ffebd40:  3ffeabb4 3ffe92ec 3ffe92f0 00000000   
3ffebd50:  3ffebd90 3fffc6fc 00000043 3ffebe64   
3ffebd60:  40206fbe 3fffdc20 3ffeabb4 40206f48   
3ffebd70:  44495353 00000000 00000000 00000000   
3ffebd80:  00000000 00000000 00000000 00000000   
3ffebd90:  00005750 00000000 00000000 00000000   
3ffebda0:  3ffeadc4 00000001 3ffe9929 4020e84c   
3ffebdb0:  3ffeadc4 00000013 3ffe92db 3ffebe6c   
3ffebdc0:  3fffdc20 00000014 3ffeadc4 4020df09   
3ffebdd0:  3ffe9900 3ffeadc4 3ffeadc4 4020df09   
3ffebde0:  00000004 00000000 3ffeadc4 4020e920   
3ffebdf0:  40201c1a 000003e8 3ffeadc4 3ffebe6c   
3ffebe00:  3ffeabb4 3ffeadc4 3ffea8b8 40202b55   
3ffebe10:  402010f5 00000000 3ffeadc4 40202af6   
3ffebe20:  3fffdc20 00000000 3ffebe64 40201152   
3ffebe30:  00000000 00000000 3ffeae20 40100398   
<<<stack<<< 
```

it is coming form timer1_isr_handler.
```
static volatile timercallback timer1_user_cb = NULL;

void timer1_isr_handler(void *para){
    if ((T1C & ((1 << TCAR) | (1 << TCIT))) == 0) TEIE &= ~TEIE1;//edge int disable
40202680:	fffc21        	l32r	a2, 40202670 <__analogWriteRange+0x14>
// ------------------------------------------------------------------ -
// timer 1

static volatile timercallback timer1_user_cb = NULL;

void timer1_isr_handler(void *para){
40202683:	f0c112        	addi	a1, a1, -16
    if ((T1C & ((1 << TCAR) | (1 << TCIT))) == 0) TEIE &= ~TEIE1;//edge int disable
40202686:	0020c0        	memw
40202689:	0238      	l32i.n	a3, a2, 0
// ------------------------------------------------------------------ -
// timer 1

static volatile timercallback timer1_user_cb = NULL;

void timer1_isr_handler(void *para){
4020268b:	3109      	s32i.n	a0, a1, 12
4020268d:	21c9      	s32i.n	a12, a1, 8
    if ((T1C & ((1 << TCAR) | (1 << TCIT))) == 0) TEIE &= ~TEIE1;//edge int disable
4020268f:	124c      	movi.n	a2, 65
40202691:	138327        	bany	a3, a2, 402026a8 <timer1_isr_handler+0x28>
40202694:	fff821        	l32r	a2, 40202674 <__analogWriteRange+0x18>
40202697:	d37c      	movi.n	a3, -3
40202699:	0020c0        	memw
4020269c:	002242        	l32i	a4, a2, 0
4020269f:	103430        	and	a3, a4, a3
402026a2:	0020c0        	memw
402026a5:	006232        	s32i	a3, a2, 0
    T1I = 0;
402026a8:	fff421        	l32r	a2, 40202678 <__analogWriteRange+0x1c>
402026ab:	030c      	movi.n	a3, 0
402026ad:	0020c0        	memw
402026b0:	0239      	s32i.n	a3, a2, 0
    if (timer1_user_cb) {
402026b2:	fff221        	l32r	a2, 4020267c <__analogWriteRange+0x20>
402026b5:	0020c0        	memw
402026b8:	0238      	l32i.n	a3, a2, 0
402026ba:	010316        	beqz	a3, 402026ce <timer1_isr_handler+0x4e>
        // to make ISR compatible to Arduino AVR model where interrupts are disabled
        // we disable them before we call the client ISR
        uint32_t savedPS = xt_rsil(15); // stop other interrupts 
402026bd:	006fc0        	rsil	a12, 15
        timer1_user_cb();
402026c0:	0020c0        	memw
402026c3:	0228      	l32i.n	a2, a2, 0
402026c5:	0002c0        	callx0	a2
        xt_wsr_ps(savedPS);
402026c8:	13e6c0        	wsr.ps	a12
402026cb:	002000        	isync
    }
}
402026ce:	3108      	l32i.n	a0, a1, 12
402026d0:	21c8      	l32i.n	a12, a1, 8
402026d2:	10c112        	addi	a1, a1, 16
402026d5:	f00d      	ret.n
402026d7:	bfd000        	excw
402026da:	fe          	.byte 0xfe
402026db:	3f          	.byte 0x3f
```

plinioseniore, 2015-09-27:
Really thanks for your support, unfortunately I'm not so in deep with this
SoC to understand.

I can understand that this timer is used for the SoftPWM, but I cannot
understand if there is a work around for this.

From Mobile.

igrr, 2015-09-28:
I have seen this behaviour as well when I tried to use Servo library. I had to enable servo *after* wifi was connected. Will look into this...

mytrain, 2015-09-28:
@Links2004 
How to you get the decompiled version of the ELF including C code.
I'm using Addr2line or readELF but I do not know how to get this result.

Links2004, 2015-09-28:
simply run: "$(XTENSA_TOOLS_ROOT)/xtensa-lx106-elf-objdump" -S $NAME.elf
all the library are plain asm but all the arduino stuff are getting c code if possible.
i have simply added to my makefile for eclipse as target.

igrr, 2015-09-30:
When the SDK calls SPIFlashWrite, flash memory is temporary unmapped from the CPU address space. If the timer interrupt fires at this moment, CPU can not read an instruction from flash, and fails with illegal instruction exception. One way to work around this would be to move interrupt handler into RAM. But then timer1_user_cb also has to be in RAM, as well as the rest of the functions it might call. Perhaps it would make sense to disable timer interrupts before calling SDK functions which save stuff into flash.

plinioseniore, 2015-09-30:
Hi @igrr disabling the timer interrupts will result in PWM output at full HIGH or LOW based on when interrupts are disabled, it would not really solve the problem.

Is SPIFlashWrite called only while joining the WiFi network?

Thanks,
Dario.

igrr, 2015-09-30:
SPIFlashWrite is called when WiFi config is changed. If you are joining the same network, you don't need to call WiFi.begin, SDK will join the last network without writing anything to flash.
Flash is also written when using SPIFFS, EEPROM, and doing OTA.

By the way, this is related to #836. As a summary, PWM needs the following things fixed:
- move timer handler and PWM routines into RAM (need to free up some RAM to do that, possibly by removing espconn from liblwip.a, or moving some strings into flash)
- use NMI instead of regular interrupt
- refactor existing critical sections in PWM routines

plinioseniore, 2015-09-30:
Thanks for the tips, in the while I will workaround as are suggesting

plinioseniore, 2015-10-01:
@igrr is there a way to read the last known network? So that at runtime we
can check if is the same of the sketch and skip the Wifi.begin().

In this way we could easily understand if we could use PWM from the very
begin or no.

Thanks.

From Mobile.

igrr, 2015-10-01:
Sure, take a look at this example:
https://github.com/esp8266/Arduino/blob/esp8266/hardware/esp8266com/esp8266/libraries/ESP8266mDNS/examples/OTA-mDNS-SPIFFS/OTA-mDNS-SPIFFS.ino#L244

commit 245c2377 (2015-10-06): Move timer ISR handlers and PWM handler into RAM (fix #819)
closed 2015-10-06 by igrr