esp8266/Arduino #820: Neopixel Library not working

ibaranov-cp, 2015-09-26:
Hey all,

I have Arduino IDE 1.6.5
Neopixel library 1.0.3
This is a ESP-12
I am using Sept 9th released Staging version

Using the standard strandtest, I cannot get a signal out on any pin, regardless of what pin I select, speed of 80 vs 160.

In fact, I noticed that when I tried the TX pin, the led light up, showing that the pin is set to active low, and simply stays there.

This seems to be a software bug, as otherwise the module works on every other example I have tried.

Is there anyone who got this working?

Protoneer, 2015-09-27:
I have instead used the NeopixelBus lib on most pins excluding 16.

hallard, 2015-09-27:
Yeah, I'm also using neopixelbus without any problem with various LED including Chinese PTH 5mm RGB LEB (with chipset inside)

jarek319, 2015-09-27:
You may have to update your neopixel library, I was using an older one and followed the instruction here: https://blog.adafruit.com/2015/06/30/updated-neopixel-support-for-esp8266-and-arduino-zero/

I'm running four neopixels off my GPIO 0 pin

sabas1080, 2015-10-05:
Worked in Huzzah 
http://www.instructables.com/id/ESP8266-with-Neopixeles/

closed 2015-10-18 by igrr