esp8266/Arduino #822: Check real flash size, suspecting wrong ID

hallard, 2015-09-27:
Hi guys,
I'm working now with ESP8266 for long and today I've got one doubt on flash memory size.
According the example sketch to display real flash size (reading flash ID) I've got a bunch of ESP07 with strange behavior. Test sketch says they got 1MB (using RealFlashSize API)
```
Connecting to: HOME-HOTSPOT...Connected
IP address   : 192.168.1.202
MAC address  : 18:FE:34:FC:B5:5C
Firmware     : Sep 27 2015 02:37:43
Flash real id: 0x001440E0
Flash Size   : 1024 KB
```
But as soon as I'm trying OTA upload stop (and works fine with ESP12 4M on same board). 
Using another ESP07, It simply crash after reboot if I compile for 1MB board (and upload using serial) and works fine as soon as I compile for 512KB. But once again RealFlashSize tell me it's 1MB flash (ID different from previous)
```
IP address   : 192.168.1.211
MAC address  : 18:FE:34:FC:B4:01
Firmware     : Sep 27 2015 01:24:09
Flash real id: 0x001440C8
Flash Size   : 1024 KB
```
So I really suspect bad flash ID or worst, counterfeit chip. So my question, is it possible to flash a sketch (said burned for 512KB) and in this sketch do read/write testing operation on flash between 
* 512Kb and 1M
* 1M and 2M
* 2M and 4M

Like this we for sure really know the real user available space ? Do you think it makes sense ?

Links2004, 2015-09-27:
the ID shows that it is an GD25Q80 form gigadevice.
http://www.gigadevice.com/product/detail/5/98.html

you only can access the range from 512Kb to 1M if you have selected 1M or more in the IDE.
the SDK blocks the access if you have selected 512Kb.

for testing you need to change the the 4M version linker script to ```eagle.flash.512k.ld``` here:
https://github.com/esp8266/Arduino/blob/esp8266/hardware/esp8266com/esp8266/boards.txt#L126
then the compiler keaps all stuff in the 512KB area and you can try to access the higher ones with the functions.

for flash access you can use:
https://github.com/esp8266/Arduino/blob/esp8266/hardware/esp8266com/esp8266/cores/esp8266/Esp.h#L122

hope this helps.

hallard, 2015-09-27:
@Links2004 
thks, exactly what I needed, I will give a try if I can find some time :-)

By the way I didn't succeed to find a global flash ID, list, the one on jedec does not match the ID brand found (or may be I read to fast) in these flash chip, any tips on this ?

Links2004, 2015-09-27:
the best list i know is in code  :)
http://code.coreboot.org/svn/flashrom/trunk/flashchips.h
note: the list sore the 1440 the other way around  as 0x4014

in case of 1440C8
14 - ? may like 2^xx is size in Byte ?
40 - ??
C8 - manufacturer ID

unfortunately i never find a good documentation for it.

hallard, 2015-09-27:
@Links2004 
I was resigned with code also, my best bet on quick dev, and on the classic question "Where's the updated documentation ?"
look In the code !!! ;-)

pgollor, 2015-09-29:
There is no documentation available with all flash chips!
I think the lower 8 bits from 0x40xx means 2^(xx -1) in kbits.
So the real size in MB is 2^(xx - 1) / 8 / 1024.

If you geht 0x4014 it is 1 MB.

pgollor, 2015-09-29:
You can get some information from Herr:
- http://www.esp8266.com/wiki/doku.php?id=esp8266-module-family
- http://code.coreboot.org/svn/flashrom/trunk/flashchips.h

closed 2015-10-18 by igrr